zigpy/zha-device-handlers #624: [BUG] Trafri blinds showing half battery again

blair287, 2020-12-07:
**Describe the bug**
A clear and concise description of what the bug is.

This happened before was fixed for a bit but the IKEA tradfri blind is showing half the battery. If you read the battery percentage remaining attribute it shows correct for example 60% but the battery entity will show 30%.

The open close remote is still ok.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

Battery to be same as cluster data

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Additional context**
Add any other context about the problem here. Please include full debug logs as well as the device signature.

dmulcahey, 2020-12-08:
Can you provide all of the signature info? The IKEA devices violate the ZCL spec for battery reporting so we need to quirk it to fix it.

blair287, 2020-12-09:
> Can you provide all of the signature info? The IKEA devices violate the ZCL spec for battery reporting so we need to quirk it to fix it.

Do I need to repair to get that for you?

dmulcahey, 2020-12-09:
There is a dialog on the device screen in HA. Just get the contents of that.

blair287, 2020-12-10:
> There is a dialog on the device screen in HA. Just get the contents of that.

Sorry would have copied but mobile app or mobile won't let me copy and not go access to pc for a bit.

[image]

blair287, 2020-12-10:
> There is a dialog on the device screen in HA. Just get the contents of that.

`{
  "node_descriptor": "NodeDescriptor(byte1=2, byte2=64, mac_capability_flags=128, manufacturer_code=4476, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=0)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0202",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0020",
        "0x0102",
        "0x1000"
      ],
      "out_clusters": [
        "0x0019",
        "0x1000"
      ]
    }
  },
  "manufacturer": "IKEA of Sweden",
  "model": "FYRTUR block-out roller blind",
  "class": "zigpy.device.Device"
}`

Adminiuga, 2020-12-10:
ah, they removed the IKEA manufacturer cluster.  we should start include device version in the zigpy signature.

MattWestb, 2020-12-10:
@blair287  Wath version is your blinds having ?
I think (I dont have any blinds) the latest version is 2.2.009.

dmulcahey, 2020-12-10:
> ah, they removed the IKEA manufacturer cluster. we should start include device version in the zigpy signature.

I think this could be helpful in quirks matching too

blair287, 2020-12-11:
> @blair287 Wath version is your blinds having ?
> I think (I dont have any blinds) the latest version is 2.2.009.

Mine are on the latest version 2.2.009 I have it's enabled.

PR #659 (2020-12-21): Add alternate ikea blinds signature
fixes: #624
closed 2020-12-21 by dmulcahey