zigpy/zha-device-handlers #631: [Device Support Request] Revolt NX-4911 Thermostatic Radiator Valve Controller (Tuya TS0601 based)

docgalaxyblock, 2020-12-11:
https://zigbee.blakadder.com/Revolt_NX-4911.html
Z2M Issues
https://github.com/Koenkk/zigbee2mqtt/issues/4933
https://github.com/Koenkk/zigbee2mqtt/issues/3821

**Describe the solution you'd like**
Full support

**Device signature - this can be acquired by removing the device from ZHA and pairing it again from the add devices screen. Be sure to add the entire content of the log panel after pairing the device to a code block below this line.**
[Device signature and pairing log at my dev VM](https://pastebin.com/MvuQtvYz)

**Additional context**
Same identifier as the tuya curtain motor

docgalaxyblock, 2020-12-11:
Signature
```
{
  "node_descriptor": "NodeDescriptor(byte1=2, byte2=64, mac_capability_flags=128, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=0)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0051",
      "in_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0xef00"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZE200_kfvq6avy",
  "model": "TS0601",
  "class": "zigpy.device.Device"
}

docgalaxyblock, 2020-12-11:
@FrnchFrgg Hi, if you have time
Would it be possible for you to create a quirk please
THX

MattWestb, 2020-12-11:
Device info
fvq6avy
by _TYST11_kfvq6avy
IKEA Billy EZSP
Zigbee info
IEEE: bc:33:ac:ff:fe:03:de:fb
Nwk: 0xce16
Device Type: EndDevice
LQI: 248
RSSI: -38
Last Seen: 2020-12-11T14:47:06
Power Source: Battery or Unknown
Quirk: zhaquirks.tuya.valve.SiterwellGS361

```
{
  "node_descriptor": "NodeDescriptor(byte1=2, byte2=64, mac_capability_flags=128, manufacturer_code=0, maximum_buffer_size=82, maximum_incoming_transfer_size=255, server_mask=11264, maximum_outgoing_transfer_size=255, descriptor_capability_field=0)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0301",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x0201",
        "0x0204",
        "0xef00"
      ],
      "out_clusters": [
        "0x0003",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TYST11_kfvq6avy",
  "model": "fvq6avy",
  "class": "zhaquirks.tuya.valve.SiterwellGS361"
}
```

The `zhaquirks.tuya.valve.SiterwellGS361` is not loaded but all other looks like my working ones.

I looking for then it was added and it its in the current HA or in the beta test (I have forcing HA loading the next ZHA version for the 1.0 HA).

MattWestb, 2020-12-11:
The zhaquirks.tuya.valve.SiterwellGS361 is for 13 days added in ZHA but you need being on the HA 1.0 beta for getting it in HA.

HA 1.0 is planed being released 13/12 so i think better waiting if you is not very urgent testing the beta of HA.

By the way it can being that you is needing the adapter for mounting it if you heating is having no 30mm mount like my harz is having 28 mm.
https://www.conrad.at/de/p/700100-heizkoerper-ventil-adapter-passend-fuer-heizkoerper-herz-560382.html is working OK for my but you can getting the same from Amazon.

docgalaxyblock, 2020-12-11:
OK, I am running b5
Do you know if it is at the dev branch? Then I will fire up my testing vm to dev and check it again
(🇩🇪) We have the 30mm ones 😉

MattWestb, 2020-12-11:
HA commit: https://github.com/home-assistant/core/pull/43707
B3 have one newer commit of ZHA so it must being in your beta.
Do you have installed some custom dev packages that is overriding the ones from B3 ?

No AT but from SE.

MattWestb, 2020-12-11:
If you have one more dongle its easier starting the vm and reload the b5 and do one cleaning before so not old junk is stored and loaded (if having one deps folder in HA configure that have folders in side its dev packages that is installed and overriding the from HA have installed then delete all in the deps folder)

MattWestb, 2020-12-11:
The devs was making some changes that was breaking the tuya model detection in between zha-quirks==0.0.47 and 0.0.48 but is should being fixed in the last HA beta.
I running 0.0.47 and its working but i cant guarantee that 0.0.48 is.
If not trying some beta 0 - 3 and see if its working.

docgalaxyblock, 2020-12-11:
I have tried b1-b5
b0,b1: it gets the MoesBHT quirk but that is not working
b3++ nothing

MattWestb, 2020-12-11:
Strange but i have not trying the betas.
@Adminiuga Its looks that the fix for the tuya devices is not working OK.
I using `quirks==0.0.47` and getting the `zhaquirks.tuya.valve.SiterwellGS361` patched in HA 0.118.
Its the same string for our devices that tuya is sending so somthing is not OK.

Anny ide wat is going wrong ??

Adminiuga, 2020-12-11:
use quirks==0.0.48

MattWestb, 2020-12-11:
I was using https://github.com/home-assistant/core/pull/43707 on HA 0.118 and its was working.
Have updating it to quirks==0.0.48 and looking if its working for my.

@docgalaxyblock is trying the 1.0 betas and its dont working for hi.
>I have tried b1-b5
b0,b1: it gets the MoesBHT quirk but that is not working
b3++ nothing

docgalaxyblock, 2020-12-11:
> 
> 
> https://zigbee.blakadder.com/Revolt_NX-4911.html
> Z2M Issues
> [Koenkk/zigbee2mqtt#4933](https://github.com/Koenkk/zigbee2mqtt/issues/4933)
> [Koenkk/zigbee2mqtt#3821](https://github.com/Koenkk/zigbee2mqtt/issues/3821)
> 
> TS0601
> von _TZE200_kfvq6avy
> 
> **Describe the solution you'd like**
> 
>     * [ ]  Temperature control
> 
>     * [ ]  Temperature sensor
> 
>     * [ ]  Mode Switch
> 
>     * [ ]  Battery
> 
>     * [ ]  Valve detection
> 
>     * [ ]  Open-Window detection
> 
>     * [ ]  Child protection
> 
> 
> **Device signature - this can be acquired by removing the device from ZHA and pairing it again from the add devices screen. Be sure to add the entire content of the log panel after pairing the device to a code block below this line.**
> [Device signature and pairing log at my dev VM](https://pastebin.com/MvuQtvYz)
> 
> **Additional context**
> Same identifier as the tuya curtain motor
> Same Zigbee module as https://zigbee.blakadder.com/Moes_HY368-ZB.html
> 
> Seems to be a different revisons as
> #581 #538

@MattWestb  Last line ^^

MattWestb, 2020-12-11:
@docgalaxyblock Can you changing the string in the quirk `zhaquirks/tuya/valve.py`
and adding your devices string in line 183  `MODELS_INFO: [("_TYST11_jeaxp72v", "eaxp72v"), ("_TYST11_kfvq6avy", "fvq6avy")],`
to      ` MODELS_INFO: [("_TYST11_jeaxp72v", "eaxp72v"), ("_TYST11_kfvq6avy", "fvq6avy"), ("_TZE200_kfvq6avy", "fvq6avy")],`

docgalaxyblock, 2020-12-11:
@MattWestb How so I edit the file? and where is it
I have no deeper "dev experience" as the vscode addon...

MattWestb, 2020-12-11:
Is your ha in native linux or in docker ?

docgalaxyblock, 2020-12-11:
Proxmox HassOS VM

MattWestb, 2020-12-11:
Than i dont knowing where its installing the packages.
Can you finding the directory `zhaquirks `somewhere ?

docgalaxyblock, 2020-12-11:
The shell is different `ash` is it I think and there I have no experience

MattWestb, 2020-12-11:
I have the same problem with the docker installation :-((

ash is working in docker as cli.

MattWestb, 2020-12-11:
In docker its  /usr/local/lib/python3.8/site-packages/zhaquirks/tuya/valve.py

MattWestb, 2020-12-11:
I was installing suplemon in docker that is more normal terminal editor but i dont knowing if its possible to do in HassOS.

If pip is supported try `pip install suplemon` to install Suplemon.

And then `suplemon /usr/local/lib/python3.8/site-packages/zhaquirks/tuya/valve.py`

MattWestb, 2020-12-11:
If you have file editor installed as addon you can copy the /usr/local/lib/python3.8/site-packages/zhaquirks/tuya/valve.py to /config
`cp /usr/local/lib/python3.8/site-packages/zhaquirks/tuya/valve.py /config/valve.py`

Then the valve.py is in your hass config folder and you can editing it.
Then copying it back and restarting HA.

docgalaxyblock, 2020-12-11:
@MattWestb Are you at the HA Discord?

MattWestb, 2020-12-11:
Sorry no :-(

I have trying making one PR for the new zigbee module and hoping i have getting it right (Was the first time doing it).

docgalaxyblock, 2020-12-11:
Ok then we stay here for debugging
I have tested B2 and b5 and it did not changed anything
At b2 it gets the `zhaquirks.tuya.electric_heating.MoesBHT`
and at b5 it gets no quirk

MattWestb, 2020-12-11:
Very true as you was saying from the beginning its one new zigbee module and its not working.
I have adding it in the repro with one pull request but its taking time getting thru.

Its only one line that is missing of the new ID and then it should working like the old one.
So the best not getting any quirk = all other things is working normal.

Its possible editing the file but not so easy describing how in the container.

tube0013, 2020-12-11:
Beta 3 and after have stricter matching, the Manufacturer and Model much match exactly for the quirk to apply. I just had to fix this in the Konke button quirk, pre beta 3 it worked with "KONKE" as the manufacture, but it was really sending "Konke" so it failed to match starting in beta 3.

docgalaxyblock, 2020-12-11:
I have edited the file
Installed the portainer addon -> bash into homeassistant container ->and used suplemon

MattWestb, 2020-12-11:
@docgalaxyblock That sounds like you have done it right. What happens after restart of HA ?

@tube0013 I was thinking it was that first but the supplier have changing the zigbee module but the manufacturer ID is the same. I have trying do one PR but black is complaining is it OK or have i done it wrong ??

tube0013, 2020-12-11:
I have trying do one PR but black is complaining is it OK or have i done it wrong ??

Install black on your local machine and run it against your quirk file, it will hopefully fix the issues failing in the check

MattWestb, 2020-12-11:
I was doing it in the githubs web editor and not local as it was only one more ID to adding. :-((

tube0013, 2020-12-11:
> I was doing it in the githubs web editor and not local as it was only one more ID to adding. :-((

I ran it and put the diff output in a comment on the PR.

MattWestb, 2020-12-11:
Thanks tube0013 I was managen getting it and updating the PR :-)))

MattWestb, 2020-12-11:
@tube0013 Do you have experience of updating the local quirk in docker container installation and explaining it for the user ? 
I have doing manual updated but not so easy explaining how  to do it and very likely its better / easier ways to do it.

tube0013, 2020-12-11:
I actually wrote the instructions that are in the readme for that, so that's the best I got, if followed I think it makes sense.

docgalaxyblock, 2020-12-11:
> I have tested B2 and b5 and it did not changed anything
> At b2 it gets the `zhaquirks.tuya.electric_heating.MoesBHT`
> and at b5 it gets no quirk
I have completly restarted the testing VM

@MattWestb This 😞 
And checked both again

MattWestb, 2020-12-11:
So https://github.com/zigpy/zha-device-handlers#testing-quirks-in-development-in-docker-based-install shall working for hi and also using suplemon for edit the file and not need downloading the complete zhaquirk ala `suplemon /usr/local/lib/python3.8/site-packages/zhaquirks/tuya/valve.py` with eding line 138 to  `MODELS_INFO: [("_TYST11_jeaxp72v", "eaxp72v"), ("_TYST11_kfvq6avy", "fvq6avy"), ("_TZE200_kfvq6avy", "fvq6avy")],`
and deleting pycache `rm -rf /usr/local/lib/python3.8/site-packages/zhaquirks/tuya/__pycache__/`

By the way good instruction I was stretching for somthing like that then testing Bellows but was using the Custom deps deployment in HA.

MattWestb, 2020-12-11:
@docgalaxyblock If you have changing the local zhaquirk file  its being overwritten if up/downgrading HA and you must redoing the changes agen. :-((

docgalaxyblock, 2020-12-11:
@MattWestb Year I know that. I am quite familar at docker ^^
I have checked after reboot if the file was resetted

MattWestb, 2020-12-11:
Im not !!
I think the deleting of pycache is needed for it  to initiating the change.
`rm -rf /usr/local/lib/python3.8/site-packages/zhaquirks/tuya/__pycache__/`

docgalaxyblock, 2020-12-11:
That fix is also not working 😢

MattWestb, 2020-12-11:
I think im being wrong about the modell :-(((

The cluster is not fitting for 100% with the old Revolt quirk so it not triggeriged.
But the MoesHY368 is the same cluster.
Try editing line 174 to:
`        MODELS_INFO: [("_TZE200_ckud7u2l", "TS0601"), ("_TZE200_kfvq6avy", "TS0601")],`

docgalaxyblock, 2020-12-11:
b5 now it is `zhaquirks.tuya.valve.MoesHY368`
Battery ✔️ ; Temp report ✔️ ; switching away mode - heating ✔️ ; Temp set ✔️

MattWestb, 2020-12-11:
THAT'S GREAT !!!

I should updating the quirk tomorrow so it can being merged for other users but is missing HA 1.0 so make one copy of your changes for sunday (the release of HA 1.0)

You can sending mode command 3 from cluster management for getting it install mode and dont need taking the battery out :-))

Great work done and also big thanks to @tube0013 for "black help" !!

docgalaxyblock, 2020-12-11:
@MattWestb Did you know what are the four F Modes (old the botton very long) F4 is reset

MattWestb, 2020-12-11:
Children lock, Valve detect, Valve position, Schudel mode and service / initiating mode is missing but it was not possible testing before the tuya TRV was merged and HA need adapting the integration of all things.

Its need more work with the tuya quirk and also ZHA integration of all functioning we is having but its need more devs to helping doing that.

I have getting the readed temperature in HA to have 0.1° precision instead of 0.5° that it is in 0.118.

I was only testing 0 - 3 and 3 was valve reset / calibrating (long press the front) the 4 i was not seeing any sign of.

I have updating the PR with the correct device type so it can being merged then the devs like.

docgalaxyblock, 2020-12-11:
@MattWestb Maybe wanna join the discord? Would you teach me a bit? That would be awseome
At this TVR F4 is the calibration procedere nad I got also F4 that is the window open feature

MattWestb, 2020-12-11:
[image]
My bathroom with 0.5° precision and battery low (not heating its warm enough 355 days / year).
[image]

docgalaxyblock, 2020-12-11:
I could not read that cluster ^^
hw_version = 1
@MattWestb Again do you have discord or not? do not want to "spam" here

docgalaxyblock, 2020-12-11:
Notes
`suplemon /usr/local/lib/python3.8/site-packages/zhaquirks/tuya/valve.py`
Edited lines:
```
138         MODELS_INFO: [("_TYST11_jeaxp72v", "eaxp72v"), ("_TYST11_kfvq6avy", "fvq6avy"), ("_TZE200_kfvq6avy", "fvq6avy")],
174         MODELS_INFO: [("_TZE200_ckud7u2l", "TS0601"), ("_TZE200_kfvq6avy", "TS0601")],

Adminiuga, 2020-12-12:
You can't have two `MODELS_INFO` lines. Should be one with opening and closing square bracket + comma and inside the brackets tuples `("manufacturer name" , "model name"),` separated by commas

MattWestb, 2020-12-12:
@Adminiuga The first (line 138) was not working then the cluster is not matching and i was restoring it but patching it in the PR for black not complaining:
```
        MODELS_INFO: [
            ("_TYST11_jeaxp72v", "eaxp72v"),
            ("_TYST11_kfvq6avy", "fvq6avy"),
        ],
```
The second (line 174) i was splitting for black then adding the second record: 
```
        MODELS_INFO: [
            ("_TZE200_ckud7u2l", "TS0601"),
            ("_TZE200_kfvq6avy", "fvq6avy"),
        ],
```
Black is happy in the PR with this 2 model blocks.

For the user then testing the "long line " version that black dont like is working (being triggered) but dont looks so  nice:
````
        MODELS_INFO: [("_TZE200_ckud7u2l", "TS0601"), ("_TZE200_kfvq6avy", "TS0601")],
````

@docgalaxyblock its better not patching the line 138 in your test then its only makes more work for the system  but its working if you have done it and loving it.

@Adminiuga Do you like have the splitted line version that black is liking or the long line (original style with multi entice) that black is complaining about in the PR ?

Adminiuga, 2020-12-12:
Black one

MattWestb, 2020-12-12:
@docgalaxyblock Is your TRVs still working well and keeping the cold outside ?

docgalaxyblock, 2020-12-12:
@MattWestb Yes
I think the bath radiator is a bit clogged or the vent is not the newest one
Because it 2°C off to the sonoff sensor so I think because it is fully opend and the bath does not feels like 23°C more like the 20°C that the sonoff reports
[image]

Need to install the two others...
Where did you get your different brandet one?
And for testing purpose I have ordered the HY368 😉

FrnchFrgg, 2020-12-12:
Beware that configuring the temperature offsets is not yet implemented. But the valve should actually support that, which means you'll be eventually able to calibrate the temperature readings.

MattWestb, 2020-12-12:
The TRV have one NTC (old very un precise temperature sensor) and if not offset the measured temperature reported  can being little (or mutch) of scale.

If the valve is not smooth put some drop good fine oil on it (Nähmaschinenöl or from shaving machine) and move it little before mounting the TRV is normally helping.

I was buying on 5 pack with GW from pearl.at its the same as yours but with other zigbee module as you knowing.

The HY368 have 2 versions and the second one have very bad looking display of numbers that i don't like. Some have getting firmware updates. They have more functions then ours is having but i dont like the design.

Good if you is buying some more types for testing then its easier for our tuya codemaster @FrnchFrgg doing the coding and you is knowing how to do thesting if the quirk :-))

And one more time thanks for the coding FrnchFrgg !!

All new functions you can getting "extracting" from our devices is more than welcome ;-)))

MattWestb, 2020-12-13:
If all is right done it should being possible using quirks==0.0.49 in HA 0.118 and 1.0.0 then the patch is merged and one new release is made :-)
Easiest is installing `Custom deps deployment`  as addon in HA and configure with and running it.

```
pypi:
  - quirks==0.0.49
apk: []
```

The "old" docker patching is working well 2 !!

MattWestb, 2020-12-13:
False alarm then dmulcahey have pushed it in the last minute in HA 1.0 https://github.com/home-assistant/core/pull/44173  !!!!

[20 more replies]

closed 2023-07-10 by github-actions[bot]