zigpy/zha-device-handlers #635: [Device Support Request] Model E1E-G7F manufactured by Sengled

ShaneSpina, 2020-12-13:
https://zigbee.blakadder.com/Sengled_E1E-G7F.html

**Is your feature request related to a problem? Please describe.**
I would like to add this remote to my HA environment

**Describe the solution you'd like**
It has 4 buttons and I'd like to make use of all 4 in an automation, either in NR or in HA

**Device signature - this can be acquired by removing the device from ZHA and pairing it again from the add devices screen. Be sure to add the entire content of the log panel after pairing the device to a code block below this line.**
{
  "node_descriptor": "NodeDescriptor(byte1=2, byte2=64, mac_capability_flags=128, manufacturer_code=4448, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=0)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0104",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x0020",
        "0xfc11"
      ],
      "out_clusters": [
        "0x0003",
        "0x0004",
        "0x0006",
        "0x0008",
        "0xfc10"
      ]
    }
  },
  "manufacturer": "sengled",
  "model": "E1E-G7F",
  "class": "zigpy.device.Device"
}

**Additional context**
Not sure what if anything you'd want in here..

compeek, 2021-01-06:
This seems to be a duplicate of https://github.com/zigpy/zha-device-handlers/issues/140

I'm mostly telling you though because there's some discussion happening there you might be interested in. I took a stab at implementing support for this device tonight, and I got something partially working that I'll try to finish up soon. See this comment: https://github.com/zigpy/zha-device-handlers/issues/140

ShaneSpina, 2021-01-06:
Thanks.  I'll delete and re-add one of my remotes tonight and see what happens!

compeek, 2021-01-06:
Well, just to be clear, I only have it working locally. You would have to copy the code I shared into the right place to test it. I'm hoping to get it polished up and submit a PR to the repo so it can be made official if they approve it.

NobleWolf, 2021-01-08:
@compeek Awesome! I'm excited for this because I got the remote as a gift. I love the design and would hate to return it for an Ikea Tradfri one.

compeek, 2021-01-08:
> Alex Hauck Awesome! I'm excited for this because I got the remote as a gift. I love the design and would hate to return it for an Ikea Tradfri one.

The Hue Dimmer Switch is by far the best remote I've used with Home Assistant, but it's pricey at $25 a piece. But the events are instant and it has no weird behavior. The next best thing I've found so far is this Sengled one, which I got for $15 on Amazon recently. It definitely feels cheaper, but it looks very nice. Unfortunately the events are slightly delayed (by about a second) in most cases, but otherwise I'm pretty happy with it..

compeek, 2021-02-04:
@ShaneSpina The new device handler is finally showing up in Home Assistant for me now with the 2021.2.0 release! Let me know how it works for you.

closed 2021-08-12 by github-actions[bot]