zigpy/zha-device-handlers #639: [Device Support Request] Aqara relay (relay_c2acn01) Interlock support

Kalypox, 2020-12-14:
**Is your feature request related to a problem? Please describe.**
Prohibit the possibility to power on the 2 ways at the same time

**Describe the solution you'd like**
Add the interlock option to HA

**Device signature - this can be acquired by removing the device from ZHA and pairing it again from the add devices screen. Be sure to add the entire content of the log panel after pairing the device to a code block below this line.**
```
Device 0x0000 (00:15:8d:00:06:32:74:40) left the network
Device 0x010e (00:15:8d:00:06:32:74:40) joined the network
Device 00:15:8d:00:06:32:74:40 changed id (0x6ac2 => 0x010e)
Canceling old initialize call
[0x010e] Requesting 'Node Descriptor'
Tries remaining: 2
[0x010e] Extending timeout for 0xa5 request
[0x010e:zdo] ZDO request ZDOCmd.Device_annce: [0x010E, 00:15:8d:00:06:32:74:40, 142]
[0x010e] Node Descriptor: NodeDescriptor(byte1=1, byte2=64, mac_capability_flags=142, manufacturer_code=4151, maximum_buffer_size=127, maximum_incoming_transfer_size=100, server_mask=0, maximum_outgoing_transfer_size=100, descriptor_capability_field=0)
[0x010e] Discovering endpoints
Tries remaining: 3
[0x010e] Discovered endpoints: [1, 2]
[0x010e:1] Discovering endpoint information
Tries remaining: 3
[0x010e:1] Discovered endpoint information: SizePrefixedSimpleDescriptor(endpoint=1, profile=260, device_type=257, device_version=2, input_clusters=[0, 3, 4, 5, 1, 2, 10, 6, 16, 2820, 12], output_clusters=[25, 10])
[0x010e:1:0x0000] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=171 command_id=Command.Read_Attributes_rsp>
[0x010e:1] Manufacturer: LUMI
[0x010e:1] Model: lumi.relay.c2acn01
[0x010e:2] Discovering endpoint information
Tries remaining: 3
[0x010e:1:0x0000] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=171 command_id=Command.Read_Attributes_rsp>
[0x010e:1:0x0000] ZCL request 0x0001: [[ReadAttributeRecord(attrid=4, status=<Status.SUCCESS: 0>, value=<TypeValue type=CharacterString, value=LUMI>), ReadAttributeRecord(attrid=5, status=<Status.SUCCESS: 0>, value=<TypeValue type=CharacterString, value=lumi.relay.c2acn01>)]]
[0x010e:1:0x0000] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=171 command_id=Command.Read_Attributes_rsp>
[0x010e:1:0x0000] ZCL request 0x0001: [[ReadAttributeRecord(attrid=4, status=<Status.SUCCESS: 0>, value=<TypeValue type=CharacterString, value=LUMI>), ReadAttributeRecord(attrid=5, status=<Status.SUCCESS: 0>, value=<TypeValue type=CharacterString, value=lumi.relay.c2acn01>)]]
[0x010e:2] Discovered endpoint information: SizePrefixedSimpleDescriptor(endpoint=2, profile=260, device_type=257, device_version=2, input_clusters=[6, 16, 4, 5], output_clusters=[])
Checking quirks for LUMI lumi.relay.c2acn01 (00:15:8d:00:06:32:74:40)
Considering <class 'zhaquirks.xiaomi.aqara.relay_c2acn01.Relay'>
Found custom device replacement for 00:15:8d:00:06:32:74:40: <class 'zhaquirks.xiaomi.aqara.relay_c2acn01.Relay'>
device - 0x010E:00:15:8d:00:06:32:74:40 entering async_device_initialized - is_new_join: True
device - 0x010E:00:15:8d:00:06:32:74:40 has joined the ZHA zigbee network
[0x010E](lumi.relay.c2acn01): started configuration
[0x010E:ZDO](lumi.relay.c2acn01): 'async_configure' stage succeeded
[0x010e:1:0x0000] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=3 command_id=Command.Report_Attributes>
[0x010e:1:0x0000] ZCL request 0x000a: [[Attribute(attrid=5, value=<TypeValue type=CharacterString, value=lumi.relay.c2acn01>), Attribute(attrid=1, value=<TypeValue type=uint8_t, value=1>)]]
[0x010e:1:0x0000] Attribute report received: model=lumi.relay.c2acn01, app_version=1
[0x010e:1:0x0000] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=3 command_id=Command.Report_Attributes>
[0x010e:1:0x0000] ZCL request 0x000a: [[Attribute(attrid=5, value=<TypeValue type=CharacterString, value=lumi.relay.c2acn01>), Attribute(attrid=1, value=<TypeValue type=uint8_t, value=1>)]]
[0x010e:1:0x0000] Attribute report received: model=lumi.relay.c2acn01, app_version=1
[0x010e:1:0x0000] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=3 command_id=Command.Report_Attributes>
[0x010e:1:0x0000] ZCL request 0x000a: [[Attribute(attrid=5, value=<TypeValue type=CharacterString, value=lumi.relay.c2acn01>), Attribute(attrid=1, value=<TypeValue type=uint8_t, value=1>)]]
[0x010e:1:0x0000] Attribute report received: model=lumi.relay.c2acn01, app_version=1
[0x010E:1:0x0006]: skipping channel configuration
[0x010E:1:0x0000]: skipping channel configuration
[0x010E:1:0x0b04]: skipping channel configuration
[0x010E:1:0x0019]: skipping channel configuration
[0x010E:2:0x0006]: skipping channel configuration
[0x010E:1:0x0006]: 'async_configure' stage succeeded
[0x010E:1:0x0000]: 'async_configure' stage succeeded
[0x010E:1:0x0b04]: 'async_configure' stage succeeded
[0x010E:1:0x0019]: 'async_configure' stage succeeded
[0x010E:2:0x0006]: 'async_configure' stage succeeded
[0x010E](lumi.relay.c2acn01): completed configuration
[0x010E](lumi.relay.c2acn01): stored in registry: ZhaDeviceEntry(name='LUMI lumi.relay.c2acn01', ieee='00:15:8d:00:06:32:74:40', last_seen=1607953705.7980297)
[0x010E](lumi.relay.c2acn01): started initialization
[0x010E:ZDO](lumi.relay.c2acn01): 'async_initialize' stage succeeded
[0x010e:1:0x0006] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=175 command_id=Command.Read_Attributes_rsp>
[0x010e:1:0x0006] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=175 command_id=Command.Read_Attributes_rsp>
[0x010e] Invalid state on future for 0xaf seq -- probably duplicate response
[0x010e:2:0x0006] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=177 command_id=Command.Read_Attributes_rsp>
[0x010e:2:0x0006] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=177 command_id=Command.Read_Attributes_rsp>
[0x010e:2:0x0006] ZCL request 0x0001: [[ReadAttributeRecord(attrid=0, status=<Status.SUCCESS: 0>, value=<TypeValue type=Bool, value=Bool.false>)]]
[0x010E:2:0x0006]: 'async_initialize' stage succeeded
[... 38 lines cut ...]
[0xd4e6] Delivery error for seq # 0xbb, on endpoint id 1 cluster 0x0b04: ZiGate doesn't answer to command
[0xD4E6:1:0x0b04]: async_update
[0x6E52:1:0x0b04]: async_update
[0x010E:1:0x0b04]: async_update
[0xd4e6:1:0x0b04] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=195 command_id=Command.Read_Attributes_rsp>
[0x6e52:1:0x0b04] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=197 command_id=Command.Read_Attributes_rsp>
[0xd4e6] Delivery error for seq # 0xc3, on endpoint id 1 cluster 0x0b04: ZiGate doesn't answer to command
[0xD4E6:1:0x0b04]: async_update
[0x6E52:1:0x0b04]: async_update
[0x010E:1:0x0b04]: async_update
[0xd4e6:1:0x0b04] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=199 command_id=Command.Read_Attributes_rsp>
[... cut ...]
[0xd4e6:1:0x0b04] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=205 command_id=Command.Read_Attributes_rsp>
Read attribute for: cluster_id: [2820] cluster_type: [in] endpoint_id: [1] attribute: [1291] manufacturer: [None] response --> response: [({1291: 0}, {})]
[0xd4e6] Delivery error for seq # 0xc7, on endpoint id 1 cluster 0x0b04: ZiGate doesn't answer to command
```

**Additional context**
I see works here:
https://github.com/Koenkk/zigbee-herdsman-converters/issues/544
https://youtu.be/omGVI5nFK04?t=40 (see at 40seconds)

Thanks !

Adminiuga, 2020-12-14:
Orig message:

Enable:
```
No.     Time           Source                Destination           Protocol Length Info
    108 134.067464     0x0000                0xed08                ZigBee HA 54     ZCL: Write Attributes, Seq: 75

Frame 108: 54 bytes on wire (432 bits), 54 bytes captured (432 bits) on interface 0
IEEE 802.15.4 Data, Dst: 0xed08, Src: 0x0000
ZigBee Network Layer Data, Dst: 0xed08, Src: 0x0000
ZigBee Application Support Layer Data, Dst Endpt: 1, Src Endpt: 1
ZigBee Cluster Library Frame, Mfr: Unknown (0x115f), Command: Write Attributes, Seq: 75
    Frame Control Field: Profile-wide (0x14)
        .... ..00 = Frame Type: Profile-wide (0x0)
        .... .1.. = Manufacturer Specific: True
        .... 0... = Direction: Client to Server
        ...1 .... = Disable Default Response: True
    Manufacturer Code: Unknown (0x115f)
    Sequence Number: 75
    Command: Write Attributes (0x02)
    Attribute Field, Boolean: 0x01
        Attribute: 0xff06
        Data Type: Boolean (0x10)
        0000 0001 = Boolean: True
```

when disabling it I get this:

```
No.     Time           Source                Destination           Protocol Length Info
    154 179.949980     0x0000                0xed08                ZigBee HA 54     ZCL: Write Attributes, Seq: 82

Frame 154: 54 bytes on wire (432 bits), 54 bytes captured (432 bits) on interface 0
IEEE 802.15.4 Data, Dst: 0xed08, Src: 0x0000
ZigBee Network Layer Data, Dst: 0xed08, Src: 0x0000
ZigBee Application Support Layer Data, Dst Endpt: 1, Src Endpt: 1
ZigBee Cluster Library Frame, Mfr: Unknown (0x115f), Command: Write Attributes, Seq: 82
    Frame Control Field: Profile-wide (0x14)
        .... ..00 = Frame Type: Profile-wide (0x0)
        .... .1.. = Manufacturer Specific: True
        .... 0... = Direction: Client to Server
        ...1 .... = Disable Default Response: True
    Manufacturer Code: Unknown (0x115f)
    Sequence Number: 82
    Command: Write Attributes (0x02)
    Attribute Field, Boolean: 0x00
        Attribute: 0xff06
        Data Type: Boolean (0x10)
        0000 0000 = Boolean: False
```

Adminiuga, 2020-12-14:
Binary output cluster on ep id == 1

Adminiuga, 2020-12-14:
if you manage to install https://github.com/Adminiuga/zha-device-handlers/tree/ac/aqara_interlock branch in your environment (needs HA 2020.12) you could try setting the interlock attribute to 1 via ZHA -> Device -> relay device -> Manage clusters -> "BinaryOutput cluster" -> pick "Interlock" attribute -> set value to 1

Kalypox, 2020-12-15:
Hi @Adminiuga,

Perfect, it works !
I also created a switch which call "zha.set_zigbee_cluster_attribute" to change the 0xFF06 cluster / interlock attribute / value 1 !

Thanks a lot !

closed 2020-12-15 by Kalypox

dj9p, 2021-03-25:
Hello, sorry I am new to home assistant and I am looking to set up this interlock function for my aqara relay. Can you please detail me the setting up of this function? Thanks in advance!

D4nnyF3nt0m, 2023-02-20:
I have enabled interlock mode (only 1 relay on at all times) by going to devices > manage zigbee devices >Binary Output Interlock > choose Attributes of the selected cluster: interlock (id: 0xff06) > enter 1 at value and enter 4447 at manufacturer code override. then click on write attribute.

D4nnyF3nt0m, 2023-02-20:
Is it possible to change the aqara relay somehow that turning of a channel is delayed when a physical switch is turned off?
In short, this should be the behaviour of the aqara switch. L1 is high > physical S1 is switched from high to low > 60 second delay > L1 is set to low.

PR #641 (2020-12-15): Add manufacturer specific "interlock" attribute for lumi.relay.c2acn01 device
Add manufacturer specific "interlock" attribute for lumi.relay.c2acn01 device.
Fixes #639