zigpy/zha-device-handlers #642: Aqara Vibration Sensor DJT11LM False Trigger[BUG]

wes1993, 2020-12-15:
**Describe the bug**
I have 10 of this sensor and sometimes have random trigger without anyone touch the sensors
**To Reproduce**
Steps to reproduce the behavior:
Pair the sensors
Check the log

**Expected behavior**
No trigger if there isn't vibration or motion

dmulcahey, 2020-12-15:
Post debug logs that show this happening without a message from the device.

wes1993, 2020-12-17:
Thanks for your reply, where can i take the logs?

wes1993, 2020-12-18:
In the log i have seen this error:
```
Logger: zigpy_znp.uart
Source: deps/lib/python3.8/site-packages/zigpy_znp/types/commands.py:419
First occurred: 17 dicembre 2020, 12:20:48 (18 occurrences)
Last logged: 8:22:15

Received an exception while passing frame to API
Traceback (most recent call last):
  File "/config/deps/lib/python3.8/site-packages/zigpy_znp/types/commands.py", line 412, in from_frame
    params[param.name], data = param.type.deserialize(data)
  File "/usr/local/lib/python3.8/site-packages/zigpy/types/struct.py", line 178, in deserialize
    value, data = field.type.deserialize(data)
  File "/usr/local/lib/python3.8/site-packages/zigpy/types/basic.py", line 53, in deserialize
    raise ValueError(f"Data is too short to contain {cls._size} bytes")
ValueError: Data is too short to contain 1 bytes

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/config/deps/lib/python3.8/site-packages/zigpy_znp/uart.py", line 77, in data_received
    self._api.frame_received(frame.payload)
  File "/config/deps/lib/python3.8/site-packages/zigpy_znp/api.py", line 334, in frame_received
    command = command_cls.from_frame(frame)
  File "/config/deps/lib/python3.8/site-packages/zigpy_znp/types/commands.py", line 419, in from_frame
    raise ValueError(
ValueError: Frame data is truncated (parsed {'Src': 0xFB6B, 'Status': <Status.NOT_SUPPORTED: 132>}), required parameter remains: Param(name='Neighbors', type=<class 'zigpy.zdo.types.Neighbors'>, description='Neighbors', optional=False)
```

wes1993, 2020-12-18:
Effetuando un'analisi specifica su un solo sensore vedo questo:
```
2020-12-18 11:32:05 DEBUG (MainThread) [zigpy.zcl] [0xf1b4:1:0x0101] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=False disable_default_response=True> manufacturer=None tsn=94 command_id=Command.Report_Attributes>
2020-12-18 11:32:05 DEBUG (MainThread) [zigpy.zcl] [0xf1b4:1:0x0101] ZCL request 0x000a: [[Attribute(attrid=1285, value=<TypeValue type=uint32_t, value=196608>)]]
2020-12-18 11:32:05 DEBUG (MainThread) [zigpy.zcl] [0xf1b4:1:0x0101] Attribute report received: 1285=196608
2020-12-18 11:32:06 DEBUG (MainThread) [zigpy.zcl] [0xf1b4:1:0x0000] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=True is_reply=False disable_default_response=True> manufacturer=4447 tsn=95 command_id=Command.Report_Attributes>
2020-12-18 11:32:06 DEBUG (MainThread) [zigpy.zcl] [0xf1b4:1:0x0000] ZCL request 0x000a: [[Attribute(attrid=65281, value=<TypeValue type=CharacterString, value=!�
                                                                                                                                                                  (!�!�>)]]
2020-12-18 11:32:06 DEBUG (MainThread) [zigpy.zcl] [0xf1b4:1:0x0000] Attribute report received: 65281=!�
                                                                                                        (!�!�
```

```
2020-12-18 11:37:06 DEBUG (MainThread) [zigpy.zcl] [0xf1b4:1:0x0101] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=False disable_default_response=True> manufacturer=None tsn=96 command_id=Command.Report_Attributes>
2020-12-18 11:37:06 DEBUG (MainThread) [zigpy.zcl] [0xf1b4:1:0x0101] ZCL request 0x000a: [[Attribute(attrid=1285, value=<TypeValue type=uint32_t, value=196608>)]]
2020-12-18 11:37:06 DEBUG (MainThread) [zigpy.zcl] [0xf1b4:1:0x0101] Attribute report received: 1285=196608
```

wes1993, 2020-12-18:
UPDATE!!
I have fixed this issue by commeting this string:
before
```
            elif attrid == RECENT_ACTIVITY_LEVEL_ATTR:
                # these seem to be sent every minute when vibration is active
                self.endpoint.device.motion_bus.listener_event(MOTION_EVENT)
```
after:
```
#            elif attrid == RECENT_ACTIVITY_LEVEL_ATTR:
#                # these seem to be sent every minute when vibration is active
#                self.endpoint.device.motion_bus.listener_event(MOTION_EVENT)
```
In this file:
/usr/local/lib/python3.8/site-packages/zhaquirks/xiaomi/aqara/vibration_aq1.py

Best Regards
Stefano

closed 2020-12-18 by wes1993