zigpy/zha-device-handlers #643: [BUG] Non-RGB Lidl Bulbs still have xy_color, hs_color and rgb_color

bfayers, 2020-12-15:
**Describe the bug**
I have three of the Lidl E27 bulbs (Manufacturer: `_TZ3000_49qchf10` Model: `TS0502A`) still present the xy,hs and rgb color attributes in Home Assistant despite not being RGB bulbs - there is also a colour wheel shown in hass.

**To Reproduce**
Steps to reproduce the behavior:
1. Pair a Non-RGB Lidl Bulb
2. Go to the entity
3. See a colour wheel which is both non-functional and redundant.

**Expected behavior**
The xy,hs and rgb color attributes should not exist and the colour wheel should not be present.

**Additional context**
Device Signature:
```
{
  "node_descriptor": "NodeDescriptor(byte1=1, byte2=64, mac_capability_flags=142, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=0)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x010c",
      "in_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0008",
        "0x0300",
        "0x1000"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "242": {
      "profile_id": 41440,
      "device_type": "0x0061",
      "in_clusters": [],
      "out_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "_TZ3000_49qchf10",
  "model": "TS0502A",
  "class": "zigpy.device.Device"
}
```

TheJulianJES, 2020-12-20:
I'll take a look at this later on. Can you check by going to "Manage Clusters" -> select Color in the first dropdown. Then select color_capabilities in the second dropdown and click "Get Zigbee attribute". Please post the output from the text field here.

bfayers, 2020-12-20:
Sure! That comes back as `None`

TheJulianJES, 2020-12-20:
> > Can you check by going to "Manage Clusters" -> select Color in the first dropdown. Then select color_capabilities in the second dropdown and click "Get Zigbee attribute".
> 
> 
> 
> Sure! That comes back as `None`
> 
> 

If you have any other Zigbee lights: do they also come back as "None"?

bfayers, 2020-12-20:
> If you have any other Zigbee lights: do they also come back as "None"?

I also have some non-RGB (so, White Spectrum only - like the Lidl ones spoken about in the original bug report) Ikea Bulbs which come back as `bitmap16.16`

TheJulianJES, 2020-12-20:
Thanks. Do you have any other signatures for CCT LIDL bulbs that shouldn't show the color picker in Home Assistant (or just the one mentioned in the first post)?

bfayers, 2020-12-20:
> Thanks. Do you have any other signatures for CCT LIDL (White-Spectrum) bulbs that shouldn't show the color picker in Home Assistant (or just the one mentioned in the first post)?

I only have three of the same model bulb from Lidl unfortunately.

TheJulianJES, 2020-12-20:
No problem, here's a quick quirk that might help fix the issue: https://github.com/TheJulianJES/zha-device-handlers/tree/lidl_cct_fix/zhaquirks/lidl

If possible (and you're comfortable with testing this), please download the repo from the link provided and then move the "lidl" folder to where all the other quirks are.
Here's a quick rundown of what to do when you're using HassOS or Supervised Home Assistant (Hass.io): https://github.com/zigpy/zha-device-handlers#testing-quirks-in-development-in-docker-based-install

bfayers, 2020-12-20:
> No problem, here's a quick quirk that might help fix the issue: [TheJulianJES/zha-device-handlers@`lidl_cct_fix`/zhaquirks/lidl](https://github.com/TheJulianJES/zha-device-handlers/tree/lidl_cct_fix/zhaquirks/lidl?rgh-link-date=2020-12-20T23%3A39%3A21Z)
> 
> If possible (and you're comfortable with testing this), please download the repo from the link provided and then move the "lidl" folder to where all the other quirks are.
> Here's a quick rundown of what to do when you're using HassOS or Supervised Home Assistant (Hass.io): [zigpy/zha-device-handlers#testing-quirks-in-development-in-docker-based-install](https://github.com/zigpy/zha-device-handlers#testing-quirks-in-development-in-docker-based-install)
> 
> When done correctly (and the quirk applies to device), please try re-pairing the device. The color wheel _should_ no longer appear in Home Assistant.

No problem! I'll try testing this tomorrow morning

bfayers, 2020-12-21:
Good news! I've just tested this and it works immediately - Didn't even need to re-pair the devices just restarted hass after putting the lidl folder in the right place!

closed 2020-12-21 by dmulcahey

mwsimmonite, 2020-12-27:
Hi @TheJulianJES I've got the E14 fitting version of Lidl's CCT bulb and it has the same issue, the bulbs show as _TZ3000_oborybow TS0502A, could this be fixed with the same quirk? Thanks for your time!

The signature of one of the bulbs is:
{
  "node_descriptor": "NodeDescriptor(byte1=1, byte2=64, mac_capability_flags=142, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=0)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x010c",
      "in_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0008",
        "0x0300",
        "0x1000"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "242": {
      "profile_id": 41440,
      "device_type": "0x0061",
      "in_clusters": [],
      "out_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "_TZ3000_oborybow",
  "model": "TS0502A",
  "class": "zigpy.device.Device"
}

TheJulianJES, 2020-12-27:
@mwsimmonite https://github.com/zigpy/zha-device-handlers/pull/672 should fix this.
Please try the modified quirk locally if possible (instructions are here: https://github.com/zigpy/zha-device-handlers/issues/643#issuecomment-748688320) and then report back if it works.

mwsimmonite, 2020-12-28:
Thanks for this! Both the E27/E14 CCT bulbs now show no colour wheel as expected