zigpy/zha-device-handlers #644: [Device Support Request] Water leak sensor TS0207

maximoei, 2020-12-16:
**Is your feature request related to a problem? Please describe.**
I have a whitelabel/noname version of the TS0207 - BlitzWolf BW-IS5 water leak sensor. It is not being recognized by ZHA. https://zigbee.blakadder.com/BlitzWolf_BW-IS5.html 

**Describe the solution you'd like**
Succesfull integration into Home Assistant, using a Sonoff Zigbee Bridge, using ZHA

**Device signature**
Lot's of debug below, I think I highlighted the relevant logrows below.
`**2020-12-14 21:22:25 INFO (MainThread) [zigpy.application] Device 0x1b0e (60:a4:23:ff:fe:11:74:4b) joined the network**
2020-12-14 21:22:25 DEBUG (MainThread) [zigpy.application] Skip initialization for existing device 60:a4:23:ff:fe:11:74:4b
2020-12-14 21:22:25 DEBUG (MainThread) [bellows.ezsp.protocol] Send command findKeyTableEntry: (60:a4:23:ff:fe:11:74:4b, True)
2020-12-14 21:22:25 DEBUG (bellows.thread_0) [bellows.uart] Sending: b'631721a9212a5ec6486ab5060e3593dcf87e'
2020-12-14 21:22:25 DEBUG (bellows.thread_0) [bellows.uart] Data frame: b'3717a1a9212aeafe497e'
2020-12-14 21:22:25 DEBUG (bellows.thread_0) [bellows.uart] Sending: b'8430fc7e'
2020-12-14 21:22:25 DEBUG (MainThread) [bellows.ezsp.protocol] Application frame 117 (findKeyTableEntry) received: b'ff'
2020-12-14 21:22:25 DEBUG (bellows.thread_0) [bellows.uart] Data frame: b'4717b5a9362a5ec6486ab5060e351c607e'
2020-12-14 21:22:25 DEBUG (bellows.thread_0) [bellows.uart] Sending: b'8520dd7e'
2020-12-14 21:22:25 DEBUG (bellows.thread_0) [bellows.uart] Data frame: b'5717b1a9112a11b259874a25aa5596499c2d6f05e3d59874f1456d92b70a2e5914ee7a0f0ffdf0477e'
2020-12-14 21:22:25 DEBUG (bellows.thread_0) [bellows.uart] Sending: b'8610be7e'
2020-12-14 21:22:25 DEBUG (MainThread) [bellows.ezsp.protocol] Application frame 98 (incomingSenderEui64Handler) received: b'4b7411feff23a460'
2020-12-14 21:22:25 DEBUG (MainThread) [bellows.zigbee.application] Received incomingSenderEui64Handler frame with [60:a4:23:ff:fe:11:74:4b]
2020-12-14 21:22:25 DEBUG (MainThread) [bellows.ezsp.protocol] Application frame 69 (incomingMessageHandler) received: b'04000013000000000400006348ae0e1bffff0c830e1b4b7411feff23a4608002'
2020-12-14 21:22:25 DEBUG (MainThread) [bellows.zigbee.application] Received incomingMessageHandler frame with [<EmberIncomingMessageType.INCOMING_BROADCAST: 4>, EmberApsFrame(profileId=0, clusterId=19, sourceEndpoint=0, destinationEndpoint=0, options=<EmberApsOption.APS_OPTION_SOURCE_EUI64: 1024>, groupId=0, sequence=99), 72, -82, 0x1b0e, 255, 255, b'\x83\x0e\x1bKt\x11\xfe\xff#\xa4`\x80']
2020-12-14 21:22:25 INFO (MainThread) [bellows.zigbee.application] ZDO Device announce: 0x1b0e, 60:a4:23:ff:fe:11:74:4b
2020-12-14 21:22:25 DEBUG (MainThread) [zigpy.zdo] [0x1b0e:zdo] ZDO request ZDOCmd.Device_annce: [0x1B0E, 60:a4:23:ff:fe:11:74:4b, 128]
2020-12-14 21:22:25 DEBUG (MainThread) [homeassistant.components.zha.core.device] [0x1B0E](TS0207): started initialization
2020-12-14 21:22:25 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x1B0E:ZDO](TS0207): 'async_initialize' stage succeeded
2020-12-14 21:22:25 DEBUG (MainThread) [zigpy.device] [0x1b0e] Extending timeout for 0x9b request
2020-12-14 21:22:25 DEBUG (MainThread) [bellows.zigbee.application] Extending timeout for 60:a4:23:ff:fe:11:74:4b/0x1b0e
2020-12-14 21:22:25 DEBUG (MainThread) [bellows.ezsp.protocol] Send command setExtendedTimeout: (60:a4:23:ff:fe:11:74:4b, True)
2020-12-14 21:22:25 DEBUG (MainThread) [zigpy.device] [0x1b0e] Extending timeout for 0x9d request
2020-12-14 21:22:25 DEBUG (MainThread) [zigpy.device] [0x1b0e] Extending timeout for 0x9f request
2020-12-14 21:22:25 DEBUG (bellows.thread_0) [bellows.uart] Sending: b'761421a92a2a5ec6486ab5060e3593f0eb7e'
2020-12-14 21:22:26 DEBUG (bellows.thread_0) [bellows.uart] Data frame: b'6014a1a92a2a0f9e7e'
2020-12-14 21:22:26 DEBUG (bellows.thread_0) [bellows.uart] Sending: b'87009f7e'
2020-12-14 21:22:26 DEBUG (MainThread) [bellows.ezsp.protocol] Application frame 126 (setExtendedTimeout) received: b''
2020-12-14 21:22:26 DEBUG (MainThread) [bellows.ezsp.protocol] Send command sendUnicast: (<EmberOutgoingMessageType.OUTGOING_DIRECT: 0>, 0x1B0E, EmberApsFrame(profileId=260, clusterId=1280, sourceEndpoint=1, destinationEndpoint=1, options=<EmberApsOption.APS_OPTION_ENABLE_ROUTE_DISCOVERY|APS_OPTION_RETRY: 320>, groupId=0, sequence=155), 156, b'\x00\x9b\x00\x02\x00\x00\x00')
2020-12-14 21:22:26 DEBUG (bellows.thread_0) [bellows.uart] Sending: b'071521a9602a15bc42904b25af5493099d4e273071c96710fdc46389fcd4d07e'
2020-12-14 21:22:26 DEBUG (bellows.thread_0) [bellows.uart] Data frame: b'7014b1a96b2a134fa6944a36aa559249984e27c8edce6779ff7e'
2020-12-14 21:22:26 DEBUG (bellows.thread_0) [bellows.uart] Sending: b'8070787e'
2020-12-14 21:22:26 DEBUG (MainThread) [bellows.ezsp.protocol] Application frame 63 (messageSentHandler) received: b'06fdff0000130000000004000063000000'
2020-12-14 21:22:26 DEBUG (MainThread) [bellows.zigbee.application] Received messageSentHandler frame with [<EmberOutgoingMessageType.OUTGOING_BROADCAST: 6>, 65533, EmberApsFrame(profileId=0, clusterId=19, sourceEndpoint=0, destinationEndpoint=0, options=<EmberApsOption.APS_OPTION_SOURCE_EUI64: 1024>, groupId=0, sequence=99), 0, <EmberStatus.SUCCESS: 0>, b'']
2020-12-14 21:22:26 DEBUG (MainThread) [bellows.zigbee.application] Unexpected message send notification tag: 0
2020-12-14 21:22:26 DEBUG (bellows.thread_0) [bellows.uart] Data frame: b'0014b1a9112a15b658944f24ab1593499c2a731ae3d59874facf4488d67e3db7e98c807e'
2020-12-14 21:22:26 DEBUG (bellows.thread_0) [bellows.uart] Sending: b'8160597e'
2020-12-14 21:22:26 DEBUG (MainThread) [bellows.ezsp.protocol] Application frame 69 (incomingMessageHandler) received: b'00040100050101400100006454b10e1bffff070927012a00021002'
2020-12-14 21:22:26 DEBUG (MainThread) [bellows.zigbee.application] Received incomingMessageHandler frame with [<EmberIncomingMessageType.INCOMING_UNICAST: 0>, EmberApsFrame(profileId=260, clusterId=1280, sourceEndpoint=1, destinationEndpoint=1, options=<EmberApsOption.APS_OPTION_ENABLE_ROUTE_DISCOVERY|APS_OPTION_RETRY: 320>, groupId=0, sequence=100), 84, -79, 0x1b0e, 255, 255, b"\t'\x01*\x00\x02\x10"]
2020-12-14 21:22:26 DEBUG (MainThread) [zigpy.zcl] [0x1b0e:1:0x0500] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=False disable_default_response=False> manufacturer=None tsn=39 command_id=1>
**2020-12-14 21:22:26 DEBUG (MainThread) [zigpy.zcl] [0x1b0e:1:0x0500] ZCL request 0x0001: [<ZoneType.Water_Sensor: 42>, 4098]
2020-12-14 21:22:26 DEBUG (MainThread) [zigpy.zcl] [0x1b0e:1:0x0500] No handler for cluster command 1**
2020-12-14 21:22:26 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x1B0E:1:0x0500]: Enroll requested
2020-12-14 21:22:26 DEBUG (MainThread) [zigpy.device] [0x1b0e] Extending timeout for 0xa1 request
2020-12-14 21:22:26 DEBUG (bellows.thread_0) [bellows.uart] Data frame: b'1115a1a9602a15b7107a7e'
2020-12-14 21:22:26 DEBUG (bellows.thread_0) [bellows.uart] Sending: b'82503a7e'
2020-12-14 21:22:26 DEBUG (MainThread) [bellows.ezsp.protocol] Application frame 52 (sendUnicast) received: b'0005'
2020-12-14 21:22:26 DEBUG (MainThread) [bellows.zigbee.application] Extending timeout for 60:a4:23:ff:fe:11:74:4b/0x1b0e
2020-12-14 21:22:26 DEBUG (MainThread) [bellows.ezsp.protocol] Send command setExtendedTimeout: (60:a4:23:ff:fe:11:74:4b, True)
2020-12-14 21:22:26 DEBUG (bellows.thread_0) [bellows.uart] Sending: b'127d3a21a92a2a5ec6486ab5060e35933c487e'
2020-12-14 21:22:26 DEBUG (bellows.thread_0) [bellows.uart] Data frame: b'221aa5a92a2aeb977e'
2020-12-14 21:22:26 DEBUG (bellows.thread_0) [bellows.uart] Sending: b'83401b7e'
2020-12-14 21:22:26 DEBUG (MainThread) [bellows.ezsp.protocol] Send command sendUnicast: (<EmberOutgoingMessageType.OUTGOING_DIRECT: 0>, 0x1B0E, EmberApsFrame(profileId=260, clusterId=0, sourceEndpoint=1, destinationEndpoint=1, options=<EmberApsOption.APS_OPTION_ENABLE_ROUTE_DISCOVERY|APS_OPTION_RETRY: 320>, groupId=0, sequence=157), 158, b'\x00\x9d\x00\x07\x00')
2020-12-14 21:22:26 DEBUG (bellows.thread_0) [bellows.uart] Data frame: b'3215b1a9112a15b658944f24ab1593499c2b4f1de3d59874f3def888fe7e3fbeebcdde6f8fcfc7d92f6c7e'
2020-12-14 21:22:26 DEBUG (bellows.thread_0) [bellows.uart] Sending: b'8430fc7e'
2020-12-14 21:22:26 DEBUG (bellows.thread_0) [bellows.uart] Data frame: b'421ab1a96b2a15bc42904b25af5493099d4e27ae71ce67a9df7e'
[... cut ...]
2020-12-14 21:22:33 DEBUG (MainThread) [zigpy.zcl] [0x1b0e:1:0x0001] ZCL request 0x000a: [[Attribute(attrid=32, value=<TypeValue type=uint8_t, value=30>), Attribute(attrid=33, value=<TypeValue type=uint8_t, value=200>)]]
2020-12-14 21:22:33 DEBUG (MainThread) [zigpy.zcl] [0x1b0e:1:0x0001] Attribute report received: battery_voltage=30, battery_percentage_remaining=200
2020-12-14 21:22:33 DEBUG (bellows.thread_0) [bellows.uart] Data frame: b'7626a1a9602a15be4afc7e'
2020-12-14 21:22:33 DEBUG (bellows.thread_0) [bellows.uart] Sending: b'8070787e'`

Adminiuga, 2020-12-16:
post device signature.
Have you tried restarting HA? There's no homeassistant.core log, so can't tell, but it is using standard IAS cluster so it should work.

maximoei, 2020-12-16:
I don't know how to get the device signature, there is no 'SimpleDescriptor' lines in the log; how do I get the signature?
No entity is added, even after fresh restart of HA. In general, no homeassistant.core log lines during scanning. 
Only other thing I see is:
```
2020-12-16 14:57:19 ERROR (MainThread) [homeassistant] Error doing job: Task exception was never retrieved
Traceback (most recent call last):
  File "/usr/local/lib/python3.8/site-packages/zigpy/device.py", line 210, in request
    result = await asyncio.wait_for(req.result, timeout)
  File "/usr/local/lib/python3.8/asyncio/tasks.py", line 498, in wait_for
    raise exceptions.TimeoutError()
asyncio.exceptions.TimeoutError
``

Adminiuga, 2020-12-16:
>2020-12-14 21:22:25 DEBUG (MainThread) [zigpy.application] Skip initialization for existing device 60:a4:23:ff:fe:11:74:4b

it save the device once, so doesn't do the second scan. To force it, you would need to remove it, wait 120s and then re-add again. But try posting zigbee info for the device 1st. When pairing, if there are any buttons on the device, try pressing those to keep the device awake

maximoei, 2020-12-16:
Thanks! Found it now.

Turns out that the device wás indeed added, but HA showed "no deviced found during scan", however, it did pop up in devices a few minutes later.

Signature for reference

```
{
  "node_descriptor": "NodeDescriptor(byte1=2, byte2=64, mac_capability_flags=128, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=0)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0402",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x0500",
        "0xef01"
      ],
      "out_clusters": [
        "0x0003",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TYZB01_sqmd19i1",
  "model": "TS0207",
  "class": "zigpy.device.Device"
}
```

Thanks a lot! I'll close.

closed 2020-12-16 by maximoei