zigpy/zha-device-handlers #648: [Device Support Request] Tuya siren zigbee 3.0

CharlieBailly, 2020-12-18:
## **Is your feature request related to a problem? Please describe.**
Hi all, i just bought this zigbee 3.0 siren : 
[Tuya Zigbee Smart Siren Alarm With temperature and Humidity Sensor Works With TUYA Smart Hub](https://a.aliexpress.com/_msosKhR)
I achieved to add it using ZHA (I have a SonOff Zigbee bridge as Coordinator)
But there is no entites in the device (no switch to toggle the alarm, no temp and humidity sensors, nothing !), here is a picture :
[image]

## **Describe the solution you'd like**
It would be great if there was switch entity for the alarm siren (so we can control alarm sound), and sensors for temperature and humidity since this device is supposed to have those.

## **Device signature - this can be acquired by removing the device from ZHA and pairing it again from the add devices screen. Be sure to add the entire content of the log panel after pairing the device to a code block below this line.**
Zigbee device signature from ZHA :
```json
{
  "node_descriptor": "NodeDescriptor(byte1=2, byte2=64, mac_capability_flags=128, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=0)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0051",
      "in_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0xef00"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZE200_d0yu2xgi",
  "model": "TS0601",
  "class": "zigpy.device.Device"
}
```

Here are the logs from ZHA while adding the device :
```txt

[0x0000:zdo] ZDO request ZDOCmd.Mgmt_Permit_Joining_req: [60, <Bool.false: 0>]
Device 0xc030 (84:2e:14:ff:fe:db:84:32) joined the network
[0xc030] Requesting 'Node Descriptor'
Tries remaining: 2
[0xc030] Extending timeout for 0x50 request
Device 0xc030 (84:2e:14:ff:fe:db:84:32) joined the network
Skip initialization for existing device 84:2e:14:ff:fe:db:84:32
Device 0xc030 (84:2e:14:ff:fe:db:84:32) joined the network
Skip initialization for existing device 84:2e:14:ff:fe:db:84:32
[0xc030:zdo] ZDO request ZDOCmd.Device_annce: [0xC030, 84:2e:14:ff:fe:db:84:32, 128]
[0xc030] Node Descriptor: NodeDescriptor(byte1=2, byte2=64, mac_capability_flags=128, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=0)
[0xc030] Discovering endpoints
Tries remaining: 3
[0xc030] Extending timeout for 0x52 request
[0xc030] Discovered endpoints: [1]
[0xc030:1] Discovering endpoint information
Tries remaining: 3
[0xc030] Extending timeout for 0x54 request
[0xc030:1] Discovered endpoint information: SizePrefixedSimpleDescriptor(endpoint=1, profile=260, device_type=81, device_version=1, input_clusters=[0, 4, 5, 61184], output_clusters=[25, 10])
Unknown cluster 61184
[0xc030] Extending timeout for 0x56 request
[0xc030:1:0x0000] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=86 command_id=Command.Read_Attributes_rsp>
[0xc030:1] Manufacturer: _TZE200_d0yu2xgi
[0xc030:1] Model: TS0601
Checking quirks for _TZE200_d0yu2xgi TS0601 (84:2e:14:ff:fe:db:84:32)
Considering <class 'bellows.zigbee.application.EZSPCoordinator'>
Fail because device_type mismatch on at least one endpoint
Considering <class 'zhaquirks.gledopto.soposhgu10.SoposhGU10'>
Fail because endpoint list mismatch: {11, 13} {1}
Considering <class 'zhaquirks.netvox.z308e3ed.Z308E3ED'>
[... 115 lines cut ...]
[0x7114:1:0x0b04] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=105 command_id=Command.Read_Attributes_rsp>
[0x8c07:1:0x0b04] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=103 command_id=Command.Read_Attributes_rsp>
[0x8c07:1:0x0b04] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=103 command_id=Command.Read_Attributes_rsp>
[0x8c07:1:0x0b04] ZCL request 0x0001: [[ReadAttributeRecord(attrid=1291, status=<Status.SUCCESS: 0>, value=<TypeValue type=int16s, value=0>)]]
[0x8c07:1:0x0b04] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=103 command_id=Command.Read_Attributes_rsp>
[0x8c07:1:0x0b04] ZCL request 0x0001: [[ReadAttributeRecord(attrid=1291, status=<Status.SUCCESS: 0>, value=<TypeValue type=int16s, value=0>)]]
[0xb9e3:1:0x0b04] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=107 command_id=Command.Read_Attributes_rsp>
[0x8C07:1:0x0b04]: async_update
[0x7114:1:0x0b04]: async_update
[0xB9E3:1:0x0b04]: async_update
[0x7114:1:0x0b04] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=111 command_id=Command.Read_Attributes_rsp>
[0x8c07:1:0x0b04] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=109 command_id=Command.Read_Attributes_rsp>
[0xb9e3:1:0x0b04] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=113 command_id=Command.Read_Attributes_rsp>
[0x8C07:1:0x0b04]: async_update
[0x7114:1:0x0b04]: async_update
[0xB9E3:1:0x0b04]: async_update
[0x7114:1:0x0b04] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=117 command_id=Command.Read_Attributes_rsp>
[0x8c07:1:0x0b04] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=115 command_id=Command.Read_Attributes_rsp>
[0xb9e3:1:0x0b04] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=119 command_id=Command.Read_Attributes_rsp>
```

## **Additional context**
I already tried to remove it and add it back in ZHA, but still nothing.
I will take a look on how to contribute, but I post this here if anyone wants to help !
Thanks !

CharlieBailly, 2020-12-19:
Ok so first I checked if there was any similar quirk already written, and it seems that #444 is a very similar product.
Moreover it has been implemented in #530, but it is not exactly the same as my device, so I will take a look, but any help is welcome !

Adminiuga, 2020-12-19:
Signature is different, but it is Tuya, so functionality wise could be the same.

CharlieBailly, 2020-12-20:
Hi thanks for your reply, but even though I am following the steps as described [here](https://github.com/zigpy/zha-device-handlers#testing-quirks-in-development-in-docker-based-install), I can't get ZHA to use a custom quirk for my device, it doesn't want to check if the quirk in `tuya/siren.py` is working for my device even though I added my device signature [here](https://github.com/zigpy/zha-device-handlers/blob/fbe1e8045f0bd471bf1d69effb413f7f67ef6a23/zhaquirks/tuya/siren.py#L150). It is also not checking for a custom quirk 

It is weird because ZHA is always 'considering' the same quirks(even for my other Zigbee devices) whatever the device signature. It is never 'considering' tuya siren class or my custom siren class if I try to add one, see here in the logs :

```txt
[0xc030:1] Manufacturer: _TZE200_d0yu2xgi
[0xc030:1] Model: TS0601
Checking quirks for _TZE200_d0yu2xgi TS0601 (84:2e:14:ff:fe:db:84:32)
Considering <class 'bellows.zigbee.application.EZSPCoordinator'>
Fail because device_type mismatch on at least one endpoint
Considering <class 'zhaquirks.gledopto.soposhgu10.SoposhGU10'>
Fail because endpoint list mismatch: {11, 13} {1}
Considering <class 'zhaquirks.netvox.z308e3ed.Z308E3ED'>
Fail because device_type mismatch on at least one endpoint
Considering <class 'zhaquirks.philips.rom001.PhilipsROM001'>
Fail because device_type mismatch on at least one endpoint
Considering <class 'zhaquirks.philips.rwl020.PhilipsRWL020'>
Fail because endpoint list mismatch: {1, 2} {1}
Considering <class 'zhaquirks.philips.rwl021.PhilipsRWL021'>
Fail because endpoint list mismatch: {1, 2} {1}
Considering <class 'zhaquirks.smartthings.multi.SmartthingsMultiPurposeSensor'>
Fail because device_type mismatch on at least one endpoint
Considering <class 'zhaquirks.smartthings.tag_v4.SmartThingsTagV4'>
Fail because device_type mismatch on at least one endpoint
Considering <class 'zhaquirks.xbee.xbee3_io.XBee3Sensor'>
Fail because endpoint list mismatch: {232, 230} {1}
Considering <class 'zhaquirks.xbee.xbee_io.XBeeSensor'>
Fail because endpoint list mismatch: {232, 230} {1}
```
The steps I am following are still valid ?
Why is ZHA always considering the same quirks even with different signature ?
Why isn't it checking for the `tuya/siren.py` quirk even though I add my device signature 
By the way I am running Home Assistant OS 2020.12.1 64 bits on RPi4 4GB  on a SSD.

Thanks !

Adminiuga, 2020-12-20:
If it is not considering it, then most likely it did not find any in the dict by manufacturer and the model. 
Make sure you are testing on 2020.12 and post the file with your changed

CharlieBailly, 2020-12-20:
Ok I understand it is possible it doesn't find it, but then why is it considering a for example `zhaquirks.philips.rom001.PhilipsROM001` quirk on each of my device ?

I am sure I am on 2020.12.1, and here are the 2 files I tested : 
[quirks.tar.gz](https://github.com/zigpy/zha-device-handlers/files/5721061/quirks.tar.gz)
In siren.py I just added my manufacturer and my model.
In siren_neo.py I also adjusted the endpoint to match mine.

But what is weird is that __pycache__ folders are not comming back when I remove them, as if my changes were not applied or something like that.
And also if I remove all the quirks except those in the tuya folder, it is still considering the defaults quirks(those I mentioned earlier).

Adminiuga, 2020-12-20:
Because those older  quirks don't have manufacturer model specified, so it tries to match it based on other criteria.

CharlieBailly, 2020-12-22:
Ok, but I am still blocked with the fact that my `__pycache__` folders are not regenerated and thus my zha is maybe not considering my updates, what do you think I am doing wrong ?

Adminiuga, 2020-12-22:
🤷 What environment and were exactly did you put the files? Have you restarted HA after the change?

CharlieBailly, 2020-12-22:
I have home assistant os on a raspberry pi 4, and I followed the steps involving the portainer addons so i put the files in the ..../zhaquirks/tuya folder, and yes i restarted after changes

Adminiuga, 2020-12-22:
Try asking in #zigbee channel on HA discord server. It is still not clear to me if this is a supervised install, HA Core or HassOS

CharlieBailly, 2020-12-22:
Ok i'll try that.
It is HassOS

Adminiuga, 2020-12-22:
on  hass os just get dev access to console and run/modify files in the container directly.

CharlieBailly, 2020-12-22:
Ok i followed this : https://developers.home-assistant.io/docs/operating-system/debugging/
I am in the container but there is no difference with the portainer method, is there ?
I can see the changes I made by using portainer console

CharlieBailly, 2021-01-05:
Ok I achieved to test my quirk, and now my device is working, see the result :
[image]
 I will try to make a pull request soon, it is just a tiny change in the `siren.py` file.

closed 2021-01-20 by dmulcahey

nachobazz, 2021-08-19:
Hi, @CharlieBailly and @Adminiuga, 

Sorry to bother you guys with this but I'm going through the setup process and the device is recognized and identified correctly, but the atributes don't update and the cluster commands don't affect the siren at all.

[image]

My setup is as follows:

- Home Assistant updated to 2021.08
- Tasmota flashed Sonoff Bridge 9.5.0
- ZHA

Also the LEDs keep blinking after adoption. Please find the logs attached below.

`
Device 0xa4fd (80:4b:50:ff:fe:04:83:1a) left the network
New device 0xa4fd (80:4b:50:ff:fe:04:83:1a) joined the network
[0xa4fd] Scheduling initialization
Tries remaining: 3
[0xa4fd] Requesting 'Node Descriptor'
Tries remaining: 2
[0xa4fd] Extending timeout for 0x1b request
Device 0xa4fd (80:4b:50:ff:fe:04:83:1a) joined the network
[0xa4fd] Scheduling initialization
[0xa4fd] Canceling old initialize call
Received frame on uninitialized device <Device model=None manuf=None nwk=0xA4FD ieee=80:4b:50:ff:fe:04:83:1a is_initialized=False> from ep 0 to ep 0, cluster 19: b'\x88\xfd\xa4\x1a\x83\x04\xfe\xffPK\x80\x80'
[0xa4fd:zdo] ZDO request ZDOCmd.Device_annce: [0xA4FD, 80:4b:50:ff:fe:04:83:1a, 128]
Tries remaining: 3
[0xa4fd] Requesting 'Node Descriptor'
Tries remaining: 2
[0xa4fd] Extending timeout for 0x1d request
Received frame on uninitialized device <Device model=None manuf=None nwk=0xA4FD ieee=80:4b:50:ff:fe:04:83:1a is_initialized=False> from ep 0 to ep 0, cluster 32770: b'\x1d\x00\xfd\xa4\x02@\x80\x02\x10RR\x00\x00,R\x00\x00'
[0xa4fd] Got Node Descriptor: NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)
[0xa4fd] Discovering endpoints
Tries remaining: 3
[0xa4fd] Extending timeout for 0x1f request
Received frame on uninitialized device <Device model=None manuf=None nwk=0xA4FD ieee=80:4b:50:ff:fe:04:83:1a is_initialized=False> from ep 0 to ep 0, cluster 32773: b'\x1f\x00\xfd\xa4\x01\x01'
[0xa4fd] Discovered endpoints: [1]
[0xa4fd] Initializing endpoints [<Endpoint id=1 in=[] out=[] status=<Status.NEW: 0>>]
[0xa4fd:1] Discovering endpoint information
Tries remaining: 3
[0xa4fd] Extending timeout for 0x21 request
Received frame on uninitialized device <Device model=None manuf=None nwk=0xA4FD ieee=80:4b:50:ff:fe:04:83:1a is_initialized=False> from ep 0 to ep 0, cluster 32772: b'!\x00\xfd\xa4\x14\x01\x04\x01Q\x00\x01\x04\x00\x00\x04\x00\x05\x00\x00\xef\x02\x19\x00\n\x00'
[0xa4fd:1] Discovered endpoint information: SizePrefixedSimpleDescriptor(endpoint=1, profile=260, device_type=81, device_version=1, input_clusters=[0, 4, 5, 61184], output_clusters=[25, 10])
Unknown cluster 61184
[0xa4fd] Extending timeout for 0x23 request
[0xa4fd:1:0x0000] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=35 command_id=Command.Read_Attributes_rsp>
[0xa4fd] Read model 'TS0601' and manufacturer '_TZE200_d0yu2xgi' from <Endpoint id=1 in=[basic:0x0000, groups:0x0004, scenes:0x0005, None:0xEF00] out=[ota:0x0019, time:0x000A] status=<Status.ZDO_INIT: 1>>
[0xa4fd] Discovered basic device information for <Device model='TS0601' manuf='_TZE200_d0yu2xgi' nwk=0xA4FD ieee=80:4b:50:ff:fe:04:83:1a is_initialized=True>
Device is initialized <Device model='TS0601' manuf='_TZE200_d0yu2xgi' nwk=0xA4FD ieee=80:4b:50:ff:fe:04:83:1a is_initialized=True>
Checking quirks for _TZE200_d0yu2xgi TS0601 (80:4b:50:ff:fe:04:83:1a)
Considering <class 'zhaquirks.tuya.siren.TuyaSiren2'>
Found custom device replacement for 80:4b:50:ff:fe:04:83:1a: <class 'zhaquirks.tuya.siren.TuyaSiren2'>
device - 0xA4FD:80:4b:50:ff:fe:04:83:1a entering async_device_initialized - is_new_join: True
device - 0xA4FD:80:4b:50:ff:fe:04:83:1a has joined the ZHA zigbee network
[0xA4FD](TS0601): started configuration
[0xA4FD:ZDO](TS0601): 'async_configure' stage succeeded
[0xA4FD:1:0x0000]: finished channel configuration
[0xA4FD:1:0x0006]: bound 'on_off' cluster: Status.SUCCESS
[0xA4FD:1:0x0006]: reporting 'on_off' attr on 'on_off' cluster: 0/900/1: Result: '[ConfigureReportingResponseRecord(status=0)]'
[0xA4FD:1:0x0402]: bound 'temperature' cluster: Status.SUCCESS
[0xA4FD:1:0x0402]: reporting 'measured_value' attr on 'temperature' cluster: 30/900/50: Result: '[ConfigureReportingResponseRecord(status=0)]'
[0xA4FD:1:0x0405]: bound 'humidity' cluster: Status.SUCCESS
[0xA4FD:1:0x0405]: reporting 'measured_value' attr on 'humidity' cluster: 30/900/100: Result: '[ConfigureReportingResponseRecord(status=0)]'
[0xA4FD:1:0x0019]: finished channel configuration
Error handling '_save_attribute' event with (80:4b:50:ff:fe:04:83:1a, 1, 0, 4, '_TZE200_d0yu2xgi') params: FOREIGN KEY constraint failed
[0xA4FD:1:0x0000]: 'async_configure' stage succeeded
[0xA4FD:1:0x0006]: 'async_configure' stage failed: 'ConfigureReportingResponseRecord' object is not subscriptable
[... cut ...]
[0xa4fd:1:0xef00] ZCL request 0x0024: [[0, 28]]
[0xa4fd:1:0xef00] No handler for cluster command 36
`

These lines looks odd (DB insert error):
`Error handling '_save_attribute' event with (80:4b:50:ff:fe:04:83:1a, 1, 0, 4, '_TZE200_d0yu2xgi') params: FOREIGN KEY constraint failed`

And there's also this unsupported cluster message
`[0xA4FD:1:0x0003]: executed 'trigger_effect' command with args: '(2, 0)' kwargs: '{}' result: [64, <Status.UNSUPPORTED_CLUSTER: 195>]`

Any ideas of what could be wrong?

PR #701 (2021-01-20): Add support for a second Tuya-based NEO Siren
Pull request to close #648