zigpy/zha-device-handlers #650: xiaomi Aqara wxkg02lm lumi.sensor_86sw2

orsgah, 2020-12-19:
Aqara wxkg02lm two buttons wireless switch
lumi.sensor_86sw2
Bottons connected successfully, but automations do not work, zha didn’t trigger buttons, but in log buttons is clicked.
I try it with two items.

Cc2531

Adminiuga, 2020-12-19:
what zigpy lib are you using?

orsgah, 2020-12-20:
> what zigpy lib are you using?

I am using last update of home assistant 2020.12 fresh installed(hass os) and znp cc2531 usb

catatonico2, 2020-12-28:
I have the same problem, have you been able to solve it?

Adminiuga, 2020-12-28:
Switch to using zigpy_znp

catatonico2, 2020-12-28:
My coordinator is a Conbee 2. I have consulted and it seems not made for this.

Adminiuga, 2020-12-29:
Well, then you problem.is completely unrelated.

Adminiuga, 2020-12-29:
You button probably just dropped off the network, cause xiaomis tend to do that

orsgah, 2020-12-30:
> You button probably just dropped off the network, cause xiaomis tend to do that

But on zigbee to mqtt work fine. And in logs i see that button work

catatonico2, 2020-12-30:
in the end using zha_event has worked for me

closed 2020-12-30 by Adminiuga

Kitser, 2021-02-08:
I have got the same problem. Using the zha_event work as a workaround. But I only have events for button 1, button 2, and both.
No separate event for double click and hold. Hopefully, this will be fixed in a later update.