zigpy/zha-device-handlers #653: [Device Support Request] Zemismart Tubular Motor ZM25TQ  (white-label of TuYa mcdj3aq) (cover)

jameshirka, 2020-12-20:
**Is your feature request related to a problem? Please describe.**
I've tried to add the Zemismart tubular motor that I bought recently and HA sees the device as a router and not a cover.

**Describe the solution you'd like**
I would like to be able to add and control the device as a cover in HA.

**Device signature - this can be acquired by removing the device from ZHA and pairing it again from the add devices screen. Be sure to add the entire content of the log panel after pairing the device to a code block below this line.**

The Zigbee device signature is as follows:

`{
  "node_descriptor": "NodeDescriptor(byte1=1, byte2=64, mac_capability_flags=142, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=0)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0051",
      "in_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0x000a",
        "0xef00"
      ],
      "out_clusters": [
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZE200_fzo2pocs",
  "model": "TS0601",
  "class": "zigpy.device.Device"
}`

Log when adding the device.
`
[0x0000:zdo] ZDO request ZDOCmd.Mgmt_Permit_Joining_req: [60, <Bool.false: 0>]
[0xab30:1:0xfc02] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=True is_reply=True disable_default_response=False> manufacturer=4673 tsn=18 command_id=Command.Report_Attributes>
[0xab30:1:0xfc02] ZCL request 0x000a: [[Attribute(attrid=16, value=<TypeValue type=bitmap8, value=bitmap8.1>), Attribute(attrid=20, value=<TypeValue type=int16s, value=27>), Attribute(attrid=18, value=<TypeValue type=int16s, value=-13>), Attribute(attrid=19, value=<TypeValue type=int16s, value=-1018>)]]
[0xab30:1:0xfc02] Attribute report received: acceleration=bitmap8.1, z_axis=27, x_axis=-13, y_axis=-1018
[0xab30:1:0xfc02] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=True is_reply=True disable_default_response=False> manufacturer=4673 tsn=19 command_id=Command.Report_Attributes>
[0xab30:1:0xfc02] ZCL request 0x000a: [[Attribute(attrid=20, value=<TypeValue type=int16s, value=37>), Attribute(attrid=18, value=<TypeValue type=int16s, value=-11>), Attribute(attrid=19, value=<TypeValue type=int16s, value=-1012>)]]
[0xab30:1:0xfc02] Attribute report received: z_axis=37, x_axis=-11, y_axis=-1012
[0xab30:1:0xfc02] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=True is_reply=True disable_default_response=False> manufacturer=4673 tsn=20 command_id=Command.Report_Attributes>
[0xab30:1:0xfc02] ZCL request 0x000a: [[Attribute(attrid=20, value=<TypeValue type=int16s, value=34>), Attribute(attrid=18, value=<TypeValue type=int16s, value=-10>), Attribute(attrid=19, value=<TypeValue type=int16s, value=-1009>)]]
[0xab30:1:0xfc02] Attribute report received: z_axis=34, x_axis=-10, y_axis=-1009
[0xab30:1:0xfc02] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=True is_reply=True disable_default_response=False> manufacturer=4673 tsn=21 command_id=Command.Report_Attributes>
[0xab30:1:0xfc02] ZCL request 0x000a: [[Attribute(attrid=20, value=<TypeValue type=int16s, value=32>), Attribute(attrid=18, value=<TypeValue type=int16s, value=-9>), Attribute(attrid=19, value=<TypeValue type=int16s, value=-1011>)]]
[0xab30:1:0xfc02] Attribute report received: z_axis=32, x_axis=-9, y_axis=-1011
[0xab30:1:0xfc02] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=True is_reply=True disable_default_response=False> manufacturer=4673 tsn=22 command_id=Command.Report_Attributes>
[0xab30:1:0xfc02] ZCL request 0x000a: [[Attribute(attrid=20, value=<TypeValue type=int16s, value=29>), Attribute(attrid=18, value=<TypeValue type=int16s, value=-11>), Attribute(attrid=19, value=<TypeValue type=int16s, value=-1009>)]]
[0xab30:1:0xfc02] Attribute report received: z_axis=29, x_axis=-11, y_axis=-1009
Device 0x7b77 (5c:02:72:ff:fe:94:32:03) joined the network
[0x7b77:zdo] ZDO request ZDOCmd.Device_annce: [0x7B77, 5c:02:72:ff:fe:94:32:03, 142]
[0x7b77] Requesting 'Node Descriptor'
Tries remaining: 2
[0x7b77] Extending timeout for 0x0d request
[0xd7eb:1:0x0402] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=76 command_id=Command.Report_Attributes>
[0xd7eb:1:0x0402] ZCL request 0x000a: [[Attribute(attrid=0, value=<TypeValue type=int16s, value=2588>)]]
[0xd7eb:1:0x0402] Attribute report received: measured_value=2588
[0xd7eb] Delivery error for seq # 0x4c, on endpoint id 1 cluster 0x0402: message send failure
[0x7b77] Node Descriptor: NodeDescriptor(byte1=1, byte2=64, mac_capability_flags=142, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=0)
[0x7b77] Discovering endpoints
Tries remaining: 3
[0x7b77] Discovered endpoints: [1]
[0x7b77:1] Discovering endpoint information
Tries remaining: 3
[0x7b77:1] Discovered endpoint information: SizePrefixedSimpleDescriptor(endpoint=1, profile=260, device_type=81, device_version=0, input_clusters=[0, 10, 4, 5, 61184], output_clusters=[25])
Unknown cluster 61184
[0xab30:1:0xfc02] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=True is_reply=True disable_default_response=False> manufacturer=4673 tsn=23 command_id=Command.Report_Attributes>
[0xab30:1:0xfc02] ZCL request 0x000a: [[Attribute(attrid=16, value=<TypeValue type=bitmap8, value=bitmap8.0>)]]
[... 53 lines cut ...]
[0xd7eb:1:0x0402] ZCL request 0x000a: [[Attribute(attrid=0, value=<TypeValue type=int16s, value=2588>)]]
[0xd7eb:1:0x0402] Attribute report received: measured_value=2588
[0xd7eb] Delivery error for seq # 0x4c, on endpoint id 1 cluster 0x0402: message send failure
[0x7b77:1:0x000a] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=False disable_default_response=False> manufacturer=None tsn=0 command_id=Command.Read_Attributes>
[0x7b77:1:0x000a] ZCL request 0x0000: [[7]]
[0x5fb7:1:0x0402] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=27 command_id=Command.Report_Attributes>
[0x5fb7:1:0x0402] ZCL request 0x000a: [[Attribute(attrid=0, value=<TypeValue type=int16s, value=2648>)]]
[0x5fb7:1:0x0402] Attribute report received: measured_value=2648
[0x7b77:1:0x0000] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=1 command_id=Command.Report_Attributes>
[0x7b77:1:0x0000] ZCL request 0x000a: [[Attribute(attrid=1, value=<TypeValue type=uint8_t, value=255>), Attribute(attrid=65506, value=<TypeValue type=uint8_t, value=21>)]]
[0x7b77:1:0x0000] Attribute report received: app_version=255, 65506=21
[0x7b77:1:0x0000] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=2 command_id=Command.Report_Attributes>
[0x7b77:1:0x0000] ZCL request 0x000a: [[Attribute(attrid=1, value=<TypeValue type=uint8_t, value=82>), Attribute(attrid=65506, value=<TypeValue type=uint8_t, value=21>)]]
[0x7b77:1:0x0000] Attribute report received: app_version=82, 65506=21
[0x7b77:1:0x0000] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=3 command_id=Command.Report_Attributes>
[0x7b77:1:0x0000] ZCL request 0x000a: [[Attribute(attrid=65503, value=<TypeValue type=CharacterString, value=��"'>)]]
[0x7b77:1:0x0000] Attribute report received: 65503=��"'
`

**Additional context**
Zigbee2Mqtt have recently added support for this:
https://www.zigbee2mqtt.io/devices/mcdj3aq.html

Github issue for it on zigbee2mqtt:
https://github.com/Koenkk/zigbee2mqtt/issues/4839

blauret, 2021-03-01:
I have one of these at home. I volunteer for testing. It is supported by zigbee2mqtt as far as I can tell. I'm trying to figure out how to port the integration.

blauret, 2021-03-03:
It's working for me.

closed 2021-09-06 by github-actions[bot]