zigpy/zha-device-handlers #654: [BUG] Sonoff ZBMINI classified as light instead of switch

malfroid, 2020-12-20:
**Describe the bug**
After adding Sonoff ZBMINI = 01MINIZB to ZHA, the entity is classified as a light.
This device is not a light, but a switch.
It does not seem possible to change the domain of an entity from light to switch. See also [my post on the HA forum](https://community.home-assistant.io/t/sonoff-zbmini-added-via-zha-is-incorrectly-classified-as-a-light-should-be-switch/257878).

**To Reproduce**
Steps to reproduce the behavior:
1. Add Sonoff ZBMINI to Home Assistant via ZHA integration
2. Look at entity ID: shows light.<name>

**Expected behavior**
Entity should be have the domain switch, ID should show switch.<name>

**Screenshots**
[image]

**Additional context**
Running Home Assistant + ZHA on Raspberry Pi 4 with Conbee II zigbee gateway.

dmulcahey, 2020-12-21:
This can be overridden in configuration.yaml in zha leveraging device config. There are examples on the forum. I don’t have any off hand.

Adminiuga, 2020-12-21:
the `device_type = 0x0100` and per Zigbee Home Automation Profile, the device with type id 0x0100 is an on/off Light.
[image]

to override, add to configuration.yaml
```
zha:
  device_config:
    00:0b:57:ff:de:ad:be:ef-1:
       type: 'switch'
```

dmulcahey, 2020-12-21:
Closing as work around supplied and the device is being treated correctly by ZHA.

closed 2020-12-21 by dmulcahey

malfroid, 2020-12-22:
I looked up the ZBMINI hardware address in the UI:
[image]

I then added the proposed configuration by @Adminiuga to my  configuration.yaml. Full configuration.yaml:

``` yaml
# Configure a default setup of Home Assistant (frontend, api, etc)
default_config:

# Text to speech
tts:
  - platform: google_translate

group: !include groups.yaml
automation: !include automations.yaml
script: !include scripts.yaml
scene: !include scenes.yaml

zha:
  zigpy_config:
    ota:
      ikea_provider: true         # Auto update Trådfri devices
      ledvance_provider: true     # Auto update LEDVANCE devices
  device_config:
    84:71:27:ff:fe:93:17:24:      # Reclassify ZBMINI (kitchen ceiling light) as switch instead of light
      type: 'switch'
    84:71:27:ff:fe:93:10:ce:      # Reclassify ZBMINI (kitchen fan) as switch instead of light
      type: 'switch'
```

After saving configuration.yaml and restarting HA, the entity is still a light: 

[image]

I tried clicking the button "Reconfigure device" in the UI, but that didn't help. Any suggestions?

dmulcahey, 2020-12-22:
You need {ieee}-{endpoint_id} not just ieee. Add `-1` to both ieee addresses before the `:`

Adminiuga, 2020-12-22:
```
00:0b:57:ff:de:ad:be:ef-1:
                       ^^^ your Endpoint ID
```

malfroid, 2020-12-22:
Thanks, by adding the endpoint it works as expected!

As I did not find anything related to this in the ZHA documentation, I made a suggestion to improve the documentation in https://github.com/home-assistant/home-assistant.io/pull/15989

ilovesmart, 2024-06-15:
Hi all. I followed the instructions for this the other day and it worked 100%.
But for some strange reason the switch it created is now greyed out and when you click into it this is the message.
This entity is no longer being provided by the zha integration. If the entity is no longer in use, delete it in settings.

The light entity that was the original entity is re-enabled and working but its a light not a switch so it turns of when I say turn off lights .

Not sure how to reverse this back to a switch.
Anyone have this issue ?

Rowserv, 2025-01-14:
Hello, I am new to home assistant, I have the same issue, trying to use sonoff zbmini as a switch, but i can't make it work, can you please help me? I've tryed to do as you posted before, but it doesn't work,