zigpy/zha-device-handlers #661: [Device Support Request] Opple 2-button remote - lumi.remote.b286opcn01 matching quirks issue

AndrewLamHK, 2020-12-22:
**Is your feature request related to a problem? Please describe.**
Hi all,
I am new on using ZHA. I have a problem to connect my Opple 2-button remote - lumi.remote.b286opcn01 to existing quirks. It seems there is mismatch on the end points. I also connected an Opple 6-button remote and it match the quirk without issue.

**Describe the solution you'd like**
It would be great if it will be able to match the existing quirk or new quirk.

**Device signature - this can be acquired by removing the device from ZHA and pairing it again from the add devices screen. Be sure to add the entire content of the log panel after pairing the device to a code block below this line.**
[image]

**Additional context**
[0x9c00:1] Manufacturer: LUMI
[0x9c00:1] Model: lumi.remote.b286opcn01 Checking quirks for LUMI lumi.remote.b286opcn01 (04:cf:8c:df:3c:75:bc:ed) Considering <class 'zhaquirks.xiaomi.aqara.opple_remote.RemoteB286OPCN01'>
Fail because endpoint list mismatch: {1, 2, 3, 4, 5, 6} {1} Considering <class 'zhaquirks.xiaomi.aqara.opple_remote.RemoteB286OPCN01Alt'>
Fail because endpoint list mismatch: {1, 2, 3, 4, 5, 6} {1} Considering <class 'zhaquirks.xiaomi.aqara.opple_remote.RemoteB286OPCN01V2'>
Fail because endpoint list mismatch: {1, 2} {1}

nelim, 2020-12-23:
There is some strange thing with this opple 2 remote.
I had same problem, i've modified V2 and removed second endpoint, but after second pairing somehow my remote entered in Alternative mode and matched RemoteB286OPCN01Alt.

On first pairing it was
[0x0938] Discovered endpoints: [1]
On other pairing it was
[0x4b79] Discovered endpoints: [1, 2, 3, 4, 5, 6]

I'm not sure how is this alt mode activated ?

dmulcahey, 2020-12-23:
The quirk activates it. We need to add another alternate to the quirk to match your original signature.

nelim, 2020-12-23:
@dmulcahey i have log from both pairings
[](https://nelim.han.bg/zha/opple.txt)
[](https://nelim.han.bg/zha/opple-fixed.txt)

I'm not sure which quirk and  how activated alt mode.. but it now works and that is my second time fixing it without understanding how :D

dmulcahey, 2020-12-23:
I’ll take care of it

TheJulianJES, 2020-12-28:
The Z2M docs mention something about the operation mode of the device: https://www.zigbee2mqtt.io/devices/WXCJKG13LM.html#binding
Perhaps this is related to "switching the modes"?

MattWestb, 2020-12-29:
Tasmota  was also getting support for the Opple switches for some weeks ago. https://github.com/arendst/Tasmota/issues/9823
They was having problem setting the operating mode with command for switching between ZLL (light commands binding) and ZHA (click clicks). 
(its looks its possible changing the mode with one switch on the backside of the switch)

Is t possible integrating the switching of the mode from manage cluster in ZHA ?

PS i dot have buying those switches then deCONZ was disabling the ZLL functionality but there is one of the best switches for integrating lights on the market to day (if not disabling ZLL).

Adminiuga, 2020-12-29:
By default we are doing "event mode" then ? https://github.com/zigpy/zha-device-handlers/blob/66c16f6719c8478b55117524a8df152bbe4dde83/zhaquirks/xiaomi/aqara/opple_remote.py#L156

But you should be able to manually write the mode attribute back to 0.

MattWestb, 2020-12-29:
Only letting the door open so its possible putting it back to ZLL command mode and also putting one comment in the quirk to setting it or not can being good.

Dont locking it to ZHA mode then its bad as deCONZ have doing.

But for my it should being more logical having ZLL as default and ZHA as alternative config for this device then its primary one light switch and not a ZHA switch

luffy934, 2021-02-13:
hello, i have exactly the same problem and i'm new. i'm on ZHA. How to fix this ?

closed 2021-02-27 by dmulcahey

Drizzt321, 2021-12-02:
@dmulcahey I'm seeing this issue today, with fully updated HomeAssistant, Home Assistant OS 6.6, core-2021.11.5, supervisor-2021.10.8 (stable). I've actually got multiple of these switches that I bought all together, and some of them get the quirk applied, and some of them don't. 

Device with the quirks applied, and works great:
```
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4447, maximum_buffer_size=127, maximum_incoming_transfer_size=100, server_mask=11264, maximum_outgoing_transfer_size=100, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0105",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x0012",
        "0xfcc0"
      ],
      "out_clusters": [
        "0x0003",
        "0x0006",
        "0x0008",
        "0x0300"
      ]
    },
    "2": {
      "profile_id": 260,
      "device_type": "0x0103",
      "in_clusters": [
        "0x0003",
        "0x0012"
      ],
      "out_clusters": [
        "0x0003",
        "0x0006"
      ]
    },
    "3": {
      "profile_id": 260,
      "device_type": "0x0103",
      "in_clusters": [
        "0x0012"
      ],
      "out_clusters": []
    },
    "4": {
      "profile_id": 260,
      "device_type": "0x0103",
      "in_clusters": [
        "0x0012"
      ],
      "out_clusters": []
    }
  },
  "manufacturer": "LUMI",
  "model": "lumi.remote.b486opcn01",
  "class": "zhaquirks.xiaomi.aqara.opple_remote.RemoteB486OPCN01V2"
}
```

Device that doesn't have the quirks applied, and so doesn't have the nice remapped commands that made working with the Quirks output so much nicer. 

```
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|MainsPowered: 132>, manufacturer_code=4447, maximum_buffer_size=127, maximum_incoming_transfer_size=100, server_mask=11264, maximum_outgoing_transfer_size=100, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=True, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0105",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0003"
      ],
      "out_clusters": [
        "0x0003",
        "0x0006",
        "0x0008",
        "0x0300"
      ]
    },
    "2": {
      "profile_id": 260,
      "device_type": "0x0103",
      "in_clusters": [
        "0x0003"
      ],
      "out_clusters": [
        "0x0003",
        "0x0006"
      ]
    },
    "3": {
      "profile_id": 260,
      "device_type": "0x0103",
      "in_clusters": [
        "0x0003"
      ],
      "out_clusters": [
        "0x0003",
        "0x0006"
      ]
    },
    "4": {
      "profile_id": 260,
      "device_type": "0x0103",
      "in_clusters": [
        "0x0003",
        "0x0012"
      ],
      "out_clusters": [
        "0x0006"
      ]
    },
    "5": {
      "profile_id": 260,
      "device_type": "0x0103",
      "in_clusters": [
        "0x0003",
        "0x0012"
      ],
      "out_clusters": [
        "0x0006"
      ]
    },
    "6": {
      "profile_id": 260,
      "device_type": "0x0103",
      "in_clusters": [
        "0x0003",
        "0x0012"
      ],
      "out_clusters": [
        "0x0006"
      ]
    }
  },
  "manufacturer": "LUMI",
  "model": "lumi.remote.b486opcn01",
  "class": "zigpy.device.Device"
}
```