zigpy/zha-device-handlers #663: [Device Support Request] Tuya zigbee remote TS0042 & TS0043 - Different manufacturer

AndrewLamHK, 2020-12-22:
**Is your feature request related to a problem? Please describe.**
Hi all,
I have a Tuya zigbee remote 2-button one TS0042 & a 3-button one TS0043. I could connected them to ZHA but they could not match the Tuya quirks.

  "manufacturer": "_TYZB02_keyjhapk",
  "model": "TS0042",

  "manufacturer": "_TYZB02_key8kk7r",
  "model": "TS0043",

I can find this two models in Tuya quirk but it seems they are not matching the Tuya quirks because of different manufacturers.

**Describe the solution you'd like**
It would be great if they can match the Tuya remote quirks.

**Device signature - this can be acquired by removing the device from ZHA and pairing it again from the add devices screen. Be sure to add the entire content of the log panel after pairing the device to a code block below this line.**
1
{
2
  "node_descriptor": "NodeDescriptor(byte1=2, byte2=64, mac_capability_flags=128, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=0)",
3
  "endpoints": {
4
    "1": {
5
      "profile_id": 260,
6
      "device_type": "0x0000",
7
      "in_clusters": [
8
        "0x0000",
9
        "0x0001",
10
        "0x0006"
11
      ],
12
      "out_clusters": [
13
        "0x0019"
14
      ]
15
    },
16
    "2": {
17
      "profile_id": 260,
18
      "device_type": "0x0000",
19
      "in_clusters": [
20
        "0x0000",
21
        "0x0001",
22
        "0x0006"
23
      ],
24
      "out_clusters": [
25
        "0x0019"
26
      ]
27
    }
28
  },
29
  "manufacturer": "_TYZB02_keyjhapk",
30
  "model": "TS0042",
31
  "class": "zigpy.device.Device"
32

1
{
2
  "node_descriptor": "NodeDescriptor(byte1=2, byte2=64, mac_capability_flags=128, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=0)",
3
  "endpoints": {
4
    "1": {
5
      "profile_id": 260,
6
      "device_type": "0x0000",
7
      "in_clusters": [
8
        "0x0000",
9
        "0x0001",
10
        "0x0006"
11
      ],
12
      "out_clusters": [
13
        "0x0019"
14
      ]
15
    },
16
    "2": {
17
      "profile_id": 260,
18
      "device_type": "0x0000",
19
      "in_clusters": [
20
        "0x0000",
21
        "0x0001",
22
        "0x0006"
23
      ],
24
      "out_clusters": [
25
        "0x0019"
26
      ]
27
    },
28
    "3": {
29
      "profile_id": 260,
30
      "device_type": "0x0000",
31
      "in_clusters": [
32
        "0x0000",
33
        "0x0001",
34
        "0x0006"
35
      ],
36
      "out_clusters": [
37
        "0x0019"
38
      ]
39
    }
40
  },
41
  "manufacturer": "_TYZB02_key8kk7r",
42
  "model": "TS0043",
43
  "class": "zigpy.device.Device"
44

**Additional context**
Add any other context or screenshots about the feature request here.

tube0013, 2020-12-22:
the clusters on these are slightly different as well. likely will need a different variant of the quirk.

can you post the signatures again use code formatting?

AndrewLamHK, 2020-12-22:
Thank you for reply. I edited my post. Are the signatures in code formatting now? I do not know much about coding and do not know how to post code correctly. Sorry about that.

DavidFFerreira, 2021-01-18:
Hi, I have this device, and I was unable to work with zha :S

`{
  "node_descriptor": "NodeDescriptor(byte1=2, byte2=64, mac_capability_flags=128, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=0)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0000",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0006"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "2": {
      "profile_id": 260,
      "device_type": "0x0000",
      "in_clusters": [
        "0x0001",
        "0x0006"
      ],
      "out_clusters": []
    }
  },
  "manufacturer": "_TZ3400_keyjhapk",
  "model": "TS0042",
  "class": "zigpy.device.Device"
}`

DavidFFerreira, 2021-01-25:
this is still a issue in home assistant new version. the _TZ3400_keyjhapk is not add in zha release. I did manage to add myself again in HA files

Adminiuga, 2021-01-25:
in the new release, not patch release.

closed 2021-08-12 by github-actions[bot]