zigpy/zha-device-handlers #670: [Device Support Request] Add Philips LMV001 as dimmable device (to support power-on behavior)

rikbosch, 2020-12-23:
**Is your feature request related to a problem? Please describe.**
Phillips bulbs support power-on behavior. The model LMV001 (an edison style dimmable Hue light) is not supported as device handler (yet). So when the bulb is turned on after a power failure, it will always toggle on.

**Describe the solution you'd like**
Add power-on off support for LMV001 device. I'm not a pyton developer, but I think it could be added to https://github.com/zigpy/zha-device-handlers/blob/b97c196c16dea43f0ea56dc822b64d487c76c15b/zhaquirks/philips/zlldimmablelight.py#L31

**Device signature - this can be acquired by removing the device from ZHA and pairing it again from the add devices screen. Be sure to add the entire content of the log panel after pairing the device to a code block below this line.**

```json
{
  "node_descriptor": "NodeDescriptor(byte1=1, byte2=64, mac_capability_flags=142, manufacturer_code=4107, maximum_buffer_size=82, maximum_incoming_transfer_size=128, server_mask=11264, maximum_outgoing_transfer_size=128, descriptor_capability_field=0)",
  "endpoints": {
    "11": {
      "profile_id": 260,
      "device_type": "0x0101",
      "in_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0008",
        "0x1000",
        "0xfc02"
      ],
      "out_clusters": [
        "0x0019"
      ]
    },
    "242": {
      "profile_id": 41440,
      "device_type": "0x0061",
      "in_clusters": [],
      "out_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "Philips",
  "model": "LWV001",
  "class": "zigpy.device.Device"
}
```

TheJulianJES, 2020-12-27:
I'll take a look at this later on. For now, you could probably use the Hue Bluetooth app to set the power on behavior. (No need to reset the bulb or disconnect it from ZHA)

TheJulianJES, 2020-12-27:
This quirk should fix the issue: https://github.com/TheJulianJES/zha-device-handlers/blob/philips_dim/zhaquirks/philips/zhadimmablelight.py

If possible, please download the repo from the link provided and then move the "zhadimmablelight.py" in the philips folder.
Here's a quick rundown of what to do when you're using HassOS or Supervised Home Assistant (Hass.io): https://github.com/zigpy/zha-device-handlers#testing-quirks-in-development-in-docker-based-install

Please then check if Home Assistant correctly applied the quirk and that you can now change the power on behavior.
If this works, PR #675 can be merged so that the next version of zha-quirks includes the fix.

rikbosch, 2020-12-27:
Wow, that was fast! @TheJulianJES.

I've tested the quirk on my HA instance and it look like it is working.

[image]

closed 2020-12-29 by dmulcahey