zigpy/zha-device-handlers #671: [Device Support Request] ZigBee IR Blaster (Split A/C Controller) AC201

bdb12, 2020-12-24:
Hi Guys

I have two of these. They pair with ZHA. The key thing I want to make work is sending the IR commands and maybe 'learn' from IR remotes as the manufacturer claims it supports most remotes. I am happy to contact the manufacture for further information. 

Very happy to test and be helpful :)

This device is already supported on Deconz but I got rid of that in favour of ZHA,

https://github.com/dresden-elektronik/deconz-rest-plugin/pull/3423

More information on the device https://www.owon-smart.com/zigbee-infrared-temperature-control-thermostat-ir-split-ac-controller-ac201-product/

Thanks in advance guys!

Device signature 

`{
  "node_descriptor": "NodeDescriptor(byte1=1, byte2=64, mac_capability_flags=142, manufacturer_code=4412, maximum_buffer_size=82, maximum_incoming_transfer_size=255, server_mask=0, maximum_outgoing_transfer_size=255, descriptor_capability_field=0)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0xc202",
      "in_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0201",
        "0x0202",
        "0xffac"
      ],
      "out_clusters": []
    }
  },
  "manufacturer": "OWON",
  "model": "AC201",
  "class": "zigpy.device.Device"
}`

pitchoun38, 2020-12-24:
I do confirm that it works also on ZiGate with Domoticz

Adminiuga, 2020-12-24:
Looks like it presents itself as a Thermostat to zigbee network. Do you get any entities for it?
Is IR learning initiated from Zigbee or from device itself? May be you initiate IR learning from the device, and then you could send thermostat commands using automations service calls

bdb12, 2020-12-24:
Hi and thanks for the replies. The only entity it shows is a Fan. It show on all the time and when you try to turn it off it turns back on. 

To the second point about learning on the device I think this is done in the their app but of course I would need their gateway to make the app work! So according to the instructions you use the app and the physical buttons on the front of the device.

Does this help?

Adminiuga, 2020-12-24:
try adding in your `configuration.yml` file

```
zha:
  device_config:
     IEEE_address_of_your_device-1:  #  replace with ieee address of your device and pay attention to -1 at the end
       type: "climate"
```

for example
```
zha:
  device_config:
    11:22:33:55:00:77:55:66-1:
       type: "climate"
```
and it is important to not miss `-1` at the end of ieee address
Restart HA after the changes

bdb12, 2020-12-24:
Sorry where do I find the ieee number?

bdb12, 2020-12-24:
Ignore that. I just found it!

bdb12, 2020-12-24:
OK that changed things. The Fan entity is grey out and I have a new one. 
[image]

bdb12, 2020-12-24:
Now I guess learning the code will be next?

Adminiuga, 2020-12-24:
yeah, but you probably need to know what manufacturer specific commands initiate IR learning. Either zigbee sniff traffic with native app or ask manufacturer

bdb12, 2020-12-24:
OK I will try. Given it works with other platforms like Deconz. Does this help?

Adminiuga, 2020-12-24:
does it do learning in deconz??? PR was saying just "initial support"?

bdb12, 2020-12-24:
Ahh. OK. Once I have these do I come back here?

bdb12, 2020-12-24:
Thanks again. So I can ask the manufacturer the correct this. What specifically should I ask please?

pipiche38, 2020-12-28:
> I do confirm that it works also on ZiGate with Domoticz

You right it works as the AC221 with the ZiGate plugin for Domoticz and would required firmware 31d

bdb12, 2021-01-06:
Hi Adminiuga

I now have the AC201 ZHA Clusters and Attributes document from the Manufacturer. I have read up on the ZHA quirks etc and seem a little beyond me. Happy to share.

closed 2021-08-12 by github-actions[bot]

hcharbonnier, 2022-10-05:
Hi bdb12,
if you still have the ZHA Clusters and Attributes document from the Manufacturer, could you please send it to me?