zigpy/zha-device-handlers #674: [Device Support Request] IKEA tradfriplug ZB3

MattWestb, 2020-12-27:
**Is your feature request related to a problem? Please describe.**

tradfriplug.py have the ZLL control outlet but not the ZB3 version.

**Describe the solution you'd like**

Adding the signature for the new clusters and GPP endpoint.

**Device signature - this can be acquired by removing the device from ZHA and pairing it again from the add devices screen. Be sure to add the entire content of the log panel after pairing the device to a code block below this line.**
```python
{
  "node_descriptor": "NodeDescriptor(byte1=1, byte2=64, mac_capability_flags=142, manufacturer_code=4476, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=0)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x010a",
      "in_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0008",
        "0x1000"
      ],
      "out_clusters": [
        "0x0005",
        "0x0019",
        "0x0020",
        "0x1000"
      ]
    },
    "242": {
      "profile_id": 41440,
      "device_type": "0x0061",
      "in_clusters": [
        "0x0021"
      ],
      "out_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "IKEA of Sweden",
  "model": "TRADFRI control outlet",
  "class": "zigpy.device.Device"
}
```
**Additional context**

I knowing its not making more functionality but its looks nicer in HA :-))
Thanks in advance !!

TheJulianJES, 2020-12-27:
To be honest, I think this should *not* be implemented. Afaik the device works perfectly fine without any quirks and it would add unnecessary code that can cause confusion later on.

MattWestb, 2020-12-27:
I was only seeing that it was one quirk for it but not loaded and was looking more and seeing it was for the ZLL ones.

I agree i cant see that its bringing any plus for the users and only more code in the system that can making trouble.

Do the old one have any useful for the user / system or should it being deleted ?
And the second thing is that users should updating the firmware and then its not being used with the ZB3 firmware.

I think the IKEA cluster (0xFC7C  # decimal = 64636) can being good defining but its one "normal cluster" so i hope its being added then normal pairing and and being used then setting up the  "IKEA scenes" if ZHA is loading it without quirk.

I letting the issue open for some days and see if some one more have some ideas of do / let or delete.

closed 2021-08-12 by github-actions[bot]