zigpy/zha-device-handlers #676: Xiaomi device with batteries reporting less than 2.8V cause negative battery % remaining

Adminiuga, 2020-12-28:
**Describe the bug**
Xiaomi device with batteries reporting less than 2.8V cause negative battery % remaining
Reported in https://github.com/home-assistant/core/issues/44578#issuecomment-751768537

**To Reproduce**
Steps to reproduce the behavior:
1. Get xiaomi to report a battery voltage less than 2.8V
2. Get negative % remaining

**Expected behavior**
Should not report negative %. Probably reporting 0% is acceptable

**Screenshots**
If applicable, add screenshots to help explain your problem.
[image]

**Additional context**
Reported in https://github.com/home-assistant/core/issues/44578#issuecomment-751768537

closed 2020-12-30 by dmulcahey

newlund, 2021-01-10:
After this fix for negative battery % my aqara sensors only report 0, 50 or 100%

Adminiuga, 2021-01-10:
Debug logs

newlund, 2021-01-10:
Sure!
What is needed?

These?

homeassistant.components.zha: debug
bellows.zigbee.application: debug
bellows.ezsp: debug
zigpy: debug
zhaquirks: debug

And send the home-assistant.log after some logging?

Adminiuga, 2021-01-10:
Yes. Send the logs for the device reporting it's battery level.

newlund, 2021-01-10:
`2021-01-10 15:58:45 DEBUG (MainThread) [bellows.zigbee.application] EZSP Counters: COUNTER_MAC_RX_BROADCAST: 304,COUNTER_MAC_TX_BROADCAST: 52,COUNTER_MAC_RX_UNICAST: 128,COUNTER_MAC_TX_UNICAST_SUCCESS: 67,COUNTER_MAC_TX_UNICAST_
RETRY: 34,COUNTER_MAC_TX_UNICAST_FAILED: 2,COUNTER_APS_DATA_RX_BROADCAST: 1,COUNTER_APS_DATA_TX_BROADCAST: 1,COUNTER_APS_DATA_RX_UNICAST: 70,COUNTER_APS_DATA_TX_UNICAST_SUCCESS: 34,COUNTER_APS_DATA_TX_UNICAST_RETRY: 5,COUNTER_AP
S_DATA_TX_UNICAST_FAILED: 2,COUNTER_ROUTE_DISCOVERY_INITIATED: 7,COUNTER_NEIGHBOR_ADDED: 12,COUNTER_NEIGHBOR_REMOVED: 0,COUNTER_NEIGHBOR_STALE: 0,COUNTER_JOIN_INDICATION: 0,COUNTER_CHILD_REMOVED: 0,COUNTER_ASH_OVERFLOW_ERROR: 0,
COUNTER_ASH_FRAMING_ERROR: 0,COUNTER_ASH_OVERRUN_ERROR: 0,COUNTER_NWK_FRAME_COUNTER_FAILURE: 0,COUNTER_APS_FRAME_COUNTER_FAILURE: 0,COUNTER_UTILITY: 0,COUNTER_APS_LINK_KEY_NOT_AUTHORIZED: 0,COUNTER_NWK_DECRYPTION_FAILURE: 0,COUN
TER_APS_DECRYPTION_FAILURE: 0,COUNTER_ALLOCATE_PACKET_BUFFER_FAILURE: 0,COUNTER_RELAYED_UNICAST: 0,COUNTER_PHY_TO_MAC_QUEUE_LIMIT_REACHED: 0,COUNTER_PACKET_VALIDATE_LIBRARY_DROPPED_COUNT: 0,COUNTER_TYPE_NWK_RETRY_OVERFLOW: 0,COU
NTER_PHY_CCA_FAIL_COUNT: 0,COUNTER_BROADCAST_TABLE_FULL: 0,COUNTER_PTA_LO_PRI_REQUESTED: 0,COUNTER_PTA_HI_PRI_REQUESTED: 0,COUNTER_PTA_LO_PRI_DENIED: 0,COUNTER_PTA_HI_PRI_DENIED: 0,COUNTER_PTA_LO_PRI_TX_ABORTED: 0,COUNTER_PTA_HI
_PRI_TX_ABORTED: 0,COUNTER_ADDRESS_CONFLICT_SENT: 0
2021-01-10 15:58:50 DEBUG (MainThread) [bellows.ezsp.protocol] Application frame 69 (incomingMessageHandler) received: b'00040100000101000100006484bdaa8bffff261c5f114b0a01ff421d0121a90b0328150421a81305212300062401000000000a21001
5641000'
2021-01-10 15:58:50 DEBUG (MainThread) [bellows.zigbee.application] Received incomingMessageHandler frame with [<EmberIncomingMessageType.INCOMING_UNICAST: 0>, EmberApsFrame(profileId=260, clusterId=0, sourceEndpoint=1, destinat
ionEndpoint=1, options=<EmberApsOption.APS_OPTION_ENABLE_ROUTE_DISCOVERY: 256>, groupId=0, sequence=100), 132, -67, 0x8baa, 255, 255, b'\x1c_\x11K\n\x01\xffB\x1d\x01!\xa9\x0b\x03(\x15\x04!\xa8\x13\x05!#\x00\x06$\x01\x00\x00\x00\
x00\n!\x00\x15d\x10\x00']
2021-01-10 15:58:50 DEBUG (MainThread) [zigpy.zcl] [0x8baa:1:0x0000] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=True is_reply=False disable_default_response=True> manu
facturer=4447 tsn=75 command_id=Command.Report_Attributes>
2021-01-10 15:58:50 DEBUG (MainThread) [zigpy.zcl] [0x8baa:1:0x0000] ZCL request 0x000a: [[Attribute(attrid=65281, value=<TypeValue type=CharacterString, value=A!�KC(UD!�SE!#>)]]
2021-01-10 15:58:50 DEBUG (MainThread) [zigpy.zcl] [0x8baa:1:0x0000] Attribute report received: 65281=A!�KC(UD!�SE!#
2021-01-10 15:58:50 DEBUG (MainThread) [zhaquirks.xiaomi] 00:15:8d:00:03:2b:89:c9 - Attribute report. attribute_id: [65281] value: [{'battery_voltage_mV': 2985, 'temperature': 21, 'X-attrib-4': 5032, 'X-attrib-5': 35, 'X-attrib-
6': 1, 'path': 5376, '0xff01-100': <Bool.false: 0>}]
2021-01-10 15:58:50 DEBUG (MainThread) [zhaquirks] LUMI lumi.sensor_magnet.aq2, Voltage [RAW]:30 [Max]:3.0 [Min]:2.8, Battery Percent: 100.0
2021-01-10 15:58:55 DEBUG (MainThread) [bellows.ezsp.protocol] Send command readCounters: ()
2021-01-10 15:58:55 DEBUG (MainThread) [bellows.ezsp.protocol] Application frame 241 (readCounters) received: b'38013500810043002200020001000100470022000500020007000c00000001000000000000000000000000000000000000000000000000000000
`

newlund, 2021-01-10:
Please let me know if you need more. Thanks!

Adminiuga, 2021-01-10:
> battery_voltage_mV': 2985
Rounded , that's 2.99v. that's 100% per thresholds I've described earlier.

newlund, 2021-01-11:
Sorry, bad example. I will wait some more to get an other voltage update.

newlund, 2021-01-11:
≤2.8 is 0%
2.9 is 50%
>3.0 is 100%

Right?

Should not 2.99 be 95% ?

And I have a sensor here reporting 2965mV. Rounded to 2.97V should be 85% ?

```
2021-01-11 07:48:30 DEBUG (MainThread) [zhaquirks.xiaomi] 00:15:8d:00:03:21:78:5b - Attribute report. attribute_id: [65281] value: [{'battery_voltage_mV': 2965, 'X-attrib-4': 17320, 'X-attrib-5': 520, 'X-attrib-6': 0, 'temperature_measurement': 2209, 'humidity_measurement': 3374, 'pressure_measurement': 100005, 'path': 9872}]
2021-01-11 07:48:30 DEBUG (MainThread) [zhaquirks] LUMI lumi.weather, Voltage [RAW]:30 [Max]:3.0 [Min]:2.8, Battery Percent: 100.0
2021-01-11 07:48:38 DEBUG (MainThread) [zhaquirks.xiaomi] 00:15:8d:00:03:4d:17:09 - Attribute report. attribute_id: [65281] value: [{'battery_voltage_mV': 2985, 'X-attrib-4': 17320, 'X-attrib-5': 38, 'X-attrib-6': 0, 'temperature_measurement': 2210, 'humidity_measurement': 3389, 'pressure_measurement': 100045, 'path': 9872}]
2021-01-11 07:48:38 DEBUG (MainThread) [zhaquirks] LUMI lumi.weather, Voltage [RAW]:30 [Max]:3.0 [Min]:2.8, Battery Percent: 100.0
2021-01-11 07:49:26 DEBUG (MainThread) [zhaquirks.xiaomi] 00:15:8d:00:03:2c:0f:dc - Attribute report. attribute_id: [65281] value: [{'battery_voltage_mV': 2995, 'temperature': 28, 'X-attrib-4': 17320, 'X-attrib-5': 38, 'X-attrib-6': 6, 'path': 21345, '0xff01-100': <Bool.false: 0>, 'illuminance_measurement': 19}]
2021-01-11 07:49:26 DEBUG (MainThread) [zhaquirks] LUMI lumi.sensor_motion.aq2, Voltage [RAW]:30 [Max]:3.0 [Min]:2.8, Battery Percent: 100.0
2021-01-11 07:50:39 DEBUG (MainThread) [zhaquirks.xiaomi] 00:15:8d:00:03:4d:17:09 - Attribute report. attribute_id: [65281] value: [{'battery_voltage_mV': 2985, 'X-attrib-4': 17320, 'X-attrib-5': 38, 'X-attrib-6': 1, 'temperature_measurement': 2215, 'humidity_measurement': 3377, 'pressure_measurement': 100043, 'path': 9872}]
2021-01-11 07:50:39 DEBUG (MainThread) [zhaquirks] LUMI lumi.weather, Voltage [RAW]:30 [Max]:3.0 [Min]:2.8, Battery Percent: 100.0
2021-01-11 07:51:48 DEBUG (MainThread) [zhaquirks.xiaomi] 00:15:8d:00:04:86:7d:56 - Attribute report. attribute_id: [65281] value: [{'battery_voltage_mV': 2945, 'X-attrib-4': 17320, 'X-attrib-5': 6, 'X-attrib-6': 851969, 'temperature_measurement': 655, 'humidity_measurement': 6865, 'pressure_measurement': 100568, 'path': 29496}]
2021-01-11 07:51:48 DEBUG (MainThread) [zhaquirks] LUMI lumi.weather, Voltage [RAW]:29 [Max]:3.0 [Min]:2.8, Battery Percent: 50.0
2021-01-11 07:57:08 DEBUG (MainThread) [zhaquirks.xiaomi] 00:15:8d:00:04:7b:3d:56 - Attribute report. attribute_id: [65281] value: [{'battery_voltage_mV': 2855, 'X-attrib-4': 17320, 'X-attrib-5': 9, 'X-attrib-6': 1, 'temperature_measurement': -1886, 'humidity_measurement': 6298, 'pressure_measurement': 100585, 'path': 21345}]
2021-01-11 07:57:08 DEBUG (MainThread) [zhaquirks] LUMI lumi.weather, Voltage [RAW]:29 [Max]:3.0 [Min]:2.8, Battery Percent: 50.0
2021-01-11 08:00:40 DEBUG (MainThread) [zhaquirks.xiaomi] 00:15:8d:00:03:21:e4:1c - Attribute report. attribute_id: [65281] value: [{'battery_voltage_mV': 2995, 'X-attrib-4': 5032, 'X-attrib-5': 36, 'X-attrib-6': 1, 'temperature_measurement': 2294, 'humidity_measurement': 3088, 'pressure_measurement': 100019, 'path': 35172}]
2021-01-11 08:00:40 DEBUG (MainThread) [zhaquirks] LUMI lumi.weather, Voltage [RAW]:30 [Max]:3.0 [Min]:2.8, Battery Percent: 100.0
```

newlund, 2021-01-11:
2845mV gives 0% now. Should be 25% if rounded to 2.85V

Perhaps you are rounding to one decimal instead of two? Maybe that's why we only get 0,50 or 100% ?

```
2021-01-11 08:16:15 DEBUG (MainThread) [zhaquirks.xiaomi] 00:15:8d:00:04:86:86:0d - Attribute report. attribute_id: [65281] value: [{'battery_voltage_mV': 2845, 'X-attrib-4': 17320, 'X-attrib-5': 7, 'X-attrib-6': 1, 'temperature_measurement': -1787, 'humidity_measurement': 4714, 'pressure_measurement': 100534, 'path': 35172}]
2021-01-11 08:16:15 DEBUG (MainThread) [zhaquirks] LUMI lumi.weather, Voltage [RAW]:28 [Max]:3.0 [Min]:2.8, Battery Percent: 0.0

```

Adminiuga, 2021-01-11:
yeah, right now it operates in 1/10th of the V and thus has only 100%, 50% and 0%. And it doesn't go negative anymore.
2.99v is a bad example, cause that gets rounded to 3.0V

newlund, 2021-01-14:
Possible to increase the resolution?

Adminiuga, 2021-01-14:
See https://github.com/zigpy/zha-device-handlers/pull/720

newlund, 2021-01-14:
Excellent! Thank you! 👍 🥇

PR #678 (2020-12-30): Refactor Xiaomi PowerConfiguration cluster
Use common code base for Xiaomi PowerConfiguration cluster to calculate battery % remaining attribute based on voltage. 
Fixes negative battery % if voltage goes below 2.8V

Fixes #676