zigpy/zha-device-handlers #677: [Device Support Request] Radiator Valve Tuya TS0601 Type _TZE200_i48qyn9s

baldisos, 2020-12-28:
My first Issue ever, i hope it's okay.
I have a TS0601 Type Radiator Valve that isn't currently supported by Tuya Quirks.

[image]

Device Signature:
`{
  "node_descriptor": "NodeDescriptor(byte1=2, byte2=64, mac_capability_flags=128, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=0)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0051",
      "in_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0x000a",
        "0xef00"
      ],
      "out_clusters": [
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZE200_i48qyn9s",
  "model": "TS0601",
  "class": "zigpy.device.Device"
}`

baldisos, 2021-01-10:
I've added the pairing log. This part looks weird:
```
[0xb7f7:1:0x0000] ZCL request 0x000a: [[Attribute(attrid=65503, value=<TypeValue type=CharacterString, value=݆�'����'����'�>)]]
[0xb7f7:1:0x0000] Attribute report received: 65503=݆�'����'����'�
```

baldisos, 2021-08-06:
This still hasn't been resolved.
Looks like [#938 ](https://github.com/zigpy/zha-device-handlers/issues/938) is closer to resolution of this issue though.

jacekk015, 2021-10-04:
@baldisos More photos of the TRV please.
Also, after pairing the TRV sends many data with the commands and attributes.
So you need to collect a little bit longer TRV data.
Best is to remove and then re-add it.
You have to enable debug level logging.

Add this to configuration.yaml
```
logger:
  default: info
  logs:
    homeassistant.components.zha: debug
    zigpy: debug
    zhaquirks: debug
```

Look for something like that below. Commands and attributes let's someone to compare it with other TRV, and sometimes there's just need to add an ID of your TRV to the code, if it's already supported.
```
2021-09-24 15:01:03 DEBUG (MainThread) [zigpy.zcl] [0xddeb:1:0xef00] ZCL request 0x0002: [Command(status=0, tsn=16, command_id=528, function=0, data=[4, 0, 0, 0, 34])]
2021-09-24 15:01:03 DEBUG (MainThread) [zhaquirks.tuya] [0xddeb:1:0xef00] Received value [0, 0, 0, 34] for attribute 0x0210 (command 0x0002)
2021-09-24 15:01:13 DEBUG (MainThread) [zigpy.zcl] [0xddeb:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=8 command_id=2>
2021-09-24 15:01:13 DEBUG (MainThread) [zigpy.zcl] [0xddeb:1:0xef00] ZCL request 0x0002: [Command(status=0, tsn=24, command_id=536, function=0, data=[4, 0, 0, 0, 233])]
2021-09-24 15:01:13 DEBUG (MainThread) [zhaquirks.tuya] [0xddeb:1:0xef00] Received value [0, 0, 0, 233] for attribute 0x0218 (command 0x0002)
2021-09-24 15:01:14 DEBUG (MainThread) [zigpy.zcl] [0xddeb:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=9 command_id=2>
2021-09-24 15:01:14 DEBUG (MainThread) [zigpy.zcl] [0xddeb:1:0xef00] ZCL request 0x0002: [Command(status=0, tsn=30, command_id=286, function=0, data=[1, 0])]
2021-09-24 15:01:14 DEBUG (MainThread) [zhaquirks.tuya] [0xddeb:1:0xef00] Received value [0] for attribute 0x011e (command 0x0002)
2021-09-24 15:01:14 DEBUG (MainThread) [zigpy.zcl] [0xddeb:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=10 command_id=2>
2021-09-24 15:01:14 DEBUG (MainThread) [zigpy.zcl] [0xddeb:1:0xef00] ZCL request 0x0002: [Command(status=0, tsn=34, command_id=546, function=0, data=[4, 0, 0, 0, 131])]
2021-09-24 15:01:14 DEBUG (MainThread) [zhaquirks.tuya] [0xddeb:1:0xef00] Received value [0, 0, 0, 131] for attribute 0x0222 (command 0x0002)
2021-09-24 15:01:25 DEBUG (MainThread) [zigpy.zcl] [0xddeb:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=11 command_id=2>
2021-09-24 15:01:25 DEBUG (MainThread) [zigpy.zcl] [0xddeb:1:0xef00] ZCL request 0x0002: [Command(status=0, tsn=101, command_id=613, function=0, data=[4, 0, 0, 0, 42])]
2021-09-24 15:01:25 DEBUG (MainThread) [zhaquirks.tuya] [0xddeb:1:0xef00] Received value [0, 0, 0, 42] for attribute 0x0265 (command 0x0002)
2021-09-24 15:01:35 DEBUG (MainThread) [zigpy.zcl] [0xddeb:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=12 command_id=2>
2021-09-24 15:01:35 DEBUG (MainThread) [zigpy.zcl] [0xddeb:1:0xef00] ZCL request 0x0002: [Command(status=0, tsn=102, command_id=614, function=0, data=[4, 0, 0, 0, 34])]
2021-09-24 15:01:35 DEBUG (MainThread) [zhaquirks.tuya] [0xddeb:1:0xef00] Received value [0, 0, 0, 34] for attribute 0x0266 (command 0x0002)
```

baldisos, 2021-10-04:
@jacekk015 

This is the output i get while re-pairing the device:

```
2021-10-04 17:44:30 INFO (MainThread) [zigpy.application] Device 0x6591 (ec:1b:bd:ff:fe:70:b3:9f) left the network
2021-10-04 17:44:35 DEBUG (bellows.thread_0) [bellows.uart] Data frame: b'27e4b1ed542e14ba59954b65ab5592806ff4b50712316083fdcc6389dc3e77e77e'
2021-10-04 17:44:35 DEBUG (bellows.thread_0) [bellows.uart] Sending: b'83401b7e'
2021-10-04 17:44:35 DEBUG (MainThread) [bellows.ezsp.protocol] Application frame 69 (incomingMessageHandler) received: b'0004010800010140010000c9f3ba92acffff0708000a00002040'
2021-10-04 17:44:35 DEBUG (MainThread) [bellows.zigbee.application] Received incomingMessageHandler frame with [<EmberIncomingMessageType.INCOMING_UNICAST: 0>, EmberApsFrame(profileId=260, clusterId=8, sourceEndpoint=1, destinationEndpoint=1, options=<EmberApsOption.APS_OPTION_ENABLE_ROUTE_DISCOVERY|APS_OPTION_RETRY: 320>, groupId=0, sequence=201), 243, -70, 0xac92, 255, 255, b'\x08\x00\n\x00\x00 @']
2021-10-04 17:44:35 DEBUG (MainThread) [zigpy.zcl] [0xac92:1:0x0008] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=0 command_id=Command.Report_Attributes>
2021-10-04 17:44:35 DEBUG (MainThread) [zigpy.zcl] [0xac92:1:0x0008] ZCL request 0x000a: [[Attribute(attrid=0, value=<TypeValue type=uint8_t, value=64>)]]
2021-10-04 17:44:35 DEBUG (MainThread) [zigpy.zcl] [0xac92:1:0x0008] Attribute report received: current_level=64
2021-10-04 17:44:35 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0xAC92:1:0x0008]: received attribute: 0 update with value: 64
2021-10-04 17:44:35 DEBUG (MainThread) [bellows.ezsp.protocol] Send command sendUnicast: (<EmberOutgoingMessageType.OUTGOING_DIRECT: 0>, 0xAC92, EmberApsFrame(profileId=260, clusterId=8, sourceEndpoint=1, destinationEndpoint=1, options=<EmberApsOption.APS_OPTION_ENABLE_ROUTE_DISCOVERY|APS_OPTION_RETRY: 320>, groupId=0, sequence=0), 35, b'\x18\x00\x0b\n\x00')
2021-10-04 17:44:35 DEBUG (bellows.thread_0) [bellows.uart] Sending: b'73e5219c54b8b9b6589c4a24ab1593499c4e04aef5ce6c81fdd5ad7e'
2021-10-04 17:44:35 DEBUG (bellows.thread_0) [bellows.uart] Data frame: b'37e4b1ed542e14b459954b65ab5592836af6b50712316083fccc6389ec7f47c07e'
2021-10-04 17:44:35 DEBUG (bellows.thread_0) [bellows.uart] Sending: b'8430fc7e'
2021-10-04 17:44:35 DEBUG (MainThread) [bellows.ezsp.protocol] Application frame 69 (incomingMessageHandler) received: b'0004010600010140010000caf6b892acffff0708010a00001001'
2021-10-04 17:44:35 DEBUG (MainThread) [bellows.zigbee.application] Received incomingMessageHandler frame with [<EmberIncomingMessageType.INCOMING_UNICAST: 0>, EmberApsFrame(profileId=260, clusterId=6, sourceEndpoint=1, destinationEndpoint=1, options=<EmberApsOption.APS_OPTION_ENABLE_ROUTE_DISCOVERY|APS_OPTION_RETRY: 320>, groupId=0, sequence=202), 246, -72, 0xac92, 255, 255, b'\x08\x01\n\x00\x00\x10\x01']
2021-10-04 17:44:35 DEBUG (MainThread) [zigpy.zcl] [0xac92:1:0x0006] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=1 command_id=Command.Report_Attributes>
2021-10-04 17:44:35 DEBUG (MainThread) [zigpy.zcl] [0xac92:1:0x0006] ZCL request 0x000a: [[Attribute(attrid=0, value=<TypeValue type=Bool, value=Bool.true>)]]
2021-10-04 17:44:35 DEBUG (MainThread) [zigpy.zcl] [0xac92:1:0x0006] Attribute report received: on_off=1
2021-10-04 17:44:35 DEBUG (bellows.thread_0) [bellows.uart] Data frame: b'40e5a19c54b5d8c47e'
2021-10-04 17:44:35 DEBUG (bellows.thread_0) [bellows.uart] Sending: b'8520dd7e'
2021-10-04 17:44:35 DEBUG (bellows.thread_0) [bellows.uart] Data frame: b'50e5b19754b8b9b6589c4a24ab1593499cd104abedbae57e'
2021-10-04 17:44:35 DEBUG (bellows.thread_0) [bellows.uart] Sending: b'8610be7e'
2021-10-04 17:44:35 DEBUG (MainThread) [bellows.ezsp.protocol] Application frame 52 (sendUnicast) received: b'009f'
2021-10-04 17:44:35 DEBUG (MainThread) [bellows.ezsp.protocol] Application frame 63 (messageSentHandler) received: b'0092ac040108000101400100009f230000'
2021-10-04 17:44:35 DEBUG (MainThread) [bellows.zigbee.application] Received messageSentHandler frame with [<EmberOutgoingMessageType.OUTGOING_DIRECT: 0>, 44178, EmberApsFrame(profileId=260, clusterId=8, sourceEndpoint=1, destinationEndpoint=1, options=<EmberApsOption.APS_OPTION_ENABLE_ROUTE_DISCOVERY|APS_OPTION_RETRY: 320>, groupId=0, sequence=159), 35, <EmberStatus.SUCCESS: 0>, b'']
2021-10-04 17:44:35 DEBUG (MainThread) [bellows.ezsp.protocol] Send command sendUnicast: (<EmberOutgoingMessageType.OUTGOING_DIRECT: 0>, 0xAC92, EmberApsFrame(profileId=260, clusterId=6, sourceEndpoint=1, destinationEndpoint=1, options=<EmberApsOption.APS_OPTION_ENABLE_ROUTE_DISCOVERY|APS_OPTION_RETRY: 320>, groupId=0, sequence=1), 36, b'\x18\x01\x0b\n\x00')
2021-10-04 17:44:35 DEBUG (bellows.thread_0) [bellows.uart] Sending: b'06ea219c54b8b9b658924a24ab1593499c4f03aef5cf6c81fd01997e'
2021-10-04 17:44:35 DEBUG (bellows.thread_0) [bellows.uart] Data frame: b'61eaa19c548a0a297e'
2021-10-04 17:44:35 DEBUG (bellows.thread_0) [bellows.uart] Sending: b'87009f7e'
2021-10-04 17:44:35 DEBUG (MainThread) [bellows.ezsp.protocol] Application frame 52 (sendUnicast) received: b'00a0'
2021-10-04 17:44:35 DEBUG (bellows.thread_0) [bellows.uart] Data frame: b'71eab19754b8b9b658924a24ab1593499cee03abed3a097e'
[... cut ...]
2021-10-04 17:44:57 DEBUG (MainThread) [zigpy.zcl] [0x0555:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=29 command_id=2>
2021-10-04 17:44:57 WARNING (MainThread) [zigpy.zcl] [0x0555:1:0xef00] Unknown cluster-specific command 2
2021-10-04 17:44:57 DEBUG (MainThread) [zigpy.zcl] [0x0555:1:0xef00] ZCL request 0x0002: b'\x00\x10\x10\x02\x00\x04\x00\x00\x00\x01'
2021-10-04 17:44:57 DEBUG (MainThread) [zigpy.zcl] [0x0555:1:0xef00] No handler for cluster command 2
```
[image]
[image]
[image]

jacekk015, 2021-10-04:
Do something more.
Add to configuration.yaml
```
# Zigbee
zha:
  custom_quirks_path: /config/zha_custom_quirks/
```
then, before restart, create that folder "zha_custom_quirks" in the config directory.
This will allow you to test custom settings for the valve.
Download valve.py file and put it in that folder.
https://github.com/zigpy/zha-device-handlers/blob/master/zhaquirks/tuya/valve.py

On the very bottom try to add your valve to the MoesHY368_Type1 class, in the MODELS_INFO area.
[image]
You need to add there your symbol - I've added it below:
("_TZE200_i48qyn9s", "TS0601")

After that restart HomeAssistant. ZHA will find your ID there and use it as a quirk.
HomeAssistant will try to talk with the valve with that code.

Remove TRV and repair it again. And please wait a little bit longer for data. When you pairing, after synchronization, the TRV sends all settings. After you add that valve.py file you should see even more data from the TRV. There are no attributes pointed in the log above.

After all of that if you are lucky you should see temperatures and battery level.
But, from the pictures you've sent, I assume that you have MoesAlt clone - the same I have[same screen], and I write a python code.
Share your logs here, from the attributes I should see if that's true - maybe you will be possible to steer it with my custom quirk test file.

jacekk015, 2021-10-04:
My valve looks like this

[image]

baldisos, 2021-10-04:
Doesn't work out of the box after adding the custom quirks.

This ist the output after adding the custom valve.py:

```
2021-10-04 18:30:07 INFO (MainThread) [zigpy.application] Device 0x0555 (ec:1b:bd:ff:fe:70:b3:9f) left the network
2021-10-04 18:30:12 DEBUG (bellows.thread_0) [bellows.uart] Data frame: b'6401b18b502bf7dfc6273adb55e889a598ba047e'
2021-10-04 18:30:12 DEBUG (bellows.thread_0) [bellows.uart] Sending: b'87009f7e'
2021-10-04 18:30:12 DEBUG (MainThread) [bellows.ezsp.protocol] Application frame 35 (childJoinHandler) received: b'0401e26d9fb370feffbd1bec04'
2021-10-04 18:30:12 DEBUG (MainThread) [bellows.zigbee.application] Received childJoinHandler frame with [4, <Bool.true: 1>, 0x6de2, ec:1b:bd:ff:fe:70:b3:9f, <EmberNodeType.SLEEPY_END_DEVICE: 4>]
2021-10-04 18:30:12 DEBUG (bellows.thread_0) [bellows.uart] Data frame: b'7401b18cb6478a01296ab598b1b993499c4e69637e'
2021-10-04 18:30:12 DEBUG (bellows.thread_0) [bellows.uart] Sending: b'8070787e'
2021-10-04 18:30:12 DEBUG (MainThread) [bellows.ezsp.protocol] Application frame 36 (trustCenterJoinHandler) received: b'e26d9fb370feffbd1bec01000000'
2021-10-04 18:30:12 DEBUG (MainThread) [bellows.zigbee.application] Received trustCenterJoinHandler frame with [0x6de2, ec:1b:bd:ff:fe:70:b3:9f, <EmberDeviceUpdate.STANDARD_SECURITY_UNSECURED_JOIN: 1>, <EmberJoinDecision.USE_PRECONFIGURED_KEY: 0>, 0x0000]
2021-10-04 18:30:12 INFO (MainThread) [zigpy.application] Device 0x6de2 (ec:1b:bd:ff:fe:70:b3:9f) joined the network
2021-10-04 18:30:12 DEBUG (MainThread) [zigpy.application] Device ec:1b:bd:ff:fe:70:b3:9f changed id (0x6de2 => 0x6de2)
2021-10-04 18:30:12 DEBUG (MainThread) [zigpy.device] [0x6de2] Skipping initialization, device is fully initialized
2021-10-04 18:30:12 DEBUG (MainThread) [zigpy.application] Device is initialized <Device model='TS0601' manuf='_TZE200_i48qyn9s' nwk=0x6DE2 ieee=ec:1b:bd:ff:fe:70:b3:9f is_initialized=True>
2021-10-04 18:30:12 DEBUG (MainThread) [zigpy.quirks.registry] Checking quirks for _TZE200_i48qyn9s TS0601 (ec:1b:bd:ff:fe:70:b3:9f)
2021-10-04 18:30:12 DEBUG (MainThread) [zigpy.quirks.registry] Considering <class 'valve.MoesHY368_Type1'>
2021-10-04 18:30:12 DEBUG (MainThread) [zigpy.quirks.registry] Fail because input cluster mismatch on at least one endpoint
2021-10-04 18:30:12 DEBUG (MainThread) [zigpy.quirks.registry] Considering <class 'zhaquirks.xbee.xbee_io.XBeeSensor'>
2021-10-04 18:30:12 DEBUG (MainThread) [zigpy.quirks.registry] Fail because endpoint list mismatch: {232, 230} {1}
2021-10-04 18:30:12 DEBUG (MainThread) [zigpy.quirks.registry] Considering <class 'zhaquirks.xbee.xbee3_io.XBee3Sensor'>
2021-10-04 18:30:12 DEBUG (MainThread) [zigpy.quirks.registry] Considering <class 'zhaquirks.smartthings.tag_v4.SmartThingsTagV4'>
2021-10-04 18:30:12 DEBUG (MainThread) [zigpy.quirks.registry] Fail because device_type mismatch on at least one endpoint
2021-10-04 18:30:12 DEBUG (MainThread) [zigpy.quirks.registry] Considering <class 'zhaquirks.smartthings.multi.SmartthingsMultiPurposeSensor'>
2021-10-04 18:30:12 DEBUG (MainThread) [zigpy.quirks.registry] Considering <class 'zhaquirks.netvox.z308e3ed.Z308E3ED'>
2021-10-04 18:30:12 DEBUG (MainThread) [zigpy.quirks.registry] Considering <class 'zhaquirks.gledopto.soposhgu10.SoposhGU10'>
2021-10-04 18:30:12 DEBUG (MainThread) [zigpy.quirks.registry] Fail because endpoint list mismatch: {11, 13} {1}
2021-10-04 18:30:12 DEBUG (MainThread) [zigpy.quirks.registry] Considering <class 'bellows.zigbee.application.EZSPCoordinator'>
2021-10-04 18:30:12 DEBUG (MainThread) [bellows.ezsp.protocol] Send command findKeyTableEntry: (ec:1b:bd:ff:fe:70:b3:9f, True)
2021-10-04 18:30:12 DEBUG (MainThread) [homeassistant.components.zha.core.gateway] device - 0x6de2:ec:1b:bd:ff:fe:70:b3:9f entering async_device_initialized - is_new_join: False
2021-10-04 18:30:12 DEBUG (MainThread) [homeassistant.components.zha.core.gateway] device - 0x6de2:ec:1b:bd:ff:fe:70:b3:9f has been reset and re-added or its nwk address changed
2021-10-04 18:30:12 DEBUG (MainThread) [homeassistant.components.zha.core.gateway] skipping discovery for previously discovered device - 0x6de2:ec:1b:bd:ff:fe:70:b3:9f
2021-10-04 18:30:12 DEBUG (MainThread) [homeassistant.components.zha.core.device] [0x6de2](TS0601): started configuration
2021-10-04 18:30:12 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x6de2:ZDO](TS0601): 'async_configure' stage succeeded
2021-10-04 18:30:12 DEBUG (bellows.thread_0) [bellows.uart] Sending: b'400621ddcb99654ca62951c9abb59b7e'
2021-10-04 18:30:12 DEBUG (bellows.thread_0) [bellows.uart] Data frame: b'0401b5cacb99654ca62951c999f17e'
2021-10-04 18:30:12 DEBUG (bellows.thread_0) [bellows.uart] Sending: b'8160597e'
2021-10-04 18:30:12 DEBUG (bellows.thread_0) [bellows.uart] Data frame: b'1401b1ed502a15a159944a25ae559291638dc5c612316b0e1fabfc3a8c80c01af0215e06377e'
2021-10-04 18:30:12 DEBUG (bellows.thread_0) [bellows.uart] Sending: b'82503a7e'
2021-10-04 18:30:12 DEBUG (bellows.thread_0) [bellows.uart] Data frame: b'2506a1ddabf0207e'
2021-10-04 18:30:12 DEBUG (bellows.thread_0) [bellows.uart] Sending: b'83401b7e'
2021-10-04 18:30:12 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x6de2:1:0x0000]: finished channel configuration
2021-10-04 18:30:12 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x6de2:1:0x0019]: finished channel configuration
[... cut ...]
2021-10-04 18:31:17 DEBUG (MainThread) [zigpy.zcl] [0x6de2:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=121 command_id=2>
2021-10-04 18:31:17 WARNING (MainThread) [zigpy.zcl] [0x6de2:1:0xef00] Unknown cluster-specific command 2
2021-10-04 18:31:17 DEBUG (MainThread) [zigpy.zcl] [0x6de2:1:0xef00] ZCL request 0x0002: b'\x00ff\x02\x00\x04\x00\x00\x00"'
2021-10-04 18:31:17 DEBUG (MainThread) [zigpy.zcl] [0x6de2:1:0xef00] No handler for cluster command 2
```

MattWestb, 2021-10-04:
Your device have the time cluster on in and the "normal" have it on out side of the cluster.
You can trying changing the cluster in the lines after there you have doing your device so its looks like this:
```
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.SMART_PLUG,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    Time.cluster_id,
                    TuyaManufClusterAttributes.cluster_id,
                ],
                OUTPUT_CLUSTERS: [Ota.cluster_id],
            }
        },
    }

    replacement = {
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.THERMOSTAT,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    Time.cluster_id,
                    MoesManufCluster,
                    MoesThermostat,
                    MoesUserInterface,
                    MoesWindowDetection,
                    TuyaPowerConfigurationCluster,
                ],
                OUTPUT_CLUSTERS: [Ota.cluster_id],
            }
```
Its only the lines with time that is changed.

baldisos, 2021-10-04:
Okay, so this actually worked in terms of pairing the device.

But i can't control it, nor does it show the battery status.

[image]
[image]

This is the output when trying to change anything in Lovelace:

```
2021-10-04 19:06:24 ERROR (MainThread) [homeassistant.components.websocket_api.http.connection] [281472758758080] 'Status' object is not subscriptable
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/websocket_api/commands.py", line 185, in handle_call_service
    await hass.services.async_call(
  File "/usr/src/homeassistant/homeassistant/core.py", line 1492, in async_call
    task.result()
  File "/usr/src/homeassistant/homeassistant/core.py", line 1527, in _execute_service
    await handler.job.target(service_call)
  File "/usr/src/homeassistant/homeassistant/helpers/entity_component.py", line 213, in handle_service
    await self.hass.helpers.service.entity_service_call(
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 658, in entity_service_call
    future.result()  # pop exception if have
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 843, in async_request_call
    await coro
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 695, in _handle_entity_call
    await result
  File "/usr/src/homeassistant/homeassistant/components/zha/switch.py", line 62, in async_turn_on
    result = await self._on_off_channel.on()
  File "/usr/src/homeassistant/homeassistant/components/zha/core/channels/base.py", line 56, in wrapper
    result = await command(*args, **kwds)
  File "/config/zha_custom_quirks/valve.py", line 750, in command
    return [command_id, res[0].status]
TypeError: 'Status' object is not subscriptable
```

This was the pairing output after your suggested changes @MattWestb :

```
2021-10-04 19:03:45 DEBUG (MainThread) [bellows.ezsp.protocol] Send command sendBroadcast: (<BroadcastAddress.ALL_ROUTERS_AND_COORDINATOR: 65532>, EmberApsFrame(profileId=0, clusterId=54, sourceEndpoint=0, destinationEndpoint=0, options=<EmberApsOption.APS_OPTION_NONE: 0>, groupId=0, sequence=96), 0, 97, b'`<\x00')
2021-10-04 19:03:45 DEBUG (bellows.thread_0) [bellows.uart] Sending: b'23c8219ea8d515b26f944a25aa559249fc4e46a88df26786487e'
2021-10-04 19:03:45 DEBUG (bellows.thread_0) [bellows.uart] Data frame: b'33c8a19e54bc853f7e'
2021-10-04 19:03:45 DEBUG (bellows.thread_0) [bellows.uart] Sending: b'8430fc7e'
2021-10-04 19:03:45 DEBUG (bellows.thread_0) [bellows.uart] Data frame: b'43c8b1ed512a158459944a25aa5592df634e27ab123164ebc1c6da757e'
2021-10-04 19:03:45 DEBUG (bellows.thread_0) [bellows.uart] Sending: b'8520dd7e'
2021-10-04 19:03:45 DEBUG (MainThread) [bellows.ezsp.protocol] Application frame 54 (sendBroadcast) received: b'0096'
2021-10-04 19:03:45 DEBUG (MainThread) [bellows.ezsp.protocol] Application frame 69 (incomingMessageHandler) received: b'050000360000000000000096ff000000ffff03603c00'
2021-10-04 19:03:45 DEBUG (MainThread) [bellows.zigbee.application] Received incomingMessageHandler frame with [<EmberIncomingMessageType.INCOMING_BROADCAST_LOOPBACK: 5>, EmberApsFrame(profileId=0, clusterId=54, sourceEndpoint=0, destinationEndpoint=0, options=<EmberApsOption.APS_OPTION_NONE: 0>, groupId=0, sequence=150), 255, 0, 0x0000, 255, 255, b'`<\x00']
2021-10-04 19:03:45 DEBUG (MainThread) [zigpy.zdo] [0x0000:zdo] ZDO request ZDOCmd.Mgmt_Permit_Joining_req: [60, <Bool.false: 0>]
2021-10-04 19:03:46 DEBUG (bellows.thread_0) [bellows.uart] Data frame: b'53c8b19752d6eab259a24a25aa5592499cd846abed199b7e'
2021-10-04 19:03:46 DEBUG (bellows.thread_0) [bellows.uart] Sending: b'8610be7e'
2021-10-04 19:03:46 DEBUG (MainThread) [bellows.ezsp.protocol] Application frame 63 (messageSentHandler) received: b'06fcff0000360000000000000096610000'
2021-10-04 19:03:46 DEBUG (MainThread) [bellows.zigbee.application] Received messageSentHandler frame with [<EmberOutgoingMessageType.OUTGOING_BROADCAST: 6>, 65532, EmberApsFrame(profileId=0, clusterId=54, sourceEndpoint=0, destinationEndpoint=0, options=<EmberApsOption.APS_OPTION_NONE: 0>, groupId=0, sequence=150), 97, <EmberStatus.SUCCESS: 0>, b'']
2021-10-04 19:03:46 DEBUG (MainThread) [bellows.ezsp.protocol] Send command permitJoining: (60,)
2021-10-04 19:03:46 DEBUG (bellows.thread_0) [bellows.uart] Sending: b'36c9218a6838bf7e'
2021-10-04 19:03:46 DEBUG (bellows.thread_0) [bellows.uart] Data frame: b'64c9a18a54a58b7e'
2021-10-04 19:03:46 DEBUG (bellows.thread_0) [bellows.uart] Sending: b'87009f7e'
2021-10-04 19:03:46 DEBUG (MainThread) [bellows.ezsp.protocol] Application frame 34 (permitJoining) received: b'00'
2021-10-04 19:03:51 DEBUG (bellows.thread_0) [bellows.uart] Data frame: b'74c9b58b502af7dfc6273adb55e889a598a91e7e'
2021-10-04 19:03:51 DEBUG (bellows.thread_0) [bellows.uart] Sending: b'8070787e'
2021-10-04 19:03:51 DEBUG (bellows.thread_0) [bellows.uart] Data frame: b'04c9b18cb6478a01296ab598b1b9904a63b145437e'
2021-10-04 19:03:51 DEBUG (bellows.thread_0) [bellows.uart] Sending: b'8160597e'
2021-10-04 19:03:51 DEBUG (MainThread) [bellows.ezsp.protocol] Application frame 35 (childJoinHandler) received: b'0400e26d9fb370feffbd1bec04'
2021-10-04 19:03:51 DEBUG (MainThread) [bellows.zigbee.application] Received childJoinHandler frame with [4, <Bool.false: 0>, 0x6de2, ec:1b:bd:ff:fe:70:b3:9f, <EmberNodeType.SLEEPY_END_DEVICE: 4>]
[... cut ...]
2021-10-04 19:04:27 DEBUG (MainThread) [zigpy.zcl] [0x183f:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=77 command_id=17>
2021-10-04 19:04:27 WARNING (MainThread) [zigpy.zcl] [0x183f:1:0xef00] Unknown cluster-specific command 17
2021-10-04 19:04:27 DEBUG (MainThread) [zigpy.zcl] [0x183f:1:0xef00] ZCL request 0x0011: b'\n\xe2\x04'
2021-10-04 19:04:27 DEBUG (MainThread) [zigpy.zcl] [0x183f:1:0xef00] No handler for cluster command 17
```

MattWestb, 2021-10-04:
Great the quirk is loaded and need adding one new type with the new cluster config.
I dont knowing the tuya DP but some users / devs is knowing it and hopefully can helping you :-))

jacekk015, 2021-10-04:
Like I've said - you have my clone - I see it from attributes.
Remove that valve.py file from zha_custom_quirks folder.
Download that ZIP below:
[moes_3.py.zip](https://github.com/zigpy/zha-device-handlers/files/7280154/moes_3.py.zip)
unpack it and upload it to the zha_custom_quirks folder.
Restart HomeAssistant.
This is a DEV version but you will have a possibility to steer the TRV and get temps.
Child lock should also work - there's a switch for that.
I'm still working on it so please be patient. There are 2pcs on my desk since over a year ;-)

baldisos, 2021-10-04:
After deleting the valve.py and adding your file, i can see the battery status, and the temperature.
But i can't set the desired temperature. The dial inside Lovelace isn't moveable and on the details modal i can only set the temp between 34°C and 29.5°C.

If i'm setting the temp on the device itself it gets updated correctly inside HA.

What data would help?

jacekk015, 2021-10-04:
Remove and re-add the valve.
Min/Max temps have not changed after file switch.

baldisos, 2021-10-04:
After deleting the pycache and re-pairing the device, the temperature control now working. Although the "heat" and "off" modes do nothing. Thanks to both of you @jacekk015 and @MattWestb!

jacekk015, 2021-10-04:
As for "heat" and "off" modes. Other valves present the valve state, or even valve open percentage. Then you can show the user if the valve is heating or not. This valve doesn't present such data.
We could guess that if the room temp is below target temp the valve will be heating, but that's not the data from the valve itself.
Like I've wrote earlier - you have a DEV version.

baldisos, 2021-10-04:
Okay, i can live with that. At least i can finally control this thing after 10 months. Thanks for your work!

MattWestb, 2021-10-04:
@jacekk015 I have putting your and this TRV as work in progress but is not in dev in the tuya TRV matrix https://github.com/zigpy/zigpy/discussions/653#discussioncomment-314395.

But i have no model name (and only tuya and brand) for your TRV ;-((
Do you have some more information i can putting in the matrix ?

jacekk015, 2021-10-04:
TRV itself is clean of the symbol, like totally clean.
In the manual I've found SH3 model name, only on the pairing image.
Internet knows this type as a SH3
https://expo.tuya.com/product/323563
https://www.youtube.com/watch?v=LQ3IWrMEF_g
https://www.aliexpress.com/item/1005001997990414.html

and this is above TRV
https://expo.tuya.com/product/842242

MattWestb, 2021-10-04:
Great and thanks for info, i have adding the model SH3 and most of the "Moes Alt" is SH4 but if its working the same its OK.

I think its only the "real" Moes TRVs that have all functions working and the other have setting the set point and mode and scheduling is missing but the basic function is better then no function at all.

Have you looking in Z2M git ? 
Is 2 treys implanting but no have being merged but many commands is implanted for this TRVs.

jacekk015, 2021-10-04:
Z2M code is a small help. Code for ZHA is totally different. Z2M is JavaScript where you do a translator toZigbee and fromZigbee and put datapoints to Tuya lib - kind of easy approach.
For ZHA, oh man, plenty of classes here and there, few level of inheritance of classes.
For example you have a "state_change" event in the valve.py, but the code is really executed in the
/zhaquirks/tuya/__init__.py file. In Chrome I have 7 tabs, of zigpy and zha source code, opened all the time.
You need to find where is which part of code implemented.
One week took me to read the code and find the connections.
Working Z2M quirk code takes about 300 lines. ZHA python now has 676 lines to achieve about same functionality. If I end below 900 lines I will be happy.

MattWestb, 2021-10-06:
I have adding  one more ID in the matrix and its the Silvercrest / LIDL TRV (I dont have any model ID for it) but its looks like this one if looking on the picture in the links.
https://github.com/Koenkk/zigbee2mqtt/issues/8934.

jacekk015, 2021-11-04:
@baldisos If you are interested you can test newest version here
https://github.com/zigpy/zha-device-handlers/issues/1061#issuecomment-961433590
Lots of changes.

MattWestb, 2021-11-09:
@jacekk015 I have putting in little functions in the matrix for your TRVs but i dont knowing if i have finding and understanding all right. Can you taking on look and saying what is wrong so i can correcting it (Battery reporting i dont knowing).
What is working OK i like putting ✅ on.
```
## MAXSMART  MS-SH1 ZG / MS-SH3 ZG / MS-SH4 ZG
![MS-SH1A](https://user-images.githubusercontent.com/49618193/137137903-0234124c-9009-4232-ac23-02c4d3ff1da2.PNG)  ![MS-SH3A](https://user-images.githubusercontent.com/49618193/137137931-86bc8a2c-b423-4228-8407-953cb17c918b.PNG)  ![MS-SH4A](https://user-images.githubusercontent.com/49618193/137137962-ea2a09a9-157a-4d70-b369-7806b91c58eb.PNG)

#### Design:
1. MS-SH1 ZG First picture.
2. MS-SH3 ZG Second picture.
3. MS-SH4 ZG Third picture.

#### Most found functions is implanted:

Modes:
-  0 Scheduled ⬜
-  1 Away ⬜
-  2 Manual ⬜

Setting / reporting Auto setpoint temperature ⬜
Setting / reporting Manual setpoint temperature ⬜
Reporting local temperature ⬜
Calibrating local temperature ⬜
Setting min heat setpoing limit ⬜
Setting max heat setpoing limit ⬜
Setting Away mode setpoint temperature date and duration ⬜
Setting Comfort mode setpoint  temperature ⬜
Setting ECO mode setpoint temperature ⬜

Setting / reporting child lock ⬜
Setting / reporting window detection (on/off, temperature, operating time (min)) ⬜
Reporting battery low warning ⬜ (% or low ?)
Setting / reporting Scheduled (Not implanted on cluster only local) ⭕

#### ZHA support: 
☑️  Yes but support in device request.
✅ =  Yes working. ⬜ = Yes but functions is not verified to working OK ⭕ = Not implanted. 

  
#### Known devices:
MAXSMART is making different software IDs for all customers and also different functions on the same design.
| OEM / Brand | Model | Manufacturer Name | Model ID | ZHA Quirk | Request |
|--------|---------|-----------------------|-----------|-------------|----------|
| MAXSMART | MS-SH4 ZG | _TZE200_fhn3negr | TS0601 | ☑️ MaxS | [ 938 ](https://github.com/zigpy/zha-device-handlers/issues/938) |
| MAXSMART | MS-SH3 ZG | _TZE200_qc4fpmcn | TS0601 | ☑️ MaxS | [ 677 ](https://github.com/zigpy/zha-device-handlers/issues/677) |
| Moes | SH4-ZB | _TZE200_zion52ef | TS0601 | ⭕MaxS ? | [ 938 ](https://github.com/zigpy/zha-device-handlers/issues/938) |
| AWOW | MS-C-4 | _TZE200_thbr5z34| TS0601 | ☑️ MaxS | [978](https://github.com/zigpy/zha-device-handlers/issues/978) |
| Essentials | 120144 | _TZE200_i48qyn9s | TS0601 | ☑️ MaxS | [ 677 ](https://github.com/zigpy/zha-device-handlers/issues/677) |
| Silvercrest (LIDL) | 100335076 | _TZE200_chyvmhay | TS0601 | ☑️ MaxS |  [1061](https://github.com/zigpy/zha-device-handlers/issues/1061) |

⭕ = Not implanted: look in the last section "Adding support for not supported device"
☑️ = Not in dev but working in the device request.
All MAXSMART TRVs need extra patching in ZHA for getting all enchanted function to working. ([Example](https://github.com/home-assistant/core/pull/52065))
```
Take the ` ``` ` for getting the formatting working but loosing the links.

The tuya TRV Matrix https://github.com/zigpy/zigpy/discussions/653#discussioncomment-314395

jacekk015, 2021-11-09:
```
## MAXSMART  MS-SH1 ZG / MS-SH3 ZG / MS-SH4 ZG
![MS-SH1A](https://user-images.githubusercontent.com/49618193/137137903-0234124c-9009-4232-ac23-02c4d3ff1da2.PNG)  ![MS-SH3A](https://user-images.githubusercontent.com/49618193/137137931-86bc8a2c-b423-4228-8407-953cb17c918b.PNG)  ![MS-SH4A](https://user-images.githubusercontent.com/49618193/137137962-ea2a09a9-157a-4d70-b369-7806b91c58eb.PNG)

#### Design:
1. MS-SH1 ZG First picture.
2. MS-SH3 ZG Second picture.
3. MS-SH4 ZG Third picture.

#### Most found functions is implanted:

Modes:
-  0 Scheduled ✅
-  1 Away ✅
-  2 Manual ✅

Setting / reporting Auto setpoint temperature ✅
Setting / reporting Manual setpoint temperature ✅
Reporting local temperature ✅
Calibrating local temperature (Not implanted on cluster only local) ⭕ - not implemented, but can be added if needed
Setting min heat setpoing limit ⬜ - no attribute exists, defined manually in code
Setting max heat setpoing limit ⬜ - no attribute exists, defined manually in code
Setting Away mode setpoint temperature date and duration ✅
Setting Comfort mode setpoint  temperature ✅
Setting ECO mode setpoint temperature ✅

Setting / reporting child lock ✅
Setting / reporting window detection (on/off, temperature, operating time (min)) ✅
Reporting battery low warning ✅ (% or low ?) - not warning but battery level
Setting / reporting Scheduled (Not implanted on cluster only local) ⭕

#### ZHA support: 
☑️  Yes but support in device request.
✅ =  Yes working. ⬜ = Yes but functions is not verified to working OK ⭕ = Not implanted. 

  
#### Known devices:
MAXSMART is making different software IDs for all customers and also different functions on the same design.
| OEM / Brand | Model | Manufacturer Name | Model ID | ZHA Quirk | Request |
|--------|---------|-----------------------|-----------|-------------|----------|
| MAXSMART | MS-SH4 ZG | _TZE200_fhn3negr | TS0601 | ☑️ MaxS | [ 938 ](https://github.com/zigpy/zha-device-handlers/issues/938) |
| MAXSMART | MS-SH3 ZG | _TZE200_qc4fpmcn | TS0601 | ☑️ MaxS | [ 677 ](https://github.com/zigpy/zha-device-handlers/issues/677) |
| Moes | SH4-ZB | _TZE200_zion52ef | TS0601 | ⭕MaxS ? | [ 938 ](https://github.com/zigpy/zha-device-handlers/issues/938) |
| AWOW | MS-C-4 | _TZE200_thbr5z34| TS0601 | ☑️ MaxS | [978](https://github.com/zigpy/zha-device-handlers/issues/978) |
| Essentials | 120144 | _TZE200_i48qyn9s | TS0601 | ☑️ MaxS | [ 677 ](https://github.com/zigpy/zha-device-handlers/issues/677) |
| Silvercrest (LIDL) | 100335076 | _TZE200_chyvmhay | TS0601 | ☑️ MaxS |  [1061](https://github.com/zigpy/zha-device-handlers/issues/1061) |

⭕ = Not implanted: look in the last section "Adding support for not supported device"
☑️ = Not in dev but working in the device request.
All MAXSMART TRVs need extra patching in ZHA for getting all enchanted function to working. ([Example](https://github.com/home-assistant/core/pull/52065))

```

MattWestb, 2021-11-09:
On my TRV then setting min and max attribute on the thermostat cluster ZHA is saving it the attributes in the DB and using it for ZHA GUI for min and max setpoint therefor i have making it "static" so the GUI have the right scale (ZHA have min temp 7°C my have 5°C).

Moes is sending it to the TRV 2 but i dont knowing if its doing some good or not.

Looks much better and not like one "Austrian corona Ample" :-))

Thanks for help fixing our TRVs !!

MattWestb, 2021-11-09:
On thermostat cluster:
```
    _CONSTANT_ATTRIBUTES = {
        0x0015: 500,  # Min heating SP
        0x0016: 3000,  # Max heating SP
```
Is making ZHA GUT setting min and max what can being sett in the GUI.

But I have losing in on my hotted code :-(((

jacekk015, 2021-11-09:
I've switched to functions in INIT. Constant attributes didn't work for me after restart or something. It was some kind of problem, so I've switched.
```
    def __init__(self, *args, **kwargs):
        """Init."""
        super().__init__(*args, **kwargs)
        self.endpoint.device.thermostat_bus.add_listener(self)
        self.endpoint.device.thermostat_bus.listener_event(
            "temperature_change",
            "min_heat_setpoint_limit",
            MAXSMART_MIN_TEMPERATURE_VAL,
        )
        self.endpoint.device.thermostat_bus.listener_event(
            "temperature_change",
            "max_heat_setpoint_limit",
            MAXSMART_MAX_TEMPERATURE_VAL,
        )
```

MattWestb, 2021-11-09:
So i have not loosing my min and max then reading the clusters and ZHA GUI is having min and max as it shall !!

I think the "simple method" is only working is making one write but is not triggering one update like it was done "in real" so you is having right and have fixing it :-)))

Thanks one more time for that  !!

Then you have implanting setting min and max attribute in MaxSmart so ZHA is getting it i can taking the function away then its not needed.
Perhaps writing the min and max temps in the setpoints like i have for some TRVs that i knowing the real ones.

jacekk015, 2021-11-09:
To everything to look nice I've just put those two near attributes, but end of the name is VAL - not ATTR
It's easier to make modifications in one place.
```
# minimum limit of temperature setting
MAXSMART_MIN_TEMPERATURE_VAL = 50  # degrees/100
# maximum limit of temperature setting
MAXSMART_MAX_TEMPERATURE_VAL = 2950  # degrees/100
```

jacekk015, 2021-12-05:
@baldisos Close that issue if all is OK. Thanks!