zigpy/zha-device-handlers #683: [Device Support Request] Schwaiger ZHS 20 Smoke Detector  (HEIMAN SMOK_YDLV10)

baldisos, 2020-12-30:
Not sure if Bug or not supported right now.
I've paired the smoke detector, and it detects smoke when tested. But it doesn't report battery status.

According to z2mqtt
[https://github.com/ioBroker/ioBroker.zigbee/issues/137](url)
[https://github.com/Koenkk/zigbee2mqtt/issues/384](url)
It should also report battery percentage. 

Accorting to deconz:
[https://github.com/dresden-elektronik/deconz-rest-plugin/issues/352](url)
It should report state.fire config.battery state.lowbattery state.tampered too.

Maybe related to [https://github.com/zigpy/zha-device-handlers/issues/518](url)?

**Device signature**
```
{
  "node_descriptor": "NodeDescriptor(byte1=2, byte2=64, mac_capability_flags=132, manufacturer_code=48042, maximum_buffer_size=64, maximum_incoming_transfer_size=0, server_mask=0, maximum_outgoing_transfer_size=0, descriptor_capability_field=3)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0402",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x0009",
        "0x0500",
        "0x0502"
      ],
      "out_clusters": [
        "0x0019"
      ]
    }
  },
  "manufacturer": "Heiman",
  "model": "SMOK_YDLV10",
  "class": "zigpy.device.Device"
}
```

Pairing Log:

```
[0x0000:zdo] ZDO request ZDOCmd.Mgmt_Permit_Joining_req: [60, <Bool.false: 0>]
[0x1ff6] Delivery error for seq # 0x7b, on endpoint id 1 cluster 0x0500: message send failure
[0x1FF6:1:0x0500]: Failed to send pro-active IAS enroll response: [0x1ff6:1:0x0500]: Message send failure
[0x1FF6:1:0x0500]: finished IASZoneChannel configuration
[0xa666:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=116 command_id=2>
[0xa666:1:0xef00] Unknown cluster-specific command 2
[0xa666:1:0xef00] ZCL request 0x0002: b'\x00\x18\x18\x02\x00\x04\x00\x00\x00\xd2'
[0xa666:1:0xef00] No handler for cluster command 2
Device 0xad3b (00:50:43:c9:03:37:19:41) joined the network
[0xad3b] Requesting 'Node Descriptor'
Tries remaining: 2
[0xad3b] Extending timeout for 0x7f request
Device 0xad3b (00:50:43:c9:03:37:19:41) joined the network
Skip initialization for existing device 00:50:43:c9:03:37:19:41
[0xad3b:zdo] ZDO request ZDOCmd.Device_annce: [0xAD3B, 00:50:43:c9:03:37:19:41, 132]
[0x6073:3:0x0b04]: async_update
[0xad3b] Node Descriptor: NodeDescriptor(byte1=2, byte2=64, mac_capability_flags=132, manufacturer_code=48042, maximum_buffer_size=64, maximum_incoming_transfer_size=0, server_mask=0, maximum_outgoing_transfer_size=0, descriptor_capability_field=3)
[0xad3b] Discovering endpoints
Tries remaining: 3
[0xad3b] Extending timeout for 0x83 request
[0x6073:3:0x0b04] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=129 command_id=Command.Read_Attributes_rsp>
Received frame on uninitialized device 00:50:43:c9:03:37:19:41 (Status.NEW) for endpoint: 1
[0xad3b] Received ZCL while uninitialized on endpoint id 1, cluster 0x0500 id, hdr: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=0 command_id=1>, payload: b'(\x00\xff\xff'
[0xad3b] Discovered endpoints: [1]
[0xad3b:1] Discovering endpoint information
Tries remaining: 3
[... 81 lines cut ...]
[0xad3b] Extending timeout for 0x9f request
[0xAD3B:1:0x0502]: initializing channel: from_cache: False
[0xAD3B:1:0x0019]: initializing channel: from_cache: False
[0xad3b:1:0x0000] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=157 command_id=Command.Read_Attributes_rsp>
[0xAD3B:1:0x0000]: initializing channel: from_cache: False
[0xad3b:1:0x0500] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=159 command_id=Command.Read_Attributes_rsp>
[0xAD3B:1:0x0500]: initializing channel: from_cache: False
[0xAD3B:1:0x0000]: 'async_initialize' stage succeeded
[0xAD3B:1:0x0500]: 'async_initialize' stage succeeded
[0xAD3B:1:0x0502]: 'async_initialize' stage succeeded
[0xAD3B:1:0x0019]: 'async_initialize' stage succeeded
[0xAD3B](SMOK_YDLV10): power source: Mains
[0xAD3B](SMOK_YDLV10): completed initialization
[0xa666:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=124 command_id=2>
[0xa666:1:0xef00] Unknown cluster-specific command 2
[0xa666:1:0xef00] ZCL request 0x0002: b'\x00\x18\x18\x02\x00\x04\x00\x00\x00\xd2'
[0xa666:1:0xef00] No handler for cluster command 2
[0xa666:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=127 command_id=2>
[0xa666:1:0xef00] Unknown cluster-specific command 2
[0xa666:1:0xef00] ZCL request 0x0002: b'\x00\x18\x18\x02\x00\x04\x00\x00\x00\xd2'
[0xa666:1:0xef00] No handler for cluster command 2
[0x6073:3:0x0b04]: async_update
[0x6073:3:0x0b04] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=161 command_id=Command.Read_Attributes_rsp>

```

Adminiuga, 2020-12-30:
It reports itself as mains powered, so ZHA does not create the battery entity for it.

baldisos, 2020-12-30:
Can this be fixed, or is this permanent? If i understood right, some people got battery stats working with zigbee2mqtt.

tube0013, 2020-12-30:
doesn't the Plaid moisture sensor report as mains as well so should be fixable with a quirk?

Adminiuga, 2020-12-30:
what plaid version do you have? the one in my possession does report the battery.

tube0013, 2020-12-30:
> what plaid version do you have? the one in my possession does report the battery.

the gen3 I think. #291 I guess it was reporting the voltage over the mains_voltage attribute. fixed in #329

Adminiuga, 2020-12-30:
ah, yes. So plaid battery was a different issue.

tube0013, 2020-12-30:
yea, sorry for the sidetrack

dmulcahey, 2020-12-30:
All we should have to do is flip a bit in the node descriptor no?

Adminiuga, 2020-12-30:
yeah, should replace node descriptor with `NodeDescriptor(2, 64, 132 & 0b1111_1011, 48042, 64, 0, 0, 0, 3)`

Adminiuga, 2020-12-30:
I thought we already did this either for Heinman smoke or CO sensor

dmulcahey, 2020-12-30:
Poole remotes IIRC

Adminiuga, 2020-12-30:
Bingo - https://github.com/zigpy/zha-device-handlers/blob/50f603e80ab57ced3c7aa43101ecbe3c18755a5b/zhaquirks/xiaomi/aqara/opple_remote.py#L203

PR #686 (2020-12-31): Add quirk for Heiman SMOK_YDLV10
Add quirk to override node descriptor and make Heiman SMOK_YDLV10 report itself as battery powered device.
Fixes #683
closed 2020-12-31 by dmulcahey