zigpy/zha-device-handlers #689: [Device Support Request] Improve support for Spirit Zigbee Radiator Thermostat SPZB0001

theorician, 2020-12-31:
**Is your feature request related to a problem? Please describe.**
As per the initial issue at https://github.com/zigpy/zha-device-handlers/issues/356, it seems as though support exists for these thermostats. I'm having trouble getting mine to join though. As per https://zigbee.blakadder.com/Eurotronic_SPZB0001.html it seems to be supported; I've followed the reset/join notes including hitting the Boost button to get them to join the network. No luck.

**Describe the solution you'd like**
I'd like to be able to join these devices to my Zigbee network. 

**Device signature**
I can't get _anything_ here. By pure chance one just ... showed up, despite me never doing a reset or adding it. I'm not familiar with how Zigbee networks authenticate, etc, but it seems to have been a holdover from the deconz network I had set up before. (The deconz container is gone and destroyed of course.) Here's how that looks: 

[image]

I bought them end of August 2019 - perhaps their firmware is older than what's currently supported here?

**Additional context**
Running on the latest Home Assistant stable build, 2020.12, in the official Docker container, using a Conbee I and of course ZHA (not deconz.) I moved from deconz to ZHA today and everything has been a breeze to set up except these. They weren't trivial to set up with deconz either, I remember.

Adminiuga, 2020-12-31:
Device handlers cannot address problems with joining. 
In your screenshot, the quirk value indicates that device joined network correctly and was interviewed/discovered successfully. 
You may want to enable debug logging and monitor the logs during the join process.

theorician, 2021-01-03:
@Adminiuga Thank you. I shall do that!

closed 2021-01-03 by theorician

theorician, 2021-01-03:
@Adminiuga I've gone and turned on the firehose. I'm not sure what I'm looking at here mostly.

I've tried to clean up most of the things though - I went and filtered out log lines containing nwks of existing devices, and did a bit more filtering. It's still a ton though. Does anything really obvious jump out at you?

(There's a couple of lines in there referencing 'eurotronics' - I assume strongly that those are the one that magically joined rather than the new ones that I'm trying to join.)

Thanks for the help!

Log follows.

```

2021-01-03 15:34:08 DEBUG (MainThread) [zigpy.zdo] [0x0000:zdo] ZDO request ZDOCmd.Mgmt_Permit_Joining_req: [60, <Bool.false: 0>]
2021-01-03 15:34:08 DEBUG (MainThread) [zigpy_deconz.api] Command Command.aps_data_confirm (0,)
2021-01-03 15:34:08 DEBUG (MainThread) [zigpy_deconz.uart] Send: 0x04450007000000
2021-01-03 15:34:08 DEBUG (MainThread) [zigpy_deconz.uart] Frame received: 0x04450012000b00221201fcff000000000000
2021-01-03 15:34:08 DEBUG (MainThread) [zigpy_deconz.api] APS data confirm response for request with id 18: 00
2021-01-03 15:34:08 DEBUG (MainThread) [zigpy_deconz.api] Request id: 0x12 'aps_data_confirm' for <DeconzAddressEndpoint address_mode=ADDRESS_MODE.GROUP address=0xfffc endpoint=None>, status: 0x00
2021-01-03 15:34:08 DEBUG (MainThread) [zigpy_deconz.api] Command Command.write_parameter (2, <NetworkParameter.permit_join: 33>, b'<')
2021-01-03 15:34:08 DEBUG (MainThread) [zigpy_deconz.uart] Send: 0x0b460009000200213c
2021-01-03 15:34:08 DEBUG (MainThread) [zigpy_deconz.uart] Frame received: 0x0b46000800010021
2021-01-03 15:34:08 DEBUG (MainThread) [zigpy_deconz.api] Write parameter permit_join: SUCCESS
2021-01-03 15:34:08 DEBUG (MainThread) [zigpy_deconz.uart] Frame received: 0x0e47000700aa00
2021-01-03 15:34:08 DEBUG (MainThread) [zigpy_deconz.api] Device state changed response: [<DeviceState.128|APSDE_DATA_REQUEST_SLOTS_AVAILABLE|APSDE_DATA_INDICATION|2: 170>, 0]
2021-01-03 15:34:08 DEBUG (MainThread) [zigpy_deconz.uart] Send: 0x1747000800010004
2021-01-03 15:34:08 DEBUG (MainThread) [zigpy_deconz.uart] Frame received: 0x17470035002e00220200000104d4fdfcdb9201008d150001040101020f0018130129008614000029400613008600afffa3000102cb
2021-01-03 15:34:08 DEBUG (MainThread) [zhaquirks.eurotronic] update attribute 001c to SystemMode.Heat... 
2021-01-03 15:34:08 DEBUG (MainThread) [zhaquirks.eurotronic] update attribute 001c to SystemMode.Heat... [ ok ]
2021-01-03 15:34:08 DEBUG (MainThread) [zhaquirks.eurotronic] update attribute 0014 to 1600... 
2021-01-03 15:34:08 DEBUG (MainThread) [zhaquirks.eurotronic] update attribute 0014 to 1600... [ ok ]
2021-01-03 15:34:08 DEBUG (MainThread) [homeassistant.components.zha.core.gateway] Loading mains powered devices
2021-01-03 15:34:08 DEBUG (MainThread) [homeassistant.components.zha.core.device] [0x0000](ConBee): started initialization
2021-01-03 15:34:08 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x0000:ZDO](ConBee): 'async_initialize' stage succeeded
2021-01-03 15:34:08 DEBUG (MainThread) [zigpy_deconz.api] Command Command.aps_data_request (20, 22, 0, <DeconzAddressEndpoint address_mode=2 address=0x0000 endpoint=1>, 260, 0, 1, b'\x00\x15\x00\x07\x00', 2, 0)
2021-01-03 15:34:08 DEBUG (MainThread) [zigpy_deconz.uart] Send: 0x1248001b0014001600020000010401000001050000150007000200
2021-01-03 15:34:08 DEBUG (MainThread) [zigpy_deconz.uart] Frame received: 0x124800090002002216
2021-01-03 15:34:08 DEBUG (MainThread) [zigpy_deconz.api] APS data request response: [2, <DeviceState.APSDE_DATA_REQUEST_SLOTS_AVAILABLE|2: 34>, 22]
2021-01-03 15:34:08 DEBUG (MainThread) [zigpy_deconz.uart] Send: 0x1249001b001400180002d983030401060001050000170000000200
2021-01-03 15:34:08 DEBUG (MainThread) [zigpy_deconz.uart] Frame received: 0x0e49000700ae00
2021-01-03 15:34:08 DEBUG (MainThread) [zigpy_deconz.api] Device state changed response: [<DeviceState.128|APSDE_DATA_REQUEST_SLOTS_AVAILABLE|APSDE_DATA_INDICATION|APSDE_DATA_CONFIRM|2: 174>, 0]
2021-01-03 15:34:08 DEBUG (MainThread) [zigpy_deconz.uart] Frame received: 0x124900090002002e18
2021-01-03 15:34:08 DEBUG (MainThread) [zigpy_deconz.api] APS data request response: [2, <DeviceState.APSDE_DATA_REQUEST_SLOTS_AVAILABLE|APSDE_DATA_INDICATION|APSDE_DATA_CONFIRM|2: 46>, 24]
2021-01-03 15:34:08 DEBUG (MainThread) [zigpy_deconz.uart] Send: 0x124a001b0014001a0002d983030401000001050000190007000200
2021-01-03 15:34:08 DEBUG (MainThread) [zigpy_deconz.uart] Frame received: 0x0e4a000700ae00
2021-01-03 15:34:08 DEBUG (MainThread) [zigpy_deconz.api] Device state changed response: [<DeviceState.128|APSDE_DATA_REQUEST_SLOTS_AVAILABLE|APSDE_DATA_INDICATION|APSDE_DATA_CONFIRM|2: 174>, 0]
2021-01-03 15:34:08 DEBUG (MainThread) [zigpy_deconz.uart] Frame received: 0x124a00090002002e1a
2021-01-03 15:34:08 DEBUG (MainThread) [zigpy_deconz.api] APS data request response: [2, <DeviceState.APSDE_DATA_REQUEST_SLOTS_AVAILABLE|APSDE_DATA_INDICATION|APSDE_DATA_CONFIRM|2: 46>, 26]
2021-01-03 15:34:08 DEBUG (MainThread) [zigpy_deconz.uart] Send: 0x124b0021001a001c0002d983030401040b010b00001b0005060304040602040200
2021-01-03 15:34:08 DEBUG (MainThread) [zigpy_deconz.uart] Frame received: 0x0e4b000700ae00
2021-01-03 15:34:08 DEBUG (MainThread) [zigpy_deconz.api] Device state changed response: [<DeviceState.128|APSDE_DATA_REQUEST_SLOTS_AVAILABLE|APSDE_DATA_INDICATION|APSDE_DATA_CONFIRM|2: 174>, 0]
[... cut ...]
2021-01-03 15:36:14 DEBUG (MainThread) [zigpy_deconz.api] Device state changed response: [<DeviceState.128|APSDE_DATA_REQUEST_SLOTS_AVAILABLE|APSDE_DATA_INDICATION|APSDE_DATA_CONFIRM|2: 174>, 0]
2021-01-03 15:36:14 DEBUG (MainThread) [zigpy_deconz.api] APS data confirm response for request with id 150: 00
2021-01-03 15:36:14 DEBUG (MainThread) [zigpy_deconz.api] Device state changed response: [<DeviceState.128|APSDE_DATA_REQUEST_SLOTS_AVAILABLE|APSDE_DATA_INDICATION|2: 170>, 0]
2021-01-03 15:36:20 DEBUG (MainThread) [zigpy_deconz.api] Device state changed response: [<DeviceState.128|APSDE_DATA_REQUEST_SLOTS_AVAILABLE|APSDE_DATA_INDICATION|2: 170>, 0]
```

Adminiuga, 2021-01-03:
Without zigpy_deconz.zigbee.application debug logging, this log is much harder to read. I do see simplified beacon frames (the ones starting with 0x1f) which indicate that device is scanning the network. However if the device does not send the Mgmt_Device_annce broadcast it wouldn't know that a new device joined. 

Try resetting the trv valve to factory settings and join again.

theorician, 2021-01-03:
@Adminiuga Thanks for the response! I added these 
```
logger:
  default: warn
  logs:  
    homeassistant.core: debug
    homeassistant.components.zha: debug
    bellows.zigbee.application: debug
    bellows.ezsp: debug
    zigpy: debug
    zigpy_cc: debug
    zigpy_deconz.zigbee.application: debug
    zigpy_deconz.api: debug
    zigpy_xbee.zigbee.application: debug
    zigpy_xbee.api: debug
    zigpy_zigate: debug
    zigpy_znp: debug
    zhaquirks: debug  
```
so I think `zigpy_deconz.zigbee.application` debug logs should be in there. As I said, I removed the ones that weren't relevant (because they referred directly to a specific device). 

Unfortunately, I've been resetting/joining the device each time - it doesn't do much good it seems. 

Anyway, is there a better place for me to ask for help? Since you said that joining issues aren't really handled by zha-device-handlers, this is probably the wrong venue. I'm happy to take it elsewhere!

Adminiuga, 2021-01-03:
> so I think zigpy_deconz.zigbee.application debug logs should be in there.

But it's not there.

 You can ask zigbee channel on ha discord server if there are any other users of the same valves

theorician, 2021-01-03:
@Adminiuga No, it is there. It's just... not outputting anything useful. The only logs that it spits out are e.g.
```
2021-01-03 18:12:09 DEBUG (MainThread) [zigpy_deconz.zigbee.application] Sending Zigbee request with tsn 147 under 148 request id, data: b'0093000b05'
```
and also this specific "unknown device" (which is not the trv. I have two that won't pair and I've tried both. It's always the same nwk address.)
```
2021-01-03 18:11:58 DEBUG (MainThread) [zigpy_deconz.zigbee.application] Received frame from unknown device: 0x418f
```

Those two lines (repeated often of course) are the only ones from `zigpy_deconz.zigbee.application`.

Adminiuga, 2021-01-03:
Well, you decided those don't matter and essentially wasted my time. There's is a reason why those are requested.

The unknown device means that it joined the network but failed to be discovered. Power cycle it, so it may send a new zdo device annce message. Also, if it joins under the same nwk every time, that usually indicates that device was not reset.

theorician, 2021-01-03:
No, what I'm saying is that `Received frame from unknown device: 0x418f` is a red herring. It appears also when both the TRVs I'm trying to add are unpowered. 

The `Sending Zigbee request with tsn 147 under 148 request id, data: b'0093000b05'` lines, I'm not sure why they aren't showing the large log block I copied in above - it might be that there just weren't any during the period I copied.

theorician, 2021-01-04:
So in case anyone else finds themselves here and wonders if I fixed it - I worked around the problem, though in a slightly unsatisfying way. The TRVs work just fine now though.

Here's what I did. 

I turned off my HA docker image and started a fresh instance of the deconz docker image with: 
```
docker run -e DECONZ_VNC_MODE=1 -e DECONZ_UPNP=0 -v $(pwd):/root/.local/share/dresden-elektronik/deCONZ -p 8080:80 -p 5900:5900 --device /dev/ttyUSB0:/dev/ttyUSB0 -it marthoc/deconz
```

It started, and then I got sidetracked and [did a firmware update](https://github.com/marthoc/docker-deconz#updating-conbeeraspbee-firmware) since it told me there's a new firmware version.

Once that was done, I VNCed in and checked the network graph. I put the TRVs into pairing mode via a factory reset, and the Conbee as well via the Phoscon app. The TRVs showed up in the network graph immediately. Once they showed up, I shut down the deconz image.

Then I stared HA again. There, I went into the ZHA device search and power cycled the TRVs. Not a factory reset! Just removing the battery and putting them back in. They showed up immediately and paired as normal. They now work fine.

@Adminiuga Thanks for all the help, you definitely set me on the correct path to getting this working.