zigpy/zha-device-handlers #691: [Device Support Request] Immax 07703L TRV

stnsmk, 2021-01-02:
Immax 07703L (https://www.immaxneo.cz/produkty/immax-neo-smart-termostaticka-hlavice-zigbee/) is another variation of Tuya TRV, supposedly similar to ETOP HT-10.

**Device signature**
```
{
  "node_descriptor": "NodeDescriptor(byte1=2, byte2=64, mac_capability_flags=128, manufacturer_code=0, maximum_buffer_size=82, maximum_incoming_transfer_size=255, server_mask=11264, maximum_outgoing_transfer_size=255, descriptor_capability_field=0)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0000",
      "in_clusters": [
        "0x0000",
        "0x0003"
      ],
      "out_clusters": [
        "0x0003",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TYST11_wlosfena",
  "model": "losfena",
  "class": "zigpy.device.Device"
}
```

**Additional context**
I've had some limited success while trying to implement support deriving from generic Tuya and SiterwellGS361 (tuya/valve.py) but it was very basic due to my limited experience with zigpy/zhaquirks.

stnsmk, 2021-01-02:
In an unexpected twist another TRV, visually identical and with identical packaging, purchased from the same retailer just one day later than the previous has different signature: 

```
{
  "node_descriptor": "NodeDescriptor(byte1=2, byte2=64, mac_capability_flags=128, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=0)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0051",
      "in_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0xef00"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZE200_wlosfena",
  "model": "TS0601",
  "class": "zigpy.device.Device"
}
```

This one seems to work fine using "Quirk: zhaquirks.tuya.valve.MoesHY368" except the power readings.

I'll try to clarify the situation with the manufacturer and test both devices a bit more using existing code.

stnsmk, 2021-01-04:
After some research it looks like the the first device requires responding to a time request in a way described in this z2m patch: https://github.com/Koenkk/zigbee-herdsman/commit/71c89ce6cae2ce26b3d50ade7281de9d05f3fcc8.

What would be the right way to implement this? Capturing cluster command 0x0024 seems to fit handle_cluster_request, however I'm unsure about the actual reply process.

Adminiuga, 2021-01-04:
as long as the quirk has the "time" cluster in the "replacement", the default time cluster does respond to incoming request https://github.com/zigpy/zigpy/blob/97b01815b0c3dd64c2512a99587a56db9318e9d3/zigpy/zcl/clusters/general.py#L507

stnsmk, 2021-01-04:
Unfortunately no, it's command 0x0024 in Tuya cluster so Time cluster does not help:

```
2021-01-04 18:57:54 DEBUG (MainThread) [zigpy.zcl] [0x7dec:1:0xef00] ZCL request 0x0024: b'\x00\x00'
2021-01-04 18:57:54 DEBUG (MainThread) [zigpy.zcl] [0x7dec:1:0xef00] No handler for cluster command 36
2021-01-04 18:57:56 DEBUG (MainThread) [zigpy.zcl] [0x7dec:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=31 command_id=36>                                                                                                                                                     
2021-01-04 18:57:56 WARNING (MainThread) [zigpy.zcl] [0x7dec:1:0xef00] Unknown cluster-specific command 36
```

However you pointed me in the right direction, I got it working using the implementation similar to Time cluster. Will prepare a PR after some more tests and code cleanup.

Thanks for your help!

Adminiuga, 2021-01-04:
Why Tuya?! Why you had to re-invent the wheel? Rhetorical question...

MattWestb, 2021-01-04:
@stnsmk The tuya controlling chip is the same (type/manufacturer config) but they have changing the zigbee module. Very likely its only putting the 2 manufacturer names in the quirk and is working if tuya not have making one 5 wheel in there zigbee devices.

closed 2021-08-12 by github-actions[bot]

stnsmk, 2021-10-18:
Thanks for checking in, I think I'll be able to test it closer to the end of the week.

MattWestb, 2023-01-19:
Its looks its start moving with fixing support for this TRVs.
Cloud Event is up and running after first try but testing and work is needed for getting all working OK.
If like testing look in https://github.com/zigpy/zha-device-handlers/issues/2072 and adding your device ID for getting the quirk loading by the system.