zigpy/zha-device-handlers #703: Xiaomi Mijia Light Sensor (GZCGQ01LM) in ZHA not detecting battery level anymore (it did in previous versions lower than 2020.12.7)

yfands, 2021-01-06:
Hai The original call is from ( lustreking aka Steve Small) , I (yfands) can confirm Steve's findings indeed no battery level.
Steve posted his issue [here ](https://community.home-assistant.io/t/xiaomi-mijia-zha-not-detecting-battery-level/257827)but no answer there, so iam trying it here. In his original call there are 2 pictures one when all was well and the second picture after the update to 2020.12.7, the one that is not working any more, the difference is the class the device was assigned to.

The Problem
A while ago I got a Xiaomi Mijia Light Sensor (GZCGQ01LM), and it worked well for me, so I eventually ordered 2 more. Before I got them, I updated to the current version of HA (2020.12.7), as well as migrated HA to a new RPi.

When I tried to add them, ZHA found the illuminance entity, but not the battery entity, which my previous sensor had.

In ZHA I see the new sensors listed as Xiaomi, while the previous sensor’s manufacturer was listed as lumi. The model for both is lumi.sen_ill.mgl01

Under the zigbee info for the new ones it says “Power Source: Mains” which is obviously incorrect!
Here are the zigbee details of the good and bad ones, (see his original link).. The configuration below is from me (yfands), so to be short in ZHA found the illuminance entity, but not the battery entity anymore!

## System Health

version | 2020.12.2
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
virtualenv | false
python_version | 3.8.6
os_name | Linux
os_version | 5.4.79-v8
arch | aarch64
timezone | Europe/Amsterdam

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
Github API Calls Remaining | 4646
Installed Version | 1.9.0
Stage | running
Available Repositories | 711
Installed Repositories | 29

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 5.9
-- | --
update_channel | stable
supervisor_version | 2020.12.7
docker_version | 19.03.13
disk_total | 27.2 GB
disk_used | 5.5 GB
healthy | true
supported | true
board | rpi4-64
supervisor_api | ok
version_api | ok
installed_addons | File editor (5.2.0), Samba share (9.3.0), Terminal & SSH (8.10.0), Mosquitto broker (5.1), Node-RED (7.2.11)

</details>

<details><summary>Lovelace</summary>

dashboards | 1
-- | --
mode | storage
views | 10
resources | 20

</details>

Adminiuga, 2021-01-06:
Post zigbee device information from zha device panel

yfands, 2021-01-06:
Sorry, forgot:

Device Zigbee Signature:
{
  "node_descriptor": "NodeDescriptor(byte1=2, byte2=64, mac_capability_flags=132, manufacturer_code=4718, maximum_buffer_size=127, maximum_incoming_transfer_size=100, server_mask=11264, maximum_outgoing_transfer_size=100, descriptor_capability_field=0)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0106",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x0400"
      ],
      "out_clusters": [
        "0x0003"
      ]
    }
  },
  "manufacturer": "XIAOMI",
  "model": "lumi.sen_ill.mgl01",
  "class": "zigpy.device.Device"
}

The battery level worked in Quirk mode: zhaquirk.xiaomi.aqara.illumination.illumination.
and not in the now zigpy class. 

Regards
Frank

Adminiuga, 2021-01-06:
looks like they've changed the manufacturer. should be easy to fix

closed 2021-01-07 by Adminiuga

yfands, 2021-01-09:
Hello,

Thanks for the effort, is it already implemented in release 2021.1.1 ?, if so I deleted the sensor and added it again but unfortunately no change, all the same .... no battery only illumination values..

Adminiuga, 2021-01-10:
Next release

yfands, 2021-01-24:
Hello,

Upgraded tot 2021.1.5
And still no battery after reconnect.

(bytheway, I received the same devices but from LUMI as manufacturer and they work flawlessly, illumination and battery is showing), the same devices from XIAOMO as manufacturer not so much)

Here is the connect log file, maybe it helps....

[0x0000:zdo] ZDO request ZDOCmd.Mgmt_Permit_Joining_req: [60, <Bool.false: 0>]
[0xfa35:1:0x0400] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=5 command_id=Command.Report_Attributes>
[0xfa35:1:0x0400] ZCL request 0x000a: [[Attribute(attrid=0, value=<TypeValue type=uint16_t, value=15564>)]]
[0xfa35:1:0x0400] Attribute report received: measured_value=15564
Device 0xc8ba (04:cf:8c:df:3c:7f:22:da) joined the network
Device 04:cf:8c:df:3c:7f:22:da changed id (0xda56 => 0xc8ba)
Checking quirks for XIAOMI lumi.sen_ill.mgl01 (04:cf:8c:df:3c:7f:22:da)
Considering <class 'zhaquirks.xbee.xbee_io.XBeeSensor'>
Fail because endpoint list mismatch: {232, 230} {1}
Considering <class 'zhaquirks.xbee.xbee3_io.XBee3Sensor'>
Fail because endpoint list mismatch: {232, 230} {1}
Considering <class 'zhaquirks.smartthings.tag_v4.SmartThingsTagV4'>
Fail because device_type mismatch on at least one endpoint
Considering <class 'zhaquirks.smartthings.multi.SmartthingsMultiPurposeSensor'>
Fail because device_type mismatch on at least one endpoint
Considering <class 'zhaquirks.philips.rwl021.PhilipsRWL021'>
Fail because endpoint list mismatch: {1, 2} {1}
Considering <class 'zhaquirks.philips.rwl020.PhilipsRWL020'>
Fail because endpoint list mismatch: {1, 2} {1}
Considering <class 'zhaquirks.philips.rom001.PhilipsROM001'>
Fail because device_type mismatch on at least one endpoint
Considering <class 'zhaquirks.netvox.z308e3ed.Z308E3ED'>
Fail because device_type mismatch on at least one endpoint
Considering <class 'zhaquirks.gledopto.soposhgu10.SoposhGU10'>
Fail because endpoint list mismatch: {11, 13} {1}
Considering <class 'bellows.zigbee.application.EZSPCoordinator'>
Fail because device_type mismatch on at least one endpoint
device - 0xc8ba:04:cf:8c:df:3c:7f:22:da entering async_device_initialized - is_new_join: False
device - 0xc8ba:04:cf:8c:df:3c:7f:22:da has been reset and re-added or its nwk address changed
skipping discovery for previously discovered device - 0xc8ba:04:cf:8c:df:3c:7f:22:da
[0xc8ba](lumi.sen_ill.mgl01): started configuration
[0xc8ba:ZDO](lumi.sen_ill.mgl01): 'async_configure' stage succeeded
[0xc8ba] Extending timeout for 0xb2 request
[0xc8ba] Extending timeout for 0xb4 request
Device 04:cf:8c:df:3c:7f:22:da already exists. Updating it.
Device 0xc8ba (04:cf:8c:df:3c:7f:22:da) joined the network
Skip initialization for existing device 04:cf:8c:df:3c:7f:22:da
[0xc8ba:zdo] ZDO request ZDOCmd.Device_annce: [0xC8BA, 04:cf:8c:df:3c:7f:22:da, 132]
[0xc8ba:zdo] ZDO request ZDOCmd.Node_Desc_req: [0x0000]
[0xc8ba:zdo] Unsupported ZDO request:ZDOCmd.Node_Desc_req
[0xc8ba:1:0x0400]: bound 'illuminance' cluster: Status.SUCCESS
[0xc8ba] Extending timeout for 0xb6 request
[0xc8ba:1:0x0000]: bound 'basic' cluster: Status.SUCCESS
[0xc8ba:1:0x0000]: finished channel configuration
[0xc8ba:1:0x0400] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=182 command_id=Command.Configure_Reporting_rsp>
[0xc8ba:1:0x0400]: reporting 'measured_value' attr on 'illuminance' cluster: 30/900/1: Result: '[[ConfigureReportingResponseRecord(status=0)]]'
[0xc8ba:1:0x0400]: finished channel configuration
[0xc8ba:1:0x0400]: 'async_configure' stage succeeded
[0xc8ba:1:0x0000]: 'async_configure' stage succeeded
[0xc8ba](lumi.sen_ill.mgl01): completed configuration
[0xc8ba](lumi.sen_ill.mgl01): stored in registry: ZhaDeviceEntry(name='XIAOMI lumi.sen_ill.mgl01', ieee='04:cf:8c:df:3c:7f:22:da', last_seen=1611492775.5501995)
[0xc8ba] Extending timeout for 0xb8 request
[0xc8ba:zdo] ZDO request ZDOCmd.Mgmt_Permit_Joining_req: [180, <Bool.true: 1>]
[0xc8ba:1:0x0003] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=184 command_id=Command.Default_Response>
[0xc8ba:1:0x0003]: executed 'trigger_effect' command with args: '(2, 0)' kwargs: '{}' result: [64, <Status.SUCCESS: 0>]
[0xc8ba](lumi.sen_ill.mgl01): started initialization
[0xc8ba:ZDO](lumi.sen_ill.mgl01): 'async_initialize' stage succeeded
[0xc8ba:1:0x0400]: initializing channel: from_cache: False
[0xc8ba] Extending timeout for 0xba request
[0xc8ba:1:0x0000]: initializing channel: from_cache: False
[0xc8ba:1:0x0000]: finished channel configuration
[... 15 lines cut ...]
[0xc8ba:1:0x0400] Attribute report received: measured_value=14914
[0xfbd1:1:0x0400] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=167 command_id=Command.Report_Attributes>
[0xfbd1:1:0x0400] ZCL request 0x000a: [[Attribute(attrid=0, value=<TypeValue type=uint16_t, value=12305>)]]
[0xfbd1:1:0x0400] Attribute report received: measured_value=12305
[... cut ...]
[0xc8ba:1:0x0400] ZCL request 0x000a: [[Attribute(attrid=0, value=<TypeValue type=uint16_t, value=14624>)]]
[0xc8ba:1:0x0400] Attribute report received: measured_value=14624
[0x129f:1:0x0400] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=185 command_id=Command.Report_Attributes>
[0x129f:1:0x0400] ZCL request 0x000a: [[Attribute(attrid=0, value=<TypeValue type=uint16_t, value=15564>)]]
[0x129f:1:0x0400] Attribute report received: measured_value=15564

MattWestb, 2021-01-24:
The commit is merged in ZHA https://github.com/zigpy/zha-device-handlers/releases/tag/0.0.52
but Home Assistant 2021.1.5 is still using the version 0.0.51 so need to waiting little more :-(
(was checking my local file then i was updating HA earlier today to latest released version)

PR #704 (2021-01-07): Add alternative manufacturer for Aqara illumination sensor
Apparently new Aqara Illumination sensors are using `"Xiaomi"` as manufacturer name. Add it to signature.
Fixes #703 
```
{
	"node_descriptor": "NodeDescriptor(byte1=2, byte2=64, mac_capability_flags=132, manufacturer_code=4718, maximum_buffer_size=127, maximum_incoming_transfer_size=100, server_mask=11264, maximum_outgoing_transfer_size=100, descriptor_capability_field=0)",
	"endpoints": {
		"1": {
			"profile_id": 260,
			"device_type": "0x0106",
			"in_clusters": [
				"0x0000",
				"0x0001",
				"0x0003",
				"0x0400"
			],
			"out [... cut]