zigpy/zha-device-handlers #712: [BUG] Philips RWL021 - false multiple presses

koying, 2021-01-10:
**Describe the bug**
False multiple presses are being reported after https://github.com/zigpy/zha-device-handlers/pull/501.

**To Reproduce**
Hard to reproduce, but see the following chain of events: https://gist.github.com/koying/b5a8bffe03b6f5fdf5d9014ce4ace514
We end up with a `on_triple_press`

**Expected behavior**
As the "off" button was pressed in between, the expected result is `on_press`

**Additional context**
I suspect the root cause is because the various `<button>_short_release` events are not handled, only the `<button>_press` ones.
Accordingly, only the 3 "press" were seen, resulting in a "triple".
No clue why "release" events were produced rather than "press" ones. That might be the root cause as well.

koying, 2021-01-16:
Follow-up:
Unrelated to "short release". The "press" are not shown in the events as they are "eaten" by the quirk.

OTOH, looking at a debug log:
```
2021-01-16 14:41:46 DEBUG (MainThread) [zhaquirks.philips] PhilipsRemoteCluster - handle_cluster_request tsn: [174] command id: 0 - args: [[1, 3145728, 0, 33, 0, 0]]
2021-01-16 14:41:46 DEBUG (MainThread) [zhaquirks.philips] PhilipsRemoteCluster - handle_cluster_request tsn: [176] command id: 0 - args: [[1, 3145728, 2, 33, 1, 0]]
2021-01-16 14:41:46 DEBUG (MainThread) [zhaquirks.philips] PhilipsRemoteCluster - handle_cluster_request tsn: [174] command id: 0 - args: [[1, 3145728, 0, 33, 0, 0]]
2021-01-16 14:41:46 DEBUG (MainThread) [zhaquirks.philips] PhilipsRemoteCluster - handle_cluster_request tsn: [176] command id: 0 - args: [[1, 3145728, 2, 33, 1, 0]]
2021-01-16 14:41:46 DEBUG (MainThread) [zhaquirks.philips] PhilipsRemoteCluster - handle_cluster_request tsn: [174] command id: 0 - args: [[1, 3145728, 0, 33, 0, 0]]
2021-01-16 14:41:46 DEBUG (MainThread) [zhaquirks.philips] PhilipsRemoteCluster - handle_cluster_request tsn: [176] command id: 0 - args: [[1, 3145728, 2, 33, 1, 0]]
2021-01-16 14:41:46 DEBUG (MainThread) [zhaquirks.philips] PhilipsRemoteCluster - send_press_event click_count: [3]

2021-01-16 14:42:43 DEBUG (MainThread) [zhaquirks.philips] PhilipsRemoteCluster - handle_cluster_request tsn: [177] command id: 0 - args: [[4, 3145728, 0, 33, 0, 0]]
2021-01-16 14:42:43 DEBUG (MainThread) [zhaquirks.philips] PhilipsRemoteCluster - handle_cluster_request tsn: [179] command id: 0 - args: [[4, 3145728, 2, 33, 1, 0]]
2021-01-16 14:42:43 DEBUG (MainThread) [zhaquirks.philips] PhilipsRemoteCluster - handle_cluster_request tsn: [177] command id: 0 - args: [[4, 3145728, 0, 33, 0, 0]]
2021-01-16 14:42:43 DEBUG (MainThread) [zhaquirks.philips] PhilipsRemoteCluster - handle_cluster_request tsn: [179] command id: 0 - args: [[4, 3145728, 2, 33, 1, 0]]
2021-01-16 14:42:43 DEBUG (MainThread) [zhaquirks.philips] PhilipsRemoteCluster - send_press_event click_count: [2]
```

I see that messages with the same TSN are sent multiple time.
No clue on the root cause, but, as this case is not handled, that definitely leads to false multiple presses.

koying, 2021-01-16:
Doesn't really make sense to capture multiple click on "press", imo.

You end up with sequence:
- "short-release" 
- "short-release" 
- "double-push"

which is not logical.

MattWestb, 2021-01-16:
If i remember right is by ZLL standard one command to one group sended 3 times because its not can being verified that all device members have receiving it (the sender dont know how many devices its in the grupe). If you is sending command 0x0002 (toggle) to cluster 0x0006 (On/Off) to one group and one light is receiving it 3 times (or 2 or 1) its only executing the command one time because all received commands have the same transections number (TSN).

It must bring somthing wrong in the logic that is "converting" the light command but i dont knowing then i is using light switches for lights and HA switches for HA thighs alla Zigbee standard.

koying, 2021-01-17:
Suggested patch.
A further issue with the current implementation is that, in case of long press, we receive first a "single press" before getting the "long release".

I made a difference between a push and a press (=click) for clarification.
Also renamed long_release to long_press, for consistency.

```patch
--- zhaquirks#philips#__init__.py.orig.20210116	2021-01-16 15:10:22.058120643 +0100
+++ zhaquirks#philips#__init__.py	2021-01-17 14:34:25.151146925 +0100
@@ -173,12 +173,24 @@
         )
     }
     BUTTONS = {1: "on", 2: "up", 3: "down", 4: "off"}
-    PRESS_TYPES = {0: "press", 1: "hold", 2: "short_release", 3: "long_release"}
+    PRESS_TYPES = {0: "push", 1: "hold", 2: "short_release", 3: "long_press"}
 
+    last_tsn = -1
     button_press_queue = ButtonPressQueue()
 
     def handle_cluster_request(self, tsn, command_id, args):
         """Handle the cluster command."""
+
+        if tsn <= self.last_tsn:
+            _LOGGER.warning(
+                "PhilipsRemoteCluster - handle_cluster_request duplicate tsn ignored: [%s] last tsn: %s - command id: %s - args: [%s]",
+                tsn,
+                self.last_tsn,
+                command_id,
+                args,
+            )
+            return
+
         _LOGGER.debug(
             "PhilipsRemoteCluster - handle_cluster_request tsn: [%s] command id: %s - args: [%s]",
             tsn,
@@ -186,6 +198,8 @@
             args,
         )
 
+        # We don't receive all TSN. Assume we miss max 2
+        self.last_tsn = tsn if tsn < 254 else -1
         button = self.BUTTONS.get(args[0], args[0])
         press_type = self.PRESS_TYPES.get(args[2], args[2])
 
@@ -219,7 +233,7 @@
                 self.listener_event(ZHA_SEND_EVENT, action, event_args)
 
         # Derive Multiple Presses
-        if press_type == "press":
+        if press_type == "short_release":
             self.button_press_queue.press(send_press_event, button)
         else:
             action = f"{button}_{press_type}"
```

MattWestb, 2021-01-18:
@dmulcahey  and / or @Adminiuga can you please taking one look if the logic is right or if somthing is missing / wrong ?

closed 2021-08-12 by github-actions[bot]