zigpy/zha-device-handlers #713: [Device Support Request] Decoupled mode for ctrl_neutral (QBKG04LM/QBKG03LM/QBKG21LM/QBKG22LM)

javicalle, 2021-01-10:
**Is your feature request related to a problem? Please describe.**
Similar to #365, but for `ctrl_neutral` implementation. I think that once implemented it can be easily adapted.

In my case I want to implement QBKG04LM decoupled mode.
It can be done easily, following @albalaing indications (https://github.com/zigpy/zha-device-handlers/issues/365#issuecomment-636510316).

The problem is that without the click button events it is not useful, so it is necessary to also implement the generation of ZHA events.

**Describe the solution you'd like**

Before implement anything you can see that device generate 4 messages:
```
2021-01-09 21:46:29 DEBUG (MainThread) [zigpy.device] Ignoring message (b'18500a0000100000001001') on cluster 6: unknown endpoint or cluster id: 4
2021-01-09 21:46:31 DEBUG (MainThread) [zigpy.device] Ignoring message (b'18510a00001000') on cluster 6: unknown endpoint or cluster id: 4
2021-01-09 21:46:32 DEBUG (MainThread) [zigpy.device] Ignoring message (b'18520a00001001') on cluster 6: unknown endpoint or cluster id: 4
2021-01-09 21:46:32 DEBUG (MainThread) [zigpy.device] Ignoring message (b'18530a00001002') on cluster 6: unknown endpoint or cluster id: 4
```

This messages are (in order):
* click
* hold
* release
* double_click

You can see that the click message corresponds in some way to the hold+release messages (00001000+00001001)

I think I'm halfway done (the decoupled part is working as expected), but I'm not satisfied with the event generation part. 

When a `replacement` for endpoint 4 (and 5) is implemented you can get the `on_off` attribute reported. But IMO this approach has the following problems:
* add a new entity in HA for click button events
* it is not possible to differentiate the `click` event from `hold` and `release`
* the `double_click` event cannot be handled (parse error)

Want I think that would be implemented is:
* no HA button entity
* full events control, not only hold and release

Any help in this regard is welcome.

**Device signature - this can be acquired by removing the device from ZHA and pairing it again from the add devices screen. Be sure to add the entire content of the log panel after pairing the device to a code block below this line.**
The device signature is known and correct.

**Additional context**
Although I am not sure if it can be done this way, I have tried without success to implement a replacement that transforms events from the `OnOff` cluster to the `MultistateInput` cluster.

My current implementation is as follows (only the relevant part):
```
    class BasicClusterDecoupled(BasicCluster):
        """Adds attributes for decoupled mode"""
        def __init__(self, *args, **kwargs):
            """Init."""
            self.attributes = BasicCluster.attributes.copy()
            self.attributes.update({ 0xFF22: ("left_decoupled_mode", t.uint8_t)})
            self.attributes.update({ 0xFF23: ("right_decoupled_mode", t.uint8_t)})
            super().__init__(*args, **kwargs)

    class CustomOnOffCluster(OnOffCluster):
        """Fire ZHA events for on off cluster."""

        cluster_id = OnOff.cluster_id

        def __init__(self, *args, **kwargs):
            """Init."""
            self._current_state = {}
            super().__init__(*args, **kwargs)

        def _update_attribute(self, attrid, value):
            _LOGGER.info("%s: %s", attrid, value)
            if attrid == 0:
                self._current_state = PRESS_TYPES.get(value)
                event_args = {
                    PRESS_TYPE: self._current_state,
                    VALUE: value,
                }
                self.listener_event(ZHA_SEND_EVENT, COMMAND_CLICK, event_args)
            super()._update_attribute(attrid, self._current_state)

    replacement = {
        SKIP_CONFIGURATION: True,
        ENDPOINTS: {
            .../...
            4: {                                                                   
                DEVICE_TYPE: zha.DeviceType.ON_OFF_SWITCH,                         
                INPUT_CLUSTERS: [                                                  
                    MultistateInput.cluster_id,                                        
                    CustomOnOffCluster,                                        
                ],                                                                 
                OUTPUT_CLUSTERS: [],                                               
            },  
            5: {                                                                   
                DEVICE_TYPE: zha.DeviceType.ON_OFF_SWITCH,                         
                INPUT_CLUSTERS: [                                                  
                    MultistateInput.cluster_id,                                        
                    CustomOnOffCluster,                                        
                ],                                                                 
                OUTPUT_CLUSTERS: [],                                               
            },  
        },
    }
```

javicalle, 2021-01-12:
By the way, reading documentation to try to understand all this better, I have located a documentation where it is indicated that the possible values of the **OnOff** _cluster_ are:
* 0x00 - Off
* 0x01 - On
* 0x02 - Toggle

https://docs.smartthings.com/en/latest/device-type-developers-guide/zigbee-primer.html#commands

Adminiuga, 2021-01-12:
there's slight misunderstanding. There are no "cluster values". But a cluster has attributes and those attributes may have values. 
While 0x02 is indeed a "toggle" command for the on_off cluster, the "OnOff" attribute id `0x0000` (section 3.8.2 of the [ZCL specs](https://zigbeealliance.org/wp-content/uploads/2019/12/07-5123-06-zigbee-cluster-library-specification.pdf)) is of **bool** type and bool type defined in section 2.6.2.3
[image]

javicalle, 2021-01-12:
Slight misunderstanding no, I'm totally lost (^.^u)

I appreciate your effort to explain the concepts to me. Thanks a lot.

On the other hand, this comment was supposed to go in #719

PR #715 (2021-01-19): Decoupled mode for xiaomi.aqara.CtrlNeutral
Some Aqara wall switches can work in decoupled mode, allowing the rocker to be independent from the relay.

The current implementation for these switches only exposes the relays, so for this change to be functional it is also necessary to implement the quirks of the buttons endpoints.

This PR implements the "decoupled mode" command and these quirks in `xiaomi.aqara.CtrlNeutral`.
This class supports devices: QBKG04LM/QBKG03LM/QBKG21LM/QBKG22LM.
Decoupled mode would be configurable from UI (thanks @albalaing):
[image]

Value:
* 254 (decoupled)
* 18 (c [... cut]
closed 2021-01-19 by Adminiuga