zigpy/zha-device-handlers #716: [Device Support Request] Tuya TS0601 _TZE200_kyfqmmyl triple light switch

antonbalaban, 2021-01-11:
Hi,
I'm trying to connect a 3-gang Tuya wall switch to ZHA. It connects, but doesn't show any switches, only appears as router (which is weird as it is a no-neutral switch, it shouldn't be a router). I tried modifying singleswitch.py quirk to match manufacturer and other parameters, but without success as I'm not a programmer. I don't think ZHA sees my modified quirk, I see that pycache in Tuya folder doesn't get regenerated. I'm running HA supervised in docker container. I saw this problem a couple of times here, but didn't find a solution.

Device signature:
```
[[0x59fc:zdo] ZDO request ZDOCmd.Device_annce: [0x59FC, bc:33:ac:ff:fe:17:09:86, 142]
[0x59fc] Requesting 'Node Descriptor'
Tries remaining: 2
[0x59fc] Extending timeout for 0x99 request
[0x59fc] Node Descriptor: NodeDescriptor(byte1=1, byte2=64, mac_capability_flags=142, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=0)
[0x59fc] Discovering endpoints
Tries remaining: 3
[0x59fc] Discovered endpoints: [1]
[0x59fc:1] Discovering endpoint information
Tries remaining: 3
[0x59fc:1] Discovered endpoint information: SizePrefixedSimpleDescriptor(endpoint=1, profile=260, device_type=81, device_version=0, input_clusters=[0, 10, 4, 5, 61184], output_clusters=[25])
Unknown cluster 61184
[0x59fc:1:0x0000] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=159 command_id=Command.Read_Attributes_rsp>
[0x59fc:1] Manufacturer: _TZE200_kyfqmmyl
[0x59fc:1] Model: TS0601
Checking quirks for _TZE200_kyfqmmyl TS0601 (bc:33:ac:ff:fe:17:09:86)
Considering <class 'bellows.zigbee.application.EZSPCoordinator'>
Fail because device_type mismatch on at least one endpoint
Considering <class 'zhaquirks.gledopto.soposhgu10.SoposhGU10'>
Fail because endpoint list mismatch: {11, 13} {1}
Considering <class 'zhaquirks.lutron.lzl4bwhl01remote.LutronLZL4BWHL01Remote2'>
Fail because profile_id mismatch on at least one endpoint
Considering <class 'zhaquirks.netvox.z308e3ed.Z308E3ED'>
Fail because device_type mismatch on at least one endpoint
Considering <class 'zhaquirks.osram.a19twhite.A19TunableWhite'>
Fail because endpoint list mismatch: {3} {1}
Considering <class 'zhaquirks.philips.rom001.PhilipsROM001'>
Fail because device_type mismatch on at least one endpoint
Considering <class 'zhaquirks.philips.rwl020.PhilipsRWL020'>
Fail because endpoint list mismatch: {1, 2} {1}
Considering <class 'zhaquirks.philips.rwl021.PhilipsRWL021'>
Fail because endpoint list mismatch: {1, 2} {1}
Considering <class 'zhaquirks.smartthings.multi.SmartthingsMultiPurposeSensor'>
Fail because device_type mismatch on at least one endpoint
Considering <class 'zhaquirks.smartthings.tag_v4.SmartThingsTagV4'>
Fail because device_type mismatch on at least one endpoint
Considering <class 'zhaquirks.xbee.xbee3_io.XBee3Sensor'>
Fail because endpoint list mismatch: {232, 230} {1}
Considering <class 'zhaquirks.xbee.xbee_io.XBeeSensor'>
Fail because endpoint list mismatch: {232, 230} {1}
Considering <class 'zhaquirks.xiaomi.mija.smoke.MijiaHoneywellSmokeDetectorSensor'>
Fail because device_type mismatch on at least one endpoint
device - 0x59FC:bc:33:ac:ff:fe:17:09:86 entering async_device_initialized - is_new_join: True
device - 0x59FC:bc:33:ac:ff:fe:17:09:86 has joined the ZHA zigbee network
[0x59FC](TS0601): started configuration
[0x59FC:ZDO](TS0601): 'async_configure' stage succeeded
[0x59FC:1:0x0000]: bound 'basic' cluster: Status.SUCCESS
[0x59FC:1:0x0000]: finished channel configuration
[0x59FC:1:0x0019]: bound 'ota' cluster: Status.SUCCESS
[0x59FC:1:0x0019]: finished channel configuration
[0x59FC:1:0x0000]: 'async_configure' stage succeeded
[0x59FC:1:0x0019]: 'async_configure' stage succeeded
[0x59FC](TS0601): completed configuration
[0x59FC](TS0601): stored in registry: ZhaDeviceEntry(name='_TZE200_kyfqmmyl TS0601', ieee='bc:33:ac:ff:fe:17:09:86', last_seen=1610389145.428656)
[0x59FC](TS0601): started initialization
[0x59FC:ZDO](TS0601): 'async_initialize' stage succeeded
[0x59FC:1:0x0000]: initializing channel: from_cache: False
[0x59FC:1:0x0000]: finished channel configuration
[0x59FC:1:0x0019]: initializing channel: from_cache: False
[0x59FC:1:0x0019]: finished channel configuration
[0x59FC:1:0x0000]: 'async_initialize' stage succeeded
[0x59FC:1:0x0019]: 'async_initialize' stage succeeded
[0x59FC](TS0601): power source: Mains
[0x59FC](TS0601): completed initialization
[0xC352:1:0x0b04]: async_update
[0x4CFF:1:0x0b04]: async_update]
```

Thanks!

bennibeloisi, 2021-02-03:
Hey there, i have the same problem, but with single and double switches. I also tried modifying the single switch without luck. i would also highly appreciate if there was a chance to support the tuya switches.

antonbalaban, 2021-02-03:
My device is supported by zigbee2mqtt 
https://www.zigbee2mqtt.io/devices/TS0601_switch_3_gang.html#tuya-ts0601_switch_3_gang
Right now I’m using it with z2m, but would definitely prefer ZHA. There are a lot of those TS0601 switches around, just today got a different one (1-gang), also use with zigbee2mqtt.

MattWestb, 2021-02-03:
Post "zigbee device signature" from the device card and / or log from pairing the device.
All adding of tuya device is not go without correct information in ZHA.

If you have very urgent need of support of on new device then making one PR from the instruction https://github.com/zigpy/zha-device-handlers#building-a-quirk.

closed 2021-08-12 by github-actions[bot]