zigpy/zha-device-handlers #718: [BUG] Centralite 3200 switch identified as light.

adolfomendes, 2021-01-12:
**Describe the bug**
The Centralite 3200 (4257050-RZHAC) switch is identified as On/Off Light ("device_type": "0x0100")

**To Reproduce**
Display the device.

**Expected behavior**
The device type should be Mains Power Outlet ("device_type": "0x0009")

**Screenshots**
[image]
[image]
[image]

TheJulianJES, 2021-01-16:
See https://github.com/zigpy/zha-device-handlers/issues/654#issuecomment-749165413

adolfomendes, 2021-01-17:
This is a work around, I would like a correction.

dmulcahey, 2021-01-20:
> This is a work around, I would like a correction.

There is a work around for this. If you would like a permanent change feel free to open a PR. There is an extensive guide in the readme.

closed 2021-08-12 by github-actions[bot]