zigpy/zha-device-handlers #721: [Device Support Request] Siterwell GS361A-H04 Radiator Valve with Thermostat - ivfvd7h

eugenemarko, 2021-01-13:
**Is your feature request related to a problem? Please describe.**
I've found on https://zigbee.blakadder.com/Siterwell_GS361A-H04.html that ZHA supports this model,, but seems that not.
After I've added ("_TYST11_zivfvd7h", "ivfvd7h"), after 140 line of zhaquirks/tuya/valve.py   - everything works fine.

**Describe the solution you'd like**
Please add ("_TYST11_zivfvd7h", "ivfvd7h"), into valve.py

**Device signature - this can be acquired by removing the device from ZHA and pairing it again from the add devices screen. Be sure to add the entire content of the log panel after pairing the device to a code block below this line.**
Before:
{
  "node_descriptor": "NodeDescriptor(byte1=2, byte2=64, mac_capability_flags=128, manufacturer_code=0, maximum_buffer_size=82, maximum_incoming_transfer_size=255, server_mask=11264, maximum_outgoing_transfer_size=255, descriptor_capability_field=0)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0000",
      "in_clusters": [
        "0x0000",
        "0x0003"
      ],
      "out_clusters": [
        "0x0003",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TYST11_zivfvd7h",
  "model": "ivfvd7h",
  "class": "zigpy.device.Device"
}

After:

{
  "node_descriptor": "NodeDescriptor(byte1=2, byte2=64, mac_capability_flags=128, manufacturer_code=0, maximum_buffer_size=82, maximum_incoming_transfer_size=255, server_mask=11264, maximum_outgoing_transfer_size=255, descriptor_capability_field=0)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0301",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x0201",
        "0x0204",
        "0xef00"
      ],
      "out_clusters": [
        "0x0003",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TYST11_zivfvd7h",
  "model": "ivfvd7h",
  "class": "zhaquirks.tuya.valve.SiterwellGS361"
}

**Additional context**

Host Operating System | Home Assistant OS 5.10
Update Channel | stable
Supervisor Version | 2020.12.7
Docker Version | 19.03.13

MattWestb, 2021-01-13:
Was adding it and waiting for our devs to  merging it :-)))

eugenemarko, 2021-02-18:
Updated to supervisor 2021.2.3 - already working out of the box. Thanks!

closed 2021-02-18 by eugenemarko

MattWestb, 2021-02-18:
I hope the devs is getting all new TRV types merged so its being easier adding the missing functions in the Siterwell TRVVs we  is having.