zigpy/zha-device-handlers #727: [Device Support Request] Saswell SEA801-Z01 TRV (Radiator Thermostat)

AstersikRoss, 2021-01-16:
The Saswell TRV Radiator Thermostat is missing a quirk to function correctly I believe this is similar to other Saswell devices and may just need to be mapped to another device. I have seen deConz has recently added the device [here](https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3109)

**Device Info**

`Device info
TS0601
by _TZE200_c88teujp
Zigbee info
IEEE: 60:a4:23:ff:fe:92:4a:be
Nwk: 0x4d77
Device Type: EndDevice
LQI: 167
RSSI: -73
Last Seen: 2021-01-16T12:33:00
Power Source: Battery or Unknown`

**Device Paring Log**
```
Device 0xfa6b (5c:02:72:ff:fe:c3:61:ce) joined the network
[0xfa6b:zdo] ZDO request ZDOCmd.Device_annce: [0xFA6B, 5c:02:72:ff:fe:c3:61:ce, 128]
[0xfa6b] Requesting 'Node Descriptor'
Tries remaining: 2
[0xfa6b] Extending timeout for 0xdc request
[0xfa6b] Node Descriptor: NodeDescriptor(byte1=2, byte2=64, mac_capability_flags=128, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=0)
[0xfa6b] Discovering endpoints
Tries remaining: 3
[0xfa6b] Extending timeout for 0xde request
[0xfa6b] Discovered endpoints: [1]
[0xfa6b:1] Discovering endpoint information
Tries remaining: 3
[0xfa6b] Extending timeout for 0xe0 request
[0xfa6b:1] Discovered endpoint information: SizePrefixedSimpleDescriptor(endpoint=1, profile=260, device_type=81, device_version=1, input_clusters=[0, 4, 5, 61184], output_clusters=[25, 10])
Unknown cluster 61184
[0xfa6b] Extending timeout for 0xe2 request
[0xfa6b:1:0x0000] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=226 command_id=Command.Read_Attributes_rsp>
[0xfa6b:1] Manufacturer: _TZE200_c88teujp
[0xfa6b:1] Model: TS0601
Checking quirks for _TZE200_c88teujp TS0601 (5c:02:72:ff:fe:c3:61:ce)
Considering <class 'bellows.zigbee.application.EZSPCoordinator'>
Fail because device_type mismatch on at least one endpoint
Considering <class 'zhaquirks.gledopto.soposhgu10.SoposhGU10'>
Fail because endpoint list mismatch: {11, 13} {1}
Considering <class 'zhaquirks.netvox.z308e3ed.Z308E3ED'>
Fail because device_type mismatch on at least one endpoint
Considering <class 'zhaquirks.philips.rom001.PhilipsROM001'>
Fail because device_type mismatch on at least one endpoint
Considering <class 'zhaquirks.philips.rwl020.PhilipsRWL020'>
Fail because endpoint list mismatch: {1, 2} {1}
Considering <class 'zhaquirks.philips.rwl021.PhilipsRWL021'>
Fail because endpoint list mismatch: {1, 2} {1}
Considering <class 'zhaquirks.smartthings.multi.SmartthingsMultiPurposeSensor'>
Fail because device_type mismatch on at least one endpoint
Considering <class 'zhaquirks.smartthings.tag_v4.SmartThingsTagV4'>
Fail because device_type mismatch on at least one endpoint
Considering <class 'zhaquirks.xbee.xbee3_io.XBee3Sensor'>
Fail because endpoint list mismatch: {232, 230} {1}
Considering <class 'zhaquirks.xbee.xbee_io.XBeeSensor'>
Fail because endpoint list mismatch: {232, 230} {1}
device - 0xFA6B:5c:02:72:ff:fe:c3:61:ce entering async_device_initialized - is_new_join: True
device - 0xFA6B:5c:02:72:ff:fe:c3:61:ce has joined the ZHA zigbee network
[0xFA6B](TS0601): started configuration
[0xFA6B:ZDO](TS0601): 'async_configure' stage succeeded
[0xfa6b] Extending timeout for 0xe4 request
[0xfa6b] Extending timeout for 0xe6 request
[0xFA6B:1:0x0000]: bound 'basic' cluster: Status.SUCCESS
[0xFA6B:1:0x0000]: finished channel configuration
[0xFA6B:1:0x0019]: bound 'ota' cluster: Status.SUCCESS
[0xFA6B:1:0x0019]: finished channel configuration
[0xFA6B:1:0x0000]: 'async_configure' stage succeeded
[0xFA6B:1:0x0019]: 'async_configure' stage succeeded
[0xFA6B](TS0601): completed configuration
[0xFA6B](TS0601): stored in registry: ZhaDeviceEntry(name='_TZE200_c88teujp TS0601', ieee='5c:02:72:ff:fe:c3:61:ce', last_seen=1610808792.066362)
[0xFA6B](TS0601): started initialization
[0xFA6B:ZDO](TS0601): 'async_initialize' stage succeeded
[0xFA6B:1:0x0000]: initializing channel: from_cache: False
[0xFA6B:1:0x0000]: finished channel configuration
[0xFA6B:1:0x0019]: initializing channel: from_cache: False
[0xFA6B:1:0x0019]: finished channel configuration
[0xFA6B:1:0x0000]: 'async_initialize' stage succeeded
[0xFA6B:1:0x0019]: 'async_initialize' stage succeeded
[0xFA6B](TS0601): power source: Battery or Unknown
[0xFA6B](TS0601): completed initialization
[0xfa6b:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=14 command_id=1>
[0xfa6b:1:0xef00] Unknown cluster-specific command 1
[0xfa6b:1:0xef00] ZCL request 0x0001: b'\x0f\xff\x00'
[0xfa6b:1:0xef00] No handler for cluster command 1

```

AstersikRoss, 2021-01-16:
I think this may have been added in dev branch

vabene1111, 2021-01-17:
i would really love to see support for this device. I recently bought two of them and they are very well priced compared to other thermometers but i cannot get them working with ZHA (the device is discoverd without any problems but no entity is shown)

AstersikRoss, 2021-01-17:
I've tried editing the valve.py and adding the relevant line but on re-adding the device there are still no 'Entities' shown under the device.

AstersikRoss, 2021-01-21:
I'm closing this out as it appears this is being worked on under:

Add Saswell 88teujp thermostat #576

closed 2021-01-21 by AstersikRoss