zigpy/zha-device-handlers #729: [Device Support Request] TS0042 by _TZ3400_keyjhapk

DavidFFerreira, 2021-01-18:
**Describe the solution you'd like**
I have a Wireless Switch 2 GANG from Tuya that advertise with Tuya Hub that Have suporte for Short Tap, Double Tap and Long Press
manufacturer": "_TZ3400_keyjhapk",
model": "TS0042",

**Device signature - this can be acquired by removing the device from ZHA and pairing it again from the add devices screen. Be 
>
 {
  "node_descriptor": "NodeDescriptor(byte1=2, byte2=64, mac_capability_flags=128, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=0)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0000",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0006"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "2": {
      "profile_id": 260,
      "device_type": "0x0000",
      "in_clusters": [
        "0x0001",
        "0x0006"
      ],
      "out_clusters": []
    }
  },
  "manufacturer": "_TZ3400_keyjhapk",
  "model": "TS0042",
  "class": "zigpy.device.Device"
}

> 

**Additional context**
[image]
[image]
[image]
[image]

DavidFFerreira, 2021-01-19:
@johanhenselmans @Adminiuga can u help with that?

DavidFFerreira, 2021-01-19:
I've merge the code in https://github.com/zigpy/zha-device-handlers/blob/b12ebc37790af34253fce4a0c549d05a8ffb995c/zhaquirks/tuya/ts0042.py#L25 into my Home Assistant
now in event listener I finally receive the shortcick, but Long press or double click never appears....affter review the code, the double click and long press are not into the code for BenexmartRemote0042 class

DavidFFerreira, 2021-01-19:
@JbPasquier  here

Adminiuga, 2021-01-20:
it is in the dev tree https://github.com/zigpy/zha-device-handlers/blob/d063c1f5c8721eb33b0ebd508e225a83b7df1b42/zhaquirks/tuya/ts0042.py#L98. will be in next release

closed 2021-01-20 by Adminiuga