zigpy/zha-device-handlers #730: [Device Support Request] TS0601 by _TZE200_dfxkcots

achaulin, 2021-01-19:
I have a dimmer WDM-1ZBA-EU that have no objects after pairing. It's integrated in Z2M, but not in ZHA. (https://zigbee.blakadder.com/Earda_EDM-1ZAB-EU.html)

**Device info:**

TS0601
by _TZE200_dfxkcots
Zigbee info
IEEE: bc:33:ac:ff:fe:5a:8e:53
Nwk: 0x571a
Device Type: Router
LQI: 144
RSSI: -64
Last Seen: 2021-01-19T11:58:22
Power Source: Mains

**Zigbee device signature**

{
  "node_descriptor": "NodeDescriptor(byte1=1, byte2=64, mac_capability_flags=142, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=0)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0051",
      "in_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0xef00"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZE200_dfxkcots",
  "model": "TS0601",
  "class": "zigpy.device.Device"
}

**Pairing log**

Sent 'mgmt_permit_joining_req' to 58:8e:81:ff:fe:ba:41:82: [<Status.SUCCESS: 0>]
[0x3a87:1:0x0020] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=64 command_id=0>
[0x3a87:1:0x0020] Data remains after deserializing ZCL frame
[0x3a87:1:0x0020] ZCL request 0x0000: []
[0x3a87:1:0x0020] No handler for cluster command 0
[0x3A87:1:0x0020]: Received 64 tsn command 'checkin': []
[0x3a87] Extending timeout for 0x40 request
[0x3a87:1:0x0020] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=64 command_id=Command.Default_Response>
[0x3A87:1:0x0020]: executed 'checkin_response' command with args: '(True, 8)' kwargs: '{'tsn': 64}' result: [0, <Status.SUCCESS: 0>]
[0x3a87] Extending timeout for 0x8c request
[0x2f87:zdo] ZDO request ZDOCmd.Node_Desc_req: [0x0000]
[0x2f87:zdo] Unsupported ZDO request:ZDOCmd.Node_Desc_req
[0x3a87:1:0x0020] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=140 command_id=Command.Default_Response>
[0x3A87:1:0x0020]: executed 'set_long_poll_interval' command with args: '(24,)' kwargs: '{}' result: [2, <Status.SUCCESS: 0>]
Device 0xda2e (bc:33:ac:ff:fe:5a:8e:53) joined the network
Device bc:33:ac:ff:fe:5a:8e:53 changed id (0x2f87 => 0xda2e)
Checking quirks for _TZE200_dfxkcots TS0601 (bc:33:ac:ff:fe:5a:8e:53)
Considering <class 'bellows.zigbee.application.EZSPCoordinator'>
Fail because device_type mismatch on at least one endpoint
Considering <class 'zhaquirks.gledopto.soposhgu10.SoposhGU10'>
Fail because endpoint list mismatch: {11, 13} {1}
Considering <class 'zhaquirks.netvox.z308e3ed.Z308E3ED'>
Fail because device_type mismatch on at least one endpoint
Considering <class 'zhaquirks.philips.rom001.PhilipsROM001'>
Fail because device_type mismatch on at least one endpoint
Considering <class 'zhaquirks.philips.rwl020.PhilipsRWL020'>
Fail because endpoint list mismatch: {1, 2} {1}
Considering <class 'zhaquirks.philips.rwl021.PhilipsRWL021'>
[... 148 lines cut ...]
[0xda2e:1:0xef00] No handler for cluster command 1
[0xda2e:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=11 command_id=1>
[0xda2e:1:0xef00] Unknown cluster-specific command 1
[0xda2e:1:0xef00] ZCL request 0x0001: b'\x00\x01\x01\x01\x00\x01\x01'
[0xda2e:1:0xef00] No handler for cluster command 1
[0xda2e:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=11 command_id=1>
[0xda2e:1:0xef00] Unknown cluster-specific command 1
[0xda2e:1:0xef00] ZCL request 0x0001: b'\x00\x01\x01\x01\x00\x01\x01'
[0xda2e:1:0xef00] No handler for cluster command 1
[0xda2e:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=11 command_id=1>
[0xda2e:1:0xef00] Unknown cluster-specific command 1
[0xda2e:1:0xef00] ZCL request 0x0001: b'\x00\x01\x01\x01\x00\x01\x01'
[0xda2e:1:0xef00] No handler for cluster command 1
[0xda2e:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=12 command_id=1>
[0xda2e:1:0xef00] Unknown cluster-specific command 1
[0xda2e:1:0xef00] ZCL request 0x0001: b'\x00\x02\x02\x02\x00\x04\x00\x00\x01\xf4'
[0xda2e:1:0xef00] No handler for cluster command 1
[0xda2e:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=13 command_id=2>
[0xda2e:1:0xef00] Unknown cluster-specific command 2
[0xda2e:1:0xef00] ZCL request 0x0002: b'\x00\x07\x03\x02\x00\x04\x00\x00\x00\n'
[0xda2e:1:0xef00] No handler for cluster command 2
[0xda2e:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=14 command_id=36>
[0xda2e:1:0xef00] Unknown cluster-specific command 36
[0xda2e:1:0xef00] ZCL request 0x0024: b'\x00\x08'
[0xda2e:1:0xef00] No handler for cluster command 36

fantomnotabene, 2021-01-23:
I'm on the `z2m` right now cuz this device is unsupported in ZHA

WiebKastanje, 2021-02-07:
I have exactly the same device (dimmer) which results in having no objects after pairing with ZHA (in Home Assistant).

jfisbein, 2021-03-29:
I have this device with the same problem:
[image]

markiceman, 2021-03-29:
I have similat problem with lighting switch
[image]

davefrooney, 2021-04-13:
I have the same device and have the same issue. The device only appears as a router and has no entitites. Let me know if there is anything I can do to contribute

maxwell-01, 2021-04-28:
Same issue for me - no entities. As others have said, works perfectly with zigbee2mqtt.

Looks like support is being worked on - how do I subscribe to updates when it makes it in?

davefrooney, 2021-04-28:
It has already been merged and released. We just need to wait for home assistant to integrate the new version of zigpy

maxwell-01, 2021-04-29:
Exciting stuff!

davefrooney, 2021-04-29:
Confirmed working with beta release of HA. Thanks for all the work

WiebKastanje, 2021-04-29:
Looking forward to the new release of HA. Very excited to be able to control this dimmer.
Thanks to all the people who worked on this to get it done.

aguilaair, 2021-05-13:
Hey!

Was testing this out and I may have discovered an issue with the implementation. If you change the brightness through HA, using the light.turn_on service to anything that's not 0, the light will not actually turn on. Although, it does appears to be on in HA, the switch is actually off. To actually turn on the light you need to call light.turn_on at brightness 0 again and then it will turn on with the brightness you had set previously.

Not a huge issue as I can just call the service twice, but it's not ideal.

achaulin, 2021-05-20:
And one more issue:
When you change brightness by dimmer, on several clicks, the brightness in attributes of device changes by 10 points, and after few seconds changes to another value, then to correct value (ex. 12 -> 22 -> 102 ->124)
The delay between dimmer turn and correct brightness in HA can reach up to 5 seconds.

aguilaair, 2021-07-11:
No updates on the aforementioned issues? Would love to buy more of these switches but this is making me look for alternatives...

aguilaair, 2022-01-07:
Still an issue

PR #1489 (2022-04-19): Implementations for `NO_MANUFACTURER_ID` dimmers
This PR adds device definitions for dimmers that require send commands with the `NO_MANUFACTURER_ID` (or what is the same `manufacturer_specific=False`).

A few comments:
* `tuya/mcu/TuyaLevelControl` now handles `move_to_level_with_on_off` command, sending an `on_off` command before the `current_level` command.
* the `TuyaLevelControlManufCluster` handles DPs up to 3 gangs.
* the new `NoManufacturerCluster` will force the `NO_MANUFACTURER_ID` in commands. This is needed for recent dimmers
* at some point I wanted to migrate the remaining classe [... cut]
closed 2022-04-19 by dmulcahey