zigpy/zha-device-handlers #731: [BUG]Moes BHT-002GCLZB thermostat doesn't respect temperature increments/deferments of 0.5 degree

haralake, 2021-01-19:
**Describe the bug**
I have installed the Moes BHT-002GCLZB thermostat via zha integration.
The thermostat functions are working properly, but there is a problem, the thermostat does not respect the temperature changes by increasing or decreasing by half a unit (step 0.5).

If I increase the desired temperature from the HA lovelace by half a unit, for example 18.5 from 18, I get 18 degrees on the thermostat (instead of the 18.5 I chose). If I enter the settings of the thermostat card and increase the desired temperature for example from 18 to 18.5 I see that after a few seconds the temperature returns to 18. It is obvious that this implementation in ZHA integration does not respect temperature changes with half-degree fluctuations.

The thermostat is advertised to provide such a feature and you can change the desired temperature by half a degree from the device itself. If the temperature of the thermostat is 18 degrees and I change the desired temperature to 18.5, the thermostat does not give a command to the burner and in HA the temperature value is not renewed.

Is there a way to fix this issue and make this thermostat respect the temperature increments/deferments of 0.5 degree?

jmoransalama, 2021-01-26:
I have the same thermostat and I just noticed the same behaviour as you describe. I was able to change by half a unit in previous versions, so I guess something was changed in the code.

Might be related to:
https://github.com/zigpy/zha-device-handlers/pull/538#issuecomment-730664185

jmoransalama, 2021-01-26:
I'm not an expert, but I believe the problem might be caused by this line:
https://github.com/zigpy/zha-device-handlers/blob/660a1f4decf78a9fb648353e0b5feedc469d3e6a/zhaquirks/tuya/electric_heating.py#L93

I'll try to remove the round function to see if it fixes the issue.

jmoransalama, 2021-01-26:
Nevermind, I removed the round function but nothing changed.

I enabled logging on HA and it seems that the thermostat is only sending whole units without decimals for the target temperature (Thermostat is manually set to target temp of 20.5 º C):
`2021-01-26 13:52:49 DEBUG (MainThread) [zhaquirks.tuya] [0xd829:1:0xef00] Received value [0, 0, 0, 20] for attribute 0x0210 (command 0x0001)`

Trying to change the temperature from ha, seems to send the proper command to the thermostat:
```
2021-01-26 13:54:12 DEBUG (MainThread) [zhaquirks.tuya] [0xd829:1:0x0201] Mapping standard occupied_heating_setpoint (0x0012) with value 2050 to custom {528: 20.5}
2021-01-26 13:54:12 DEBUG (MainThread) [zhaquirks.tuya] [0xd829:1:0xef00] Received value [0, 0, 0, 20] for attribute 0x0210 (command 0x0001)
```
But as you can see the thermostat sends back the value with a whole number, without decimals.

So I think that the problem might be related to the thermostat itself and not ZHA. Let's see if someone who knows more about the code can confirm this.

haralake, 2021-01-26:
Thanks @jmoransalama for your effort.  This is a very strange behavior if you consider that the thermostat has the ability to change the target temperature by half a unit from the device itself. This feature is also mentioned in the thermostat description.

closed 2021-08-12 by github-actions[bot]