zigpy/zha-device-handlers #732: [BUG] Hue motion sensor SML001 (indoor) & SML002 (outdoor) not correctly reported

Jdruwe, 2021-01-20:
**Describe the bug**

My Hue motion sensors which I have been using for 3+ months keep reporting a battery level of 100% which much be incorrect, right?

**To Reproduce**

[image]

[image]

**Expected behavior**

A battery level to drop over time

**Logs**

When I ask for the battery_percentage_remaining attribute from the PowerConfiguration cluster I get a value of 200. I have no idea which attribute home assistant uses behind the scenes to create the battery_level sensor

```
DEBUG (MainThread) [homeassistant.components.zha.entity] light.diningroom_ceiling_2: polling current state
DEBUG (MainThread) [zigpy_deconz.zigbee.application] Sending Zigbee request with tsn 137 under 138 request id, data: b'0089000000'
DEBUG (MainThread) [zigpy_deconz.api] Command Command.aps_data_request (20, 138, 0, <DeconzAddressEndpoint address_mode=2 address=0xBDB7 endpoint=11>, 260, 6, 1, b'\x00\x89\x00\x00\x00', 2, 0)
DEBUG (MainThread) [zigpy_deconz.api] APS data request response: [2, <DeviceState.APSDE_DATA_REQUEST_SLOTS_AVAILABLE|2: 34>, 138]
DEBUG (MainThread) [zigpy_deconz.api] Device state changed response: [<DeviceState.128|APSDE_DATA_REQUEST_SLOTS_AVAILABLE|APSDE_DATA_CONFIRM|2: 166>, 0]
DEBUG (MainThread) [zigpy_deconz.api] Command Command.aps_data_confirm (0,)
DEBUG (MainThread) [zigpy_deconz.api] APS data confirm response for request with id 138: 00
DEBUG (MainThread) [zigpy_deconz.api] Request id: 0x8a 'aps_data_confirm' for <DeconzAddressEndpoint address_mode=ADDRESS_MODE.NWK address=0xbdb7 endpoint=11>, status: 0x00
DEBUG (MainThread) [zigpy.device] [0x4e28] Extending timeout for 0x8b request
DEBUG (MainThread) [zigpy_deconz.zigbee.application] Sending Zigbee request with tsn 139 under 140 request id, data: b'008b002100'
DEBUG (MainThread) [zigpy_deconz.api] Command Command.aps_data_request (20, 140, 0, <DeconzAddressEndpoint address_mode=2 address=0x4E28 endpoint=2>, 260, 1, 1, b'\x00\x8b\x00!\x00', 2, 0)
DEBUG (MainThread) [zigpy_deconz.api] APS data request response: [2, <DeviceState.APSDE_DATA_REQUEST_SLOTS_AVAILABLE|2: 34>, 140]
DEBUG (MainThread) [zigpy_deconz.api] Device state changed response: [<DeviceState.128|APSDE_DATA_REQUEST_SLOTS_AVAILABLE|APSDE_DATA_CONFIRM|2: 166>, 0]
DEBUG (MainThread) [zigpy_deconz.api] Command Command.aps_data_confirm (0,)
DEBUG (MainThread) [zigpy_deconz.api] APS data confirm response for request with id 140: 00
DEBUG (MainThread) [zigpy_deconz.api] Request id: 0x8c 'aps_data_confirm' for <DeconzAddressEndpoint address_mode=ADDRESS_MODE.NWK address=0x4e28 endpoint=2>, status: 0x00
DEBUG (MainThread) [zigpy_deconz.api] Device state changed response: [<DeviceState.128|APSDE_DATA_REQUEST_SLOTS_AVAILABLE|APSDE_DATA_INDICATION|2: 170>, 0]
DEBUG (MainThread) [zigpy_deconz.api] Command Command.aps_data_indication (1, 1)
DEBUG (MainThread) [zigpy_deconz.api] APS data indication response: [31, <DeviceState.APSDE_DATA_REQUEST_SLOTS_AVAILABLE|2: 34>, <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0x0000>, 1, <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0x4e28>, 2, 260, 1, b'\x18\x8b\x01!\x00\x00 \xc8', 0, 175, 255, 8, 227, 84, 1, -55]
DEBUG (MainThread) [zigpy.zcl] [0x4e28:2:0x0001] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=139 command_id=Command.Read_Attributes_rsp>
DEBUG (MainThread) [zigpy_deconz.api] 'aps_data_indication' response from <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0x4e28>, ep: 2, profile: 0x0104, cluster_id: 0x0001, data: b'188b0121000020c8'
DEBUG (MainThread) [homeassistant.components.zha.api] Read attribute for: cluster_id: [1] cluster_type: [in] endpoint_id: [2] attribute: [33] manufacturer: [None] response: [200] failure: [{}],
DEBUG (MainThread) [zigpy_deconz.zigbee.application] Sending Zigbee request with tsn 141 under 142 request id, data: b'008d000000'
DEBUG (MainThread) [zigpy_deconz.api] Command Command.aps_data_request (20, 142, 0, <DeconzAddressEndpoint address_mode=2 address=0xBDB7 endpoint=11>, 260, 8, 1, b'\x00\x8d\x00\x00\x00', 2, 0)
DEBUG (MainThread) [zigpy_deconz.api] APS data request response: [2, <DeviceState.APSDE_DATA_REQUEST_SLOTS_AVAILABLE|2: 34>, 142]
DEBUG (MainThread) [zigpy_deconz.api] Device state changed response: [<DeviceState.128|APSDE_DATA_REQUEST_SLOTS_AVAILABLE|APSDE_DATA_CONFIRM|2: 166>, 0]
DEBUG (MainThread) [zigpy_deconz.api] Command Command.aps_data_confirm (0,)
DEBUG (MainThread) [zigpy_deconz.api] APS data confirm response for request with id 142: 00
DEBUG (MainThread) [zigpy_deconz.api] Request id: 0x8e 'aps_data_confirm' for <DeconzAddressEndpoint address_mode=ADDRESS_MODE.NWK address=0xbdb7 endpoint=11>, status: 0x00
```
Is anyone else experiencing this issue or is there a way I am not aware of to fix this?

Adminiuga, 2021-01-20:
check the battery again, in about 1 year

Jdruwe, 2021-01-20:
> check the battery again, in about 1 year

Does it really take that long for the battery to lose enough power to update state

MattWestb, 2021-01-21:
I have 2 indoor ones for over 3 years 76% and 12% with batteries and the last have periods of jumping around in all possible and not possible routers in my network. 
The 2 first year they was in my deCONZ network and the difference was beginning ther (I think its was problem with ack and pull control config that was draining on more then the other).so you need waiting little for getting them going down or putting in rechargeable batteries that is going down faster but holing longer before being compleet empty.

Adminiuga, 2021-01-21:
> Does it really take that long for the battery to lose enough power to update state?

Update state as in report a "value" -- no. Update state as in "report a different value" is going to take a while.

Jdruwe, 2021-01-21:
> > Does it really take that long for the battery to lose enough power to update state?
> 
> Update state as in report a "value" -- no. Update state as in "report a different value" is going to take a while.

Do you know which cluster attributen is used by home assistant?

MattWestb, 2021-01-21:
Its have 2 batteries so the red value / 2 = HA value.

[image]

Its the other one with 76% battery left.

Jdruwe, 2021-01-21:
Ahhhhh that's who it's reporting 200… thanks!

MattWestb, 2021-01-21:
But attribute "battery_quantity (id: 0x0033)" = None !!
 
Some IKEA controlled with one battery is reporting 2 and 100% battery so ZHA is displaying 50% without quirk :-(

But my math is saying 153 X 0 = 0  ;-)

closed 2021-08-12 by github-actions[bot]