zigpy/zha-device-handlers #733: [Help] ZHA doesn't see the quirk for TS0043 device

antonbalaban, 2021-01-21:
I got a Benexmart 3-button remote which should be supported by ts0043.py quirk, but can't make it work. The device pairs successfully, but HA sees it as three switches, not the remote. No events in zha_events from button pressing either. Here's connection log:

```
Device 0xcdeb (bc:33:ac:ff:fe:ef:6c:32) joined the network
[0xcdeb:zdo] ZDO request ZDOCmd.Device_annce: [0xCDEB, bc:33:ac:ff:fe:ef:6c:32, 128]
[0xcdeb] Requesting 'Node Descriptor'
Tries remaining: 2
[0xcdeb] Extending timeout for 0xfc request
[0xcdeb] Node Descriptor: NodeDescriptor(byte1=2, byte2=64, mac_capability_flags=128, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=0)
[0xcdeb] Discovering endpoints
Tries remaining: 3
[0xcdeb] Extending timeout for 0xfe request
[0xcdeb] Discovered endpoints: [1, 2, 3]
[0xcdeb:1] Discovering endpoint information
Tries remaining: 3
[0xcdeb] Extending timeout for 0x00 request
[0xcdeb:1] Discovered endpoint information: SizePrefixedSimpleDescriptor(endpoint=1, profile=260, device_type=0, device_version=1, input_clusters=[0, 1, 6], output_clusters=[25, 10])
[0xcdeb] Extending timeout for 0x02 request
[0xcdeb:1:0x0000] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=2 command_id=Command.Read_Attributes_rsp>
[0xcdeb:1] Manufacturer: _TZ3000_qzjcsmar
[0xcdeb:1] Model: TS0043
[0xcdeb:2] Discovering endpoint information
Tries remaining: 3
[0xcdeb] Extending timeout for 0x04 request
[0xcdeb:2] Discovered endpoint information: SizePrefixedSimpleDescriptor(endpoint=2, profile=260, device_type=0, device_version=1, input_clusters=[1, 6], output_clusters=[])
[0xcdeb:3] Discovering endpoint information
Tries remaining: 3
[0xcdeb] Extending timeout for 0x06 request
Received frame on uninitialized device bc:33:ac:ff:fe:ef:6c:32 endpoint 1 for cluster: 10
[0xcdeb:3] Discovered endpoint information: SizePrefixedSimpleDescriptor(endpoint=3, profile=260, device_type=0, device_version=1, input_clusters=[1, 6], output_clusters=[])
Checking quirks for _TZ3000_qzjcsmar TS0043 (bc:33:ac:ff:fe:ef:6c:32)
Considering <class 'zhaquirks.xiaomi.mija.smoke.MijiaHoneywellSmokeDetectorSensor'>
Fail because endpoint list mismatch: {1} {1, 2, 3}
Considering <class 'zhaquirks.xbee.xbee_io.XBeeSensor'>
Fail because endpoint list mismatch: {232, 230} {1, 2, 3}
Considering <class 'zhaquirks.xbee.xbee3_io.XBee3Sensor'>
Fail because endpoint list mismatch: {232, 230} {1, 2, 3}
Considering <class 'zhaquirks.smartthings.tag_v4.SmartThingsTagV4'>
Fail because endpoint list mismatch: {1} {1, 2, 3}
Considering <class 'zhaquirks.smartthings.multi.SmartthingsMultiPurposeSensor'>
Fail because endpoint list mismatch: {1} {1, 2, 3}
Considering <class 'zhaquirks.philips.rwl021.PhilipsRWL021'>
Fail because endpoint list mismatch: {1, 2} {1, 2, 3}
Considering <class 'zhaquirks.philips.rwl020.PhilipsRWL020'>
Fail because endpoint list mismatch: {1, 2} {1, 2, 3}
Considering <class 'zhaquirks.philips.rom001.PhilipsROM001'>
Fail because endpoint list mismatch: {1} {1, 2, 3}
Considering <class 'zhaquirks.osram.a19twhite.A19TunableWhite'>
Fail because endpoint list mismatch: {3} {1, 2, 3}
Considering <class 'zhaquirks.netvox.z308e3ed.Z308E3ED'>
Fail because endpoint list mismatch: {1} {1, 2, 3}
Considering <class 'zhaquirks.lutron.lzl4bwhl01remote.LutronLZL4BWHL01Remote2'>
Fail because endpoint list mismatch: {1} {1, 2, 3}
Considering <class 'zhaquirks.gledopto.soposhgu10.SoposhGU10'>
Fail because endpoint list mismatch: {11, 13} {1, 2, 3}
Considering <class 'bellows.zigbee.application.EZSPCoordinator'>
Fail because endpoint list mismatch: {1} {1, 2, 3}
device - 0xCDEB:bc:33:ac:ff:fe:ef:6c:32 entering async_device_initialized - is_new_join: True
device - 0xCDEB:bc:33:ac:ff:fe:ef:6c:32 has joined the ZHA zigbee network
[0xCDEB](TS0043): started configuration
[0xCDEB:ZDO](TS0043): 'async_configure' stage succeeded
[0xcdeb] Extending timeout for 0x08 request
[0xcdeb] Extending timeout for 0x0a request
[0xcdeb] Extending timeout for 0x0c request
[0xcdeb:1:0x000a] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=False disable_default_response=False> manufacturer=None tsn=65 command_id=Command.Read_Attributes>
[0xcdeb:1:0x000a] ZCL request 0x0000: [[7]]
[0xcdeb:1:0x000a] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=False disable_default_response=False> manufacturer=None tsn=65 command_id=Command.Read_Attributes>
[0xcdeb:1:0x000a] ZCL request 0x0000: [[7]]
[0xCDEB:1:0x0006]: bound 'on_off' cluster: Status.SUCCESS
[0xcdeb] Extending timeout for 0x10 request
[... 152 lines cut ...]
      "in_clusters": [
        "0x0001",
        "0x0006"
      ],
      "out_clusters": []
    },
    "3": {
      "profile_id": 260,
      "device_type": "0x0000",
      "in_clusters": [
        "0x0001",
        "0x0006"
      ],
      "out_clusters": []
    }
  },
  "manufacturer": "_TZ3000_qzjcsmar",
  "model": "TS0043",
  "class": "zigpy.device.Device"
}
```

I have the latest version of Home Assistant in supervised mode (docker container), installed latest quirks there with `pip install zha-quirks==0.0.52` Hardware is Raspberry Pi 4B USB SSD install, ConBee II. Can't figure out what I'm doing wrong.

Thanks!

Adminiuga, 2021-01-22:
> installed latest quirks there with pip install zha-quirks==0.0.52

Installing quirks manually won't  help, because HA would install the version it comes with ==0.0.51

antonbalaban, 2021-01-22:
I installed quirks after updating core to 2020.1.4. Checked in Portainer - tuya folder and ts0043 quirk are present.

Adminiuga, 2021-01-22:
if you installed quirks with pip, then HA Core re-installed the quirks version that came bundled with 2021.1 release. Run pip freeze

antonbalaban, 2021-01-22:
You are right, it shows zha-quirks v 0.0.51. But this device seems to be supported in 0.0.51 as well, I copied the ts0043.py to /config/ folder, opened it in editor, it's in there:

```
class BenexmartRemote0043(CustomDevice):
    """Benexmart/Tuya 3-button remote device."""

    signature = {
        # SizePrefixedSimpleDescriptor(endpoint=1, profile=260, device_type=0, device_version=1, input_clusters=[0, 1, 6], output_clusters=[10, 25]))
        # SizePrefixedSimpleDescriptor(endpoint=2, profile=260, device_type=0, device_version=1, input_clusters=[1, 6], output_clusters=[])
        # SizePrefixedSimpleDescriptor(endpoint=3, profile=260, device_type=0, device_version=1, input_clusters=[1, 6], output_clusters=[])
        MODELS_INFO: [("_TZ3000_qzjcsmar", "TS0043")],
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_SWITCH,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    PowerConfiguration.cluster_id,
                    OnOff.cluster_id,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id]
```

But ZHA doesn't pick it up.

Adminiuga, 2021-01-22:
it is not supported in 0.0.51, because it does not match the model

> `MODELS_INFO: [("_TZ3000_qzjcsmar", "TS0043")],`

check the readme on how to override the quirks to test the new version

Adminiuga, 2021-01-22:
or install the dev tree of HA Core

antonbalaban, 2021-01-22:
Thank you Alexei, it's working now! Quirks don't survive HA restart, but paired device works anyway.

closed 2021-01-22 by antonbalaban