zigpy/zha-device-handlers #734: [BUG][InDEV] Wyze Lock quirk doesn't launch to handle_message

Fmstrat, 2021-01-21:
**Describe the bug**
When using the Wyze Lock quirk that is a part of the `dm/wyze-lock` branch, the lock is recognized and sensors are added to Home Assistant, however the `handle_message` call is never run on manual events.

**To Reproduce**
Steps to reproduce the behavior:
1. Follow the "guide" at: https://community.home-assistant.io/t/wyze-lock-zigbee-home-assistant-hassio-hassos-zha-integration-working-as-expected-remove-cloud-control/211884/67 to install the quirk
2. Restart HA and watch the binary sensor for open/close get added
3. Manually unlock and open the door, notice that the log entries do not fire.

**Expected behavior**
Log entries should fire to allow the user to find the entries needed for `lock.py`

**Additional context**
No matter what I do, including forcing some `_LOGGER` entries, nothing is ever hit except the `__init__` call.

To be specific, if I add this (see last line), nothing happens:
```
class WyzeCluster(CustomCluster, Basic):
    """Wyze manufacturer specific cluster implementation."""

    cluster_id = WYZE_CLUSTER_ID
    ep_attribute = "wyze_lock_cluster"
    attributes = {}   
    server_commands = {}
    client_commands = {}

    def handle_message(self, hdr, args):
        _LOGGER.warning("MINE: WyzeCluster handle_message")
```
Also, if I add some logger statements to the `init` functions for both `DoorLockCluster` and `MotionCluster` those messages do show up in the logs when I start the container.

I have also tried clearing the `__pycache__` folder in the quirk, and it does rebuild on restart.

I have confirmed that the `handle_message` routine in the quirk seems to run on the hour as others have reported in the linked thread, but not on an actual event such as manually unlocking.

I have also confirmed that `handle_message` in `/usr/local/lib/python3.8/site-packages/zigpy/zcl/__init__.py` does appear to run when I lock/unlock from Home Assistant, but does not run (nor does the quirk's version) for manual events or to show an open door.

Fmstrat, 2021-01-22:
I am also guessing that the quirk is applied appropriately since it shows up in the device screen as well as having 8 input clusters vs the original 7:
```
{
  "node_descriptor": "NodeDescriptor(byte1=2, byte2=64, mac_capability_flags=128, manufacturer_code=4698, maximum_buffer_size=82, maximum_incoming_transfer_size=512, server_mask=11264, maximum_outgoing_transfer_size=512, descriptor_capability_field=0)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x000a",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x0020",
        "0x0101",
        "0x0500",
        "0x0b05",
        "0xfc00"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019",
        "0xfc00"
      ]
    }
  },
  "manufacturer": "Yunding",
  "model": "Ford",
  "class": "zhaquirks.wyze.lock.WyzeLock"
}
```

mike1627, 2021-02-13:
Have you had any progress with this?  I recently had my lock replaced under warranty and need to get the new values for the quirk.  This worked a long time ago for my old lock however it's no longer giving me the expected debug data in the logs.  I'm starting to suspect that either a newer build of Home Assistant or python is preventing something in the quirk from working properly.

theupmachine, 2021-02-13:
I was having the same problem, but managed to get this working again with a small change. Hope this works for you.

```
class WyzeCluster(CustomCluster, Basic):
    """Wyze manufacturer specific cluster implementation."""

    cluster_id = WYZE_CLUSTER_ID
    ep_attribute = "wyze_lock_cluster"
    attributes = {}
    server_commands = {}
    client_commands = {}

    def handle_message(self, hdr, args, dst_addressing):
```

mike1627, 2021-02-13:
> I was having the same problem, but managed to get this working again with a small change. Hope this works for you.
>
> ```
> class WyzeCluster(CustomCluster, Basic):
>     """Wyze manufacturer specific [...]

That did the trick!  Thanks very much!

closed 2021-08-19 by github-actions[bot]