zigpy/zha-device-handlers #738: [Device Support Request] _TZE200_zivfvd7h TS0601 Qiumi/Siterwell QM-GS361A-H04 TRV

welshcoder, 2021-01-22:
**Is your feature request related to a problem? Please describe.**
I bought a Zigbee TRV off Amazon (https://www.amazon.co.uk/gp/product/B08357D6LJ/ref=ppx_yo_dt_b_asin_title_o01_s00?ie=UTF8&psc=1), marketed as a "Qiumi Smart TRV", but is labelled "GS361A-H04" by Siterwell Electronics on the device itself. It joined the network, but it did not pick up a quirk to make it work correctly in ZHA.

**Describe the solution you'd like**
I added ```("_TZE200_zivfvd7h", "TS0601"),``` to the MoesHY368 signature at line 180 in tuya/valve.py to make the TRV appear in HA. It seems to work fine - I can set the required temperature on the device, and it reports back - so you may want to consider adding this modification to that file.

**Device signature - this can be acquired by removing the device from ZHA and pairing it again from the add devices screen. Be sure to add the entire content of the log panel after pairing the device to a code block below this line.**
This is the log after I got after the change to valve.py.
```
[0xcb0b] Extending timeout for 0x06 request
[0xcb0b] Node Descriptor: NodeDescriptor(byte1=2, byte2=64, mac_capability_flags=128, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=0)
[0xcb0b] Discovering endpoints
Tries remaining: 3
[0xcb0b] Extending timeout for 0x07 request
[0xcb0b] Discovered endpoints: [1]
[0xcb0b:1] Discovering endpoint information
Tries remaining: 3
[0xcb0b] Extending timeout for 0x08 request
[0xcb0b:1] Discovered endpoint information: SizePrefixedSimpleDescriptor(endpoint=1, profile=260, device_type=81, device_version=1, input_clusters=[0, 4, 5, 61184], output_clusters=[25, 10])
Unknown cluster 61184
[0xcb0b] Extending timeout for 0x09 request
[0xcb0b:1:0x0000] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=9 command_id=Command.Read_Attributes_rsp>
[0xcb0b:1] Manufacturer: _TZE200_zivfvd7h
[0xcb0b:1] Model: TS0601
Checking quirks for _TZE200_zivfvd7h TS0601 (5c:02:72:ff:fe:c3:e3:7c)
Considering <class 'zhaquirks.tuya.valve.MoesHY368'>
Found custom device replacement for 5c:02:72:ff:fe:c3:e3:7c: <class 'zhaquirks.tuya.valve.MoesHY368'>
device - 0xCB0B:5c:02:72:ff:fe:c3:e3:7c entering async_device_initialized - is_new_join: True
device - 0xCB0B:5c:02:72:ff:fe:c3:e3:7c has joined the ZHA zigbee network
[0xCB0B](TS0601): started configuration
[0xCB0B:ZDO](TS0601): 'async_configure' stage succeeded
[0xCB0B:1:0x0201]: bound 'thermostat' cluster: Status.SUCCESS
[0xCB0B:1:0x0201]: attr reporting for '{'local_temp': (30, 900, 25), 'occupied_cooling_setpoint': (30, 900, 25), 'occupied_heating_setpoint': (30, 900, 25), 'unoccupied_cooling_setpoint': (30, 900, 25)}' on 'thermostat': ConfigureReportingResponseRecord(status=0)
[0xCB0B:1:0x0201]: attr reporting for '{'unoccupied_heating_setpoint': (30, 900, 25), 'running_mode': (30, 900, 25), 'running_state': (30, 900, 5), 'system_mode': (30, 900, 25)}' on 'thermostat': ConfigureReportingResponseRecord(status=0)
[0xCB0B:1:0x0201]: attr reporting for '{'occupancy': (30, 900, 1), 'pi_cooling_demand': (30, 900, 5), 'pi_heating_demand': (30, 900, 5)}' on 'thermostat': ConfigureReportingResponseRecord(status=0)
[0xCB0B:1:0x0201]: finished channel configuration
[0xCB0B:1:0x0001]: bound 'power' cluster: Status.SUCCESS
[0xCB0B:1:0x0001]: reporting 'battery_voltage' attr on 'power' cluster: 3600/10800/1: Result: '[ConfigureReportingResponseRecord(status=0)]'
[0xCB0B:1:0x0001]: reporting 'battery_percentage_remaining' attr on 'power' cluster: 3600/10800/1: Result: '[ConfigureReportingResponseRecord(status=0)]'
[0xCB0B:1:0x0001]: finished channel configuration
[0xcb0b] Extending timeout for 0x0a request
[0xcb0b] Extending timeout for 0x0b request
[0xCB0B:1:0x0000]: bound 'basic' cluster: Status.SUCCESS
[0xCB0B:1:0x0000]: finished channel configuration
[0xCB0B:1:0x0019]: bound 'ota' cluster: Status.SUCCESS
[0xCB0B:1:0x0019]: finished channel configuration
[0xCB0B:1:0x0201]: 'async_configure' stage succeeded
[0xCB0B:1:0x0001]: 'async_configure' stage succeeded
[0xCB0B:1:0x0000]: 'async_configure' stage succeeded
[0xCB0B:1:0x0019]: 'async_configure' stage succeeded
[0xCB0B](TS0601): completed configuration
[0xCB0B](TS0601): stored in registry: ZhaDeviceEntry(name='_TZE200_zivfvd7h TS0601', ieee='5c:02:72:ff:fe:c3:e3:7c', last_seen=1611342996.7547195)
[0xCB0B](TS0601): started initialization
[0xCB0B:ZDO](TS0601): 'async_initialize' stage succeeded
[0xCB0B:1:0x0201]: initializing channel: from_cache: False
[0xCB0B:1:0x0201]: read attributes: Success: {}. Failed: {'abs_min_heat_setpoint_limit': <Status.UNSUPPORTED_ATTRIBUTE: 134>, 'abs_max_heat_setpoint_limit': <Status.UNSUPPORTED_ATTRIBUTE: 134>, 'abs_min_cool_setpoint_limit': <Status.UNSUPPORTED_ATTRIBUTE: 134>, 'abs_max_cool_setpoint_limit': <Status.UNSUPPORTED_ATTRIBUTE: 134>}
[0xCB0B:1:0x0201]: read attributes: Success: {}. Failed: {'max_cool_setpoint_limit': <Status.UNSUPPORTED_ATTRIBUTE: 134>, 'max_heat_setpoint_limit': <Status.UNSUPPORTED_ATTRIBUTE: 134>, 'min_cool_setpoint_limit': <Status.UNSUPPORTED_ATTRIBUTE: 134>, 'min_heat_setpoint_limit': <Status.UNSUPPORTED_ATTRIBUTE: 134>}
[0xCB0B:1:0x0201]: Attribute report 'SiterwellThermostat'[ctrl_seqe_of_oper] = ControlSequenceOfOperation.Heating_Only
[0xCB0B:1:0x0201]: read attributes: Success: {'ctrl_seqe_of_oper': <ControlSequenceOfOperation.Heating_Only: 2>}. Failed: {'local_temp': <Status.UNSUPPORTED_ATTRIBUTE: 134>, 'occupancy': <Status.UNSUPPORTED_ATTRIBUTE: 134>, 'occupied_cooling_setpoint': <Status.UNSUPPORTED_ATTRIBUTE: 134>}
[0xCB0B:1:0x0201]: read attributes: Success: {}. Failed: {'occupied_heating_setpoint': <Status.UNSUPPORTED_ATTRIBUTE: 134>, 'pi_cooling_demand': <Status.UNSUPPORTED_ATTRIBUTE: 134>, 'pi_heating_demand': <Status.UNSUPPORTED_ATTRIBUTE: 134>, 'running_mode': <Status.UNSUPPORTED_ATTRIBUTE: 134>}
[0xCB0B:1:0x0201]: read attributes: Success: {}. Failed: {'running_state': <Status.UNSUPPORTED_ATTRIBUTE: 134>, 'system_mode': <Status.UNSUPPORTED_ATTRIBUTE: 134>, 'unoccupied_heating_setpoint': <Status.UNSUPPORTED_ATTRIBUTE: 134>, 'unoccupied_cooling_setpoint': <Status.UNSUPPORTED_ATTRIBUTE: 134>}
[0xCB0B:1:0x0201]: finished channel configuration
[0xCB0B:1:0x0001]: initializing channel: from_cache: False
[0xCB0B:1:0x0001]: finished channel configuration
[0xCB0B:1:0x0000]: initializing channel: from_cache: False
[0xCB0B:1:0x0000]: finished channel configuration
[0xCB0B:1:0x0019]: initializing channel: from_cache: False
[0xCB0B:1:0x0019]: finished channel configuration
[0xCB0B:1:0x0201]: 'async_initialize' stage succeeded
[0xCB0B:1:0x0001]: 'async_initialize' stage succeeded
[0xCB0B:1:0x0000]: 'async_initialize' stage succeeded
[0xCB0B:1:0x0019]: 'async_initialize' stage succeeded
[0xCB0B](TS0601): power source: Battery or Unknown
[0xCB0B](TS0601): completed initialization
[0xcb0b:1:0x000a] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=False disable_default_response=False> manufacturer=None tsn=107 command_id=Command.Read_Attributes>
[0xcb0b:1:0x000a] ZCL request 0x0000: [[7]]
```

**Additional context**
Home Assistant 2021.1.4, FreeBSD 12.2-RELEASE, Python 3.9.1, Zzh! CC2652 Zigbee USB Stick.

MattWestb, 2021-01-23:
I have making on pull request with the change and its going in the dev branch then its being merged.

Thanks for helping adding unsupported device in our systems !!!

welshcoder, 2021-01-23:
That's great, thank you very much!

As aside, I've been looking in my HA logs, and noticed that I get the following warning every now and again:
```WARNING (MainThread) [zigpy.zcl] [0xcb0b:1:0xef00] Unknown cluster-specific command 36```

After digging around, this appears to be a time synchronisation request by the TRV, as detailed here: https://developer.tuya.com/en/docs/iot/device-development/embedded-software-development/mcu-development-access/zigbee-general-solution/tuya-zigbee-module-uart-communication-protocol?id=K9ear5khsqoty#title-12-Time%20synchronization. As I understand it, since the quirk only handles commands 0x00, 0x01 and 0x02 in cluster 0xEF00, this command is unhandled.

It doesn't seem to affect the operation, since I'm not intending to use the TRV's built-in automatic mode, but it might be something to consider investigating further in the future. If I can help with that, let me know.

MattWestb, 2021-01-23:
I have the same TRV but the other model and all commands is not implanted and the dev dont have any TRV so not easy getting it solved. I dont have the time sync problem but i only using the manual mode from AH and changing the temperature.

Its one or 2 more devs starting implanting TRVs with more functions and if there is being merged its easier using there functions and implanting them.
Look on the other not merged PR if you is finding how is done and trying it on yours and its possible you get it working.

welshcoder, 2021-01-23:
Thank you for pointing me to the PR - I didn't spot that it was already being fixed!

Thanks for your efforts on this.

MattWestb, 2021-01-23:
Then look on Z2M somewhere I have reading disabling the `Time.cluster_id,` then the devices dont support the time function for not getting the error in the log.

closed 2021-01-23 by Adminiuga