zigpy/zha-device-handlers #741: [Device Support Request] Pre request IKEA KNYCKLAN receiver: E1842 and remote: E1841

MattWestb, 2021-01-23:
**Is your feature request related to a problem? Please describe.**
In the scandinavian countries you need shutting the water of the washing machine then its not being used. Ikea have doing one Zigbee valve and remote for do that but its not being sold on all market.
**Describe the solution you'd like**
This is one "pre request" as i dont have the device (wrong market but i can thinking using it for my balcony in the summer for watering my plants) but its being requested in deCONZ (and no interest from there devs) and one Z2M user have adding support for it in Z2M. 
The user @frassefraze have making good screenshots in his  [deCONZ](https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3824) request and the [Z2M PR](https://github.com/Koenkk/zigbee-herdsman-converters/pull/2130) have all info in inside.
If some one like to doing on quirk for the 2 devices i think its good then users can adding it without problems then they is installing the hardware in there home.  
  
**Device signature - this can be acquired by removing the device from ZHA and pairing it again from the add devices screen. Be sure to add the entire content of the log panel after pairing the device to a code block below this line.**  
I dont have the devices bit i think its enough information in the 2 linked issues above for implanting both.  
  
**Additional context**
One "Zigbee "anti drop" device :-))
[[image]](https://user-images.githubusercontent.com/20070414/105564506-c167ab00-5d22-11eb-9fcf-421447f5407b.jpg)

Adminiuga, 2021-01-23:
Ikea makes ZCL complaint devices, so they should work out of the box

closed 2021-01-23 by Adminiuga

MattWestb, 2021-01-23:
Only 2 things that i think is very likely missing: battery is very likely not OK (then IKEA is reporting battery % OK but 2 batteries => 1/2 % in ZHA) and ZHA events. 
But the binding and sterling should working out of the box.

So better waiting then some have the devices in their hands.