zigpy/zha-device-handlers #749: [BUG] OSRAM Plug missing electrical measurement with version 0.0.52

pdecat, 2021-01-26:
**Describe the bug**

Since version 0.0.52, all of my 8 OSRAM plugs are no longer reporting electrical measurement via sensor entities.
Downgrading to version 0.0.51 resolves the issue.

This is with Home Assistant 2021.1.5 that comes with zigpy 0.30.0.

**To Reproduce**

1. Use _Custom deps deployment_ to install `zha-quirks==0.0.52` in /usr/share/hassio/homeassistant/deps/:
    ```
    pypi:
      - zha-quirks==0.0.52
    apk: []
    ```
2. Restart Home Assistant
3. Navigate to OSRAM Plug device page
4. See missing sensor.

**Expected behavior**

Sensor of type "electrical_measurement" should appear for `osram_plug_01` devices.

**Screenshots**

With 0.0.52:
[image]

[image]

With 0.0.51:

[image]

[image]

**Additional context**

Before 0.0.52, these plugs used to only report 400W when on (a friend of mine also reported he seen 40W with some low power loads but I never witnessed it despite trying it with low power LEDs lights) and 28000W when off (another weird behavior).

I hoped this was improved by https://github.com/zigpy/zha-device-handlers/pull/723 that adds a specific quirk for this device which is why I was eager to try 0.0.52, but it seems worse.

Maybe a firmware version issue?

Adminiuga, 2021-01-26:
#723 removed the electrical measurement because device does not report valid data on it.

dmulcahey, 2021-01-26:
So the question becomes who is right? Do they report valid data or not?

pdecat, 2021-01-26:
> #723 removed the electrical measurement because device does not report valid data on it.

Ok, I really misunderstood the PR description and did not look at the code change :upside_down_face: 

> So the question becomes who is right? Do they report valid data or not?

Well, data is there, not much accurate, but there.

Is there any log I could capture to help here?

pdecat, 2021-01-26:
When off:

[image]

When on:

[image]

pdecat, 2021-01-26:
Not sure if this helps, here are some logs from querying `ElectricalMeasurement` cluster.

* `active_power`:

[image]

```
2021-01-26 17:54:24 DEBUG (MainThread) [zigpy_zigate.zigbee.application] request (0x4492, 260, 2820, 1, 3, 62, b'\x00>\x00\x0b\x05', True, False)
2021-01-26 17:54:24 DEBUG (MainThread) [zigpy_zigate.uart] Send: 0x0530 b'02449201030b040104000005003e000b05'
2021-01-26 17:54:24 DEBUG (MainThread) [zigpy_zigate.uart] Frame to send: b'\x050\x00\x11\xcd\x02D\x92\x01\x03\x0b\x04\x01\x04\x00\x00\x05\x00>\x00\x0b\x05'
2021-01-26 17:54:24 DEBUG (MainThread) [zigpy_zigate.uart] Frame escaped: b'\x02\x150\x02\x10\x11\xcd\x02\x12D\x92\x02\x11\x02\x13\x02\x1b\x02\x14\x02\x11\x02\x14\x02\x10\x02\x10\x02\x15\x02\x10>\x02\x10\x02\x1b\x02\x15'
2021-01-26 17:54:24 DEBUG (MainThread) [zigpy_zigate.uart] Frame received: 800000076600d40530000000
2021-01-26 17:54:24 DEBUG (MainThread) [zigpy_zigate.api] data received 0x8000 b'00d405300000' LQI:0
2021-01-26 17:54:24 DEBUG (MainThread) [zigpy_zigate.zigbee.application] zigate_callback_handler [0, 212, 1328, b'\x00\x00']
2021-01-26 17:54:24 DEBUG (MainThread) [zigpy_zigate.uart] Frame received: 80020017d00001040b040301024492020000183e010b050029606d96
2021-01-26 17:54:24 DEBUG (MainThread) [zigpy_zigate.api] data received 0x8002 b'0001040b040301024492020000183e010b050029606d' LQI:150
2021-01-26 17:54:24 DEBUG (MainThread) [zigpy_zigate.zigbee.application] zigate_callback_handler [0, 260, 2820, 3, 1, <Address address_mode=ADDRESS_MODE.NWK address=0x4492>, <Address address_mode=ADDRESS_MODE.NWK address=0x0000>, b'\x18>\x01\x0b\x05\x00)`m']
2021-01-26 17:54:24 DEBUG (MainThread) [zigpy.zcl] [0x4492:3:0x0b04] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=62 command_id=Command.Read_Attributes_rsp>
2021-01-26 17:54:24 DEBUG (MainThread) [homeassistant.components.zha.api] Read attribute for: cluster_id: [2820] cluster_type: [in] endpoint_id: [3] attribute: [1291] manufacturer: [None] response: [28000] failure: [{}],
```

* `measurement_type`: `bitmap32.8`:

[image]

```
2021-01-26 17:52:28 DEBUG (MainThread) [zigpy_zigate.zigbee.application] request (0x4492, 260, 2820, 1, 3, 29, b'\x00\x1d\x00\x00\x00', True, False)
2021-01-26 17:52:28 DEBUG (MainThread) [zigpy_zigate.uart] Send: 0x0530 b'02449201030b040104000005001d000000'
2021-01-26 17:52:28 DEBUG (MainThread) [zigpy_zigate.uart] Frame to send: b'\x050\x00\x11\xe0\x02D\x92\x01\x03\x0b\x04\x01\x04\x00\x00\x05\x00\x1d\x00\x00\x00'
2021-01-26 17:52:28 DEBUG (MainThread) [zigpy_zigate.uart] Frame escaped: b'\x02\x150\x02\x10\x11\xe0\x02\x12D\x92\x02\x11\x02\x13\x02\x1b\x02\x14\x02\x11\x02\x14\x02\x10\x02\x10\x02\x15\x02\x10\x1d\x02\x10\x02\x10\x02\x10'
2021-01-26 17:52:28 DEBUG (MainThread) [zigpy_zigate.uart] Frame received: 800000071d00af0530000000
2021-01-26 17:52:28 DEBUG (MainThread) [zigpy_zigate.api] data received 0x8000 b'00af05300000' LQI:0
2021-01-26 17:52:28 DEBUG (MainThread) [zigpy_zigate.zigbee.application] zigate_callback_handler [0, 175, 1328, b'\x00\x00']
2021-01-26 17:52:28 DEBUG (MainThread) [zigpy_zigate.uart] Frame received: 80020019c40001040b040301024492020000181d010000001b0800000096
2021-01-26 17:52:28 DEBUG (MainThread) [zigpy_zigate.api] data received 0x8002 b'0001040b040301024492020000181d010000001b08000000' LQI:150
2021-01-26 17:52:28 DEBUG (MainThread) [zigpy_zigate.zigbee.application] zigate_callback_handler [0, 260, 2820, 3, 1, <Address address_mode=ADDRESS_MODE.NWK address=0x4492>, <Address address_mode=ADDRESS_MODE.NWK address=0x0000>, b'\x18\x1d\x01\x00\x00\x00\x1b\x08\x00\x00\x00']
2021-01-26 17:52:28 DEBUG (MainThread) [zigpy.zcl] [0x4492:3:0x0b04] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=29 command_id=Command.Read_Attributes_rsp>
2021-01-26 17:52:28 DEBUG (MainThread) [homeassistant.components.zha.api] Read attribute for: cluster_id: [2820] cluster_type: [in] endpoint_id: [3] attribute: [0] manufacturer: [None] response: [bitmap32.8] failure: [{}],
2021-01-26 17:52:29 DEBUG (MainThread) [zigpy_zigate.uart] Frame received: 870100056dd0000b3500
2021-01-26 17:52:29 DEBUG (MainThread) [zigpy_zigate.api] data received 0x8701 b'd0000b35' LQI:0
2021-01-26 17:52:29 DEBUG (MainThread) [zigpy_zigate.zigbee.application] zigate_callback_handler [208, 0]
```

deepcoder, 2021-01-26:
Perhaps this should be under its own topic. When I added this A19 lightbulb to ZHA, ZHA included an electrical measurement attribute, see below. Pretty sure it does not have such, and sure hope this bulb is not using 1,600 watts:

```
- type: entities
  entities:
    - sensor.osram_lightify_a19_tunable_white_01_electrical_measurement
- type: light
  entity: light.osram_lightify_a19_tunable_white_01_level_light_color_on_off
```

[image]

pdecat, 2021-01-26:
Maybe some conversion issue somewhere affecting all Osram devices measurements?

deepcoder, 2021-01-27:
Bit off topic. But as I reviewed the debug level of logs as  attempts to apply quirks, if I read these correctly it seems like zigpy tries to apply every quirk it has (?) to each of my devices. Then if it finds something that 'matches' it applies it. Am I reading this logic correctly? Seems like this will not scale too well, and second as maybe is shown here can incorrectly apply quirks.

koying, 2021-01-27:
OSRAM plug does not report valid consumption, afaik.
It reports fixed 400W when ON, and fixed 28000W when OFF for me.

pdecat, 2021-01-28:
Ok then, let's close this issue.

closed 2021-01-28 by pdecat