zigpy/zha-device-handlers #750: [Device Support Request] OTA update for Eurotronic Spirit Zigbee

malfroid, 2021-01-26:
**Is your feature request related to a problem? Please describe.**
Current Eurotronic Spirit Zigbee firmware is unstable, which leads to many issues. See these issues: https://github.com/search?q=%22spirit+zigbee%22&state=open&type=Issues

Eurotronic recently started publishing [firmware releases on github](https://github.com/EUROTRONIC-Technology/Spirit-ZigBee/releases), together with a [guide on how to OTA update using deCONZ](https://github.com/EUROTRONIC-Technology/Spirit-ZigBee/releases).

**Describe the solution you'd like**
Support OTA update via ZHA using the firmware listed on https://github.com/EUROTRONIC-Technology/Spirit-ZigBee

ZHA already supports OTA updates for IKEA and LEDVANCE devices: https://www.home-assistant.io/integrations/zha#ota-firmware-updates

**Device signature**

``` json
{
  "node_descriptor": "NodeDescriptor(byte1=2, byte2=64, mac_capability_flags=128, manufacturer_code=4151, maximum_buffer_size=127, maximum_incoming_transfer_size=100, server_mask=11264, maximum_outgoing_transfer_size=100, descriptor_capability_field=0)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0301",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x000a",
        "0x0019",
        "0x0201"
      ],
      "out_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x0004",
        "0x000a",
        "0x0019",
        "0x0201"
      ]
    }
  },
  "manufacturer": "Eurotronic",
  "model": "SPZB0001",
  "class": "zhaquirks.eurotronic.spzb0001.SPZB0001"
}
```

Adminiuga, 2021-01-26:
for now you are probably better off downloading the firmware manually and placing it into ota folder

MattWestb, 2021-01-26:
https://github.com/zigpy/zigpy/wiki/OTA-Device-Firmware-Updates

The otau_directory and ikea_update_url is optional, **and is the file provider for any firmware files you have locally on your system**.

Hedda, 2021-02-01:
This issue should probably be moved to the zigpy repository instead? -> https://github.com/zigpy/zigpy

@malfroid please note that "zha-device-handlers" now only handles "quirks" and currently does not handle OTA providers.

You can read https://github.com/zigpy/zigpy/issues/154 for more information on how 'OTA Provider' (OTAProvider) works in zigpy.

Hedda, 2021-02-01:
> PS: Maybe it is a good idea if zha-device-handlers also provides custom 'OTA Provider' (OTAProvider) for Zigbee devices?

FYI, I started a new discussion about these specific questions/ideas here -> https://github.com/zigpy/zigpy/discussions/654

closed 2021-08-12 by github-actions[bot]