zigpy/zha-device-handlers #752: [Device Support Request] Xiaomi plug EU consumption (again)

koying, 2021-01-27:
I received an EU (mmeu01) plug from China yesterday, and that one has this signature:
```json
{
  "node_descriptor": "NodeDescriptor(byte1=1, byte2=64, mac_capability_flags=142, manufacturer_code=4447, maximum_buffer_size=127, maximum_incoming_transfer_size=100, server_mask=11264, maximum_outgoing_transfer_size=100, descriptor_capability_field=0)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0051",
      "in_clusters": [
        "0x0000",
        "0x0002",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0xfcc0"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "242": {
      "profile_id": 41440,
      "device_type": "0x0061",
      "in_clusters": [],
      "out_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "LUMI",
  "model": "lumi.plug.mmeu01",
  "class": "zigpy.device.Device"
}
```

Maybe a firmware issue, but I don't have a xiaomi bridge to upgrade...

Empirically (as in "I don't know what I'm doing" ;) ), I added this quirk class:
```python
class PlugV1(XiaomiCustomDevice):
    """lumi.plug.mmeu01 plug."""

    def __init__(self, *args, **kwargs):
        """Init."""
        self.voltage_bus = Bus()
        self.consumption_bus = Bus()
        self.power_bus = Bus()
        super().__init__(*args, **kwargs)

    signature = {
        MODELS_INFO: [(LUMI, "lumi.plug.mmeu01")],
        ENDPOINTS: {
            # <SimpleDescriptor endpoint=1 profile=260 device_type=81
            # device_version=1
            # input_clusters=[0, 2, 3, 4, 5, 6]
            # output_clusters=[10, 25]>
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.SMART_PLUG,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    DeviceTemperature.cluster_id,
                    Identify.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    OnOff.cluster_id,
                    0xfcc0
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            },
            # <SimpleDescriptor endpoint=242 profile=41440 device_type=97
            # device_version=0
            # input_clusters=[]
            # output_clusters=[33]>
            242: {
                PROFILE_ID: XIAOMI_PROFILE_ID,
                DEVICE_TYPE: XIAOMI_DEVICE_TYPE,
                OUTPUT_CLUSTERS: [GreenPowerProxy.cluster_id],
            },
        },
    }
    replacement = {
        SKIP_CONFIGURATION: True,
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.SMART_PLUG,
                INPUT_CLUSTERS: [
                    BasicCluster,
                    DeviceTemperature.cluster_id,
                    Identify.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    OnOff.cluster_id,
                    Alarms.cluster_id,
                    Metering.cluster_id,
                    ElectricalMeasurementCluster,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            },
            21: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.MAIN_POWER_OUTLET,
                INPUT_CLUSTERS: [AnalogInputCluster],
                OUTPUT_CLUSTERS: [AnalogInput.cluster_id, Groups.cluster_id],
            },
            242: {
                PROFILE_ID: XIAOMI_PROFILE_ID,
                DEVICE_TYPE: XIAOMI_DEVICE_TYPE,
                OUTPUT_CLUSTERS: [GreenPowerProxy.cluster_id],
            },
        },
    }
```

I now have 3 additonal entities:
- sensor.lumi_lumi_plug_mmeu01_6c268e3c_analog_input: Gives me consumption as float, e.g., 14.2
- sensor.lumi_lumi_plug_mmeu01_6c268e3c_electrical_measurement: Gives me consumption as W, e.g., 14W
- sensor.lumi_lumi_plug_mmeu01_6c268e3c_smartenergy_metering: Always "Unknown"

I would understand if you don't want to chase all the different Xiaomi BS implementation, but it would be appreciated if could indicate me what replacement to implement to, ideally, only have the `sensor.lumi_lumi_plug_mmeu01_6c268e3c_electrical_measurement`, and get rid of the 2 others.

Thanks

MattWestb, 2021-01-27:
If the device have the function i think it should being implanted in the quirk so other user that like using can do it.
If you have something you dont like being presented in HA you can always disable the entry in HA and you dont see it in the future (like on my network router i only like see the wan port statistics in HA and not all other "internal" ports statistic). 
(doing knowing if i have understanding your question right)

koying, 2021-01-27:
Sure I can disable those, but the question is more "What did I do wrong in the quirk that made those undesirable entities to pop up".
That is unless it is so by design, ofc.

Adminiuga, 2021-01-27:
This plug does not have endpoint id # 21 and this is the endpoint which used to send usage data in the older versions. In the `replacement` dict, remove `SKIP_CONFIGURATION` step and I assume endpoint id 21 could be removed too.
Then enable debug, leave the plug on and monitor logs for any communication from the plug.

koying, 2021-01-27:
I did the debug already, and factually, the plug is sending on 21, the signature is plainly wrong.
The error without adding 21:
```log
Ignoring message (b'182c0a550039f853e33e') on cluster 12: unknown endpoint or cluster id: 21
```

Adminiuga, 2021-01-27:
Ok, then having the endpoint 21 in the `replacement` is correct. 

>- sensor.lumi_lumi_plug_mmeu01_6c268e3c_analog_input: Gives me consumption as float, e.g., 14.2

That comes from endpoint id 21 and is a "raw" data of the elec measurement. It was done to have at least some data from devices which don't have the quirk. Feel free to hide/disable that entity.

> - sensor.lumi_lumi_plug_mmeu01_6c268e3c_electrical_measurement: Gives me consumption as W, e.g., 14W

Works as expected

> - sensor.lumi_lumi_plug_mmeu01_6c268e3c_smartenergy_metering: Always "Unknown"

remove `Metering.cluster_id,` from the replacement. After the restart, you may have to delete the dangling entity for this sensor. 

Once you are happy with the result, consider submitting a PR, see https://github.com/zigpy/zha-device-handlers#how-to-contribute

koying, 2021-01-27:
Meh.

Looks like this thing doesn't always send the same signature...
I was doing tests and getting very confused until I noticed.

Sometimes, I get 21+22, sometimes 21 and sometimes neither...

Here is the signature with the latest test
```json
{
  "node_descriptor": "NodeDescriptor(byte1=1, byte2=64, mac_capability_flags=142, manufacturer_code=4447, maximum_buffer_size=127, maximum_incoming_transfer_size=100, server_mask=11264, maximum_outgoing_transfer_size=100, descriptor_capability_field=0)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0051",
      "in_clusters": [
        "0x0000",
        "0x0002",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0xfcc0"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "21": {
      "profile_id": 260,
      "device_type": "0x0051",
      "in_clusters": [
        "0x000c"
      ],
      "out_clusters": []
    },
    "22": {
      "profile_id": 260,
      "device_type": "0x0051",
      "in_clusters": [
        "0x000c"
      ],
      "out_clusters": []
    },
    "242": {
      "profile_id": 41440,
      "device_type": "0x0061",
      "in_clusters": [],
      "out_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "LUMI",
  "model": "lumi.plug.mmeu01",
  "class": "zigpy.device.Device"
}
```

With 21+22, I indeed get 2 analogs. First one is Watt, no clue what the second is
[image]

Maybe this is relevant:
```log
2021-01-27 20:01:29 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0xCF16:21:0x000c]: reporting 'present_value' attr on 'analog_input' cluster: 30/900/1: Result: '[[ConfigureReportingResponseRecord(status=193, direction=0, attrid=85)]]'
2021-01-27 20:01:29 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0xCF16:21:0x000c]: finished channel configuration
[...]
2021-01-27 20:01:29 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0xCF16:22:0x000c]: reporting 'present_value' attr on 'analog_input' cluster: 30/900/1: Result: '[[ConfigureReportingResponseRecord(status=193, direction=0, attrid=85)]]'
2021-01-27 20:01:29 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0xCF16:22:0x000c]: finished channel configuration

```

Adminiuga, 2021-01-27:
Second analog cluster could be the summation of consumed power

Adminiuga, 2021-01-27:
Would love to know why it reports different signatures and what controls it

koying, 2021-01-28:
Trying to make sense of it, but seems random to me so far

closed 2021-08-12 by github-actions[bot]