zigpy/zha-device-handlers #754: [Device Support Request] Develco Products A/S AQSZB-110 Air Quality Sensor - VOC

BFFonseca, 2021-01-27:
**Is your feature request related to a problem? Please describe.**
I got an Air Quality Sensor from Develco (https://www.develcoproducts.com/products/sensors-and-alarms/air-quality-sensor/ )
I'm trying to use it with ZHA.
The Temperature and Humidity sensors work fine. 
The battery is always displaying "unknown" status.
The VOC sensor doesn't appear as an entity.

Checking the clusters, I can see it is receiving the VOC as ManufacturerSpecificCluster with ID: 0xfc03 - which is the correct ID according to the technical document on the product: https://www.develcoproducts.com/media/1674/aqszb-110-technical-manual-air-quality-sensor-04-08-20.pdf.
<details>
<summary>Clusters Print Screen</summary>

[image]

</details>

There's also an on/off cluster that I'm not sure what it is. The device has only 1 physical button (reset/pairing).
<details>
<summary>Device Info print screen</summary>

[image]

</details>

**Describe the solution you'd like**
Integration for both the VOC sensor and the battery status.

**Device signature - this can be acquired by removing the device from ZHA and pairing it again from the add devices screen. Be sure to add the entire content of the log panel after pairing the device to a code block below this line.**
<details>
<summary>Log Output</summary>

```
Device 0x9edc (00:15:bc:00:36:00:0a:c3) left the network
Device 0x998e (00:15:bc:00:36:00:0a:c3) joined the network [0x998e:zdo] ZDO request ZDOCmd.Device_annce: [0x998E, 00:15:bc:00:36:00:0a:c3, 0]
[0x998e] Requesting 'Node Descriptor'
Tries remaining: 2
[0x998e] Extending timeout for 0x05 request
[0x998e] Node Descriptor: NodeDescriptor(byte1=2, byte2=64, mac_capability_flags=0, manufacturer_code=4117, maximum_buffer_size=80, maximum_incoming_transfer_size=80, server_mask=0, maximum_outgoing_transfer_size=80, descriptor_capability_field=0)
[0x998e] Discovering endpoints
Tries remaining: 3
[0x998e] Extending timeout for 0x06 request
[0x998e] Discovered endpoints: [1, 38]
[0x998e:1] Discovering endpoint information
Tries remaining: 3
[0x998e] Extending timeout for 0x07 request
Received frame on uninitialized device 00:15:bc:00:36:00:0a:c3 endpoint 1 for cluster: 25
[0x998e:1] Discovered endpoint information: SizePrefixedSimpleDescriptor(endpoint=1, profile=49353, device_type=1, device_version=1, input_clusters=[3, 5, 6], output_clusters=[])
[0x998e:38] Discovering endpoint information
Tries remaining: 3
[0x998e] Extending timeout for 0x08 request
[0x998e:38] Discovered endpoint information: SizePrefixedSimpleDescriptor(endpoint=38, profile=260, device_type=770, device_version=0, input_clusters=[0, 1, 3, 32, 1026, 1029, 64515], output_clusters=[3, 10, 25])
[0x998e] Extending timeout for 0x09 request
[0x998e:38:0x0000] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=9 command_id=Command.Read_Attributes_rsp>
[0x998e:38] Manufacturer: Develco Products A/S
[0x998e:38] Model: AQSZB-110
Checking quirks for Develco Products A/S AQSZB-110 (00:15:bc:00:36:00:0a:c3)
Considering <class 'zhaquirks.xbee.xbee_io.XBeeSensor'>
Fail because endpoint list mismatch: {232, 230} {1, 38}
Considering <class 'zhaquirks.xbee.xbee3_io.XBee3Sensor'>
Fail because endpoint list mismatch: {232, 230} {1, 38}
Considering <class 'zhaquirks.smartthings.tag_v4.SmartThingsTagV4'>
Fail because endpoint list mismatch: {1} {1, 38}
Considering <class 'zhaquirks.smartthings.multi.SmartthingsMultiPurposeSensor'>
Fail because endpoint list mismatch: {1} {1, 38}
Considering <class 'zhaquirks.philips.rwl021.PhilipsRWL021'>
Fail because endpoint list mismatch: {1, 2} {1, 38}
Considering <class 'zhaquirks.philips.rwl020.PhilipsRWL020'>
Fail because endpoint list mismatch: {1, 2} {1, 38}
Considering <class 'zhaquirks.philips.rom001.PhilipsROM001'>
Fail because endpoint list mismatch: {1} {1, 38}
Considering <class 'zhaquirks.netvox.z308e3ed.Z308E3ED'>
[... 71 lines cut ...]
[0x998e:38:0x0402] Attribute report received: measured_value=-32768, tolerance=50
[0x998E:1:0x0006]: failed to set reporting for 'on_off' attr on 'on_off' cluster: 
[0x998E:1:0x0006]: finished channel configuration
[0x998E:1:0x0006]: 'async_configure' stage succeeded
[0x998E](AQSZB-110): completed configuration
[0x998E](AQSZB-110): stored in registry: ZhaDeviceEntry(name='Develco Products A/S AQSZB-110', ieee='00:15:bc:00:36:00:0a:c3', last_seen=1611771629.39609)
[0x998e] Extending timeout for 0x17 request
[0x998E:1:0x0003]: command failed: 'trigger_effect' args: '(2, 0)' kwargs '{}' exception: ''
[0x998E](AQSZB-110): started initialization
[0x998E:ZDO](AQSZB-110): 'async_initialize' stage succeeded
[0x998E:1:0x0006]: initializing channel: from_cache: False
[0x998e] Extending timeout for 0x18 request
[0x998E:38:0x0020]: initializing channel: from_cache: False
[0x998E:38:0x0020]: finished channel configuration
[0x998E:38:0x0405]: initializing channel: from_cache: False
[0x998e] Extending timeout for 0x19 request
[0x998E:38:0x0000]: initializing channel: from_cache: False
[0x998E:38:0x0000]: finished channel configuration
[0x998E:38:0x0001]: initializing channel: from_cache: False
[0x998e] Extending timeout for 0x1a request~
```

</details>
**Additional context**
I'm fairly new to ZHA and Zigbee in general but I could try to implement this by myself if you can give me some directions as what needs to be done.

closed 2021-11-22 by github-actions[bot]

Hedda, 2021-11-23:
> Wondering if the root cause is the same as that described by dmulcahey in #605 issue? That is, a a limitation in [ZHA integration](https://www.home-assistant.io/integrations/zha/) for Home Assistant which currently only support 1 sensor per cluster, and these type of devices requiring 2 or more sensor per cluster?

FYI, ZHA integration in HA does now support more than 1 sensor per cluster, however, support may have to be extended.

peter-funktionIT, 2022-01-04:
> FYI, ZHA integration in HA does now support more than 1 sensor per cluster, however, support may have to be extended.

My AQSZB-110 is found by the ZHA integration in Home Assistant 2021.12.7, but displays no sensors at all.

What kind of support needs to be extended? Quirks added?

Wolverine80, 2022-01-04:
I just got the sensor and it's showing temp, voc and humidity. It uses quirk: zhaquirks.develco.air_quality.AQSZB110

peter-funktionIT, 2022-01-04:
> I just got the sensor and it's showing temp, voc and humidity. It uses quirk: zhaquirks.develco.air_quality.AQSZB110 
Is this in Home Assistant? If so, which version?

Wolverine80, 2022-01-04:
Yes, 2021.12.7 and I use a "Sonoff Zigbee 3.0 USB Dongle Plus" Zigbee Stick with ZHA.

peter-funktionIT, 2022-01-04:
Weird that mine isn't working in the same version. I use the Popp ZB-stick (Silicon Labs EmberZNet based radio using the EZSP protocol).
Thanks for confirming, I'll keep troubleshooting mine.

mex1254, 2022-02-04:
with latest HA Version i get more entities now. but i don't get values for µg/m³

[image]

what am i doing wrong?

i get temperature and humidity

peter-funktionIT, 2022-02-04:
Same for me, I only get temperature and humidity after upgrading to 2022.2.0

X52p, 2022-02-09:
Same problem here.
The device is only showing Temperature and humidity.

It is not using any quirks (though I thought it should if I understand this correctly)

Device Signature is:
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.NONE: 0>, manufacturer_code=4117, maximum_buffer_size=80, maximum_incoming_transfer_size=80, server_mask=10752, maximum_outgoing_transfer_size=80, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=False, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 49353,
      "device_type": "0x0001",
      "in_clusters": [
        "0x0003",
        "0x0005",
        "0x0006"
      ],
      "out_clusters": []
    },
    "38": {
      "profile_id": 260,
      "device_type": "0x0302",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x0020",
        "0x0402",
        "0x0405",
        "0x042e",
        "0xfc03"
      ],
      "out_clusters": [
        "0x0003",
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "frient A/S",
  "model": "AQSZB-110",
  "class": "zigpy.device.Device"
}

Should we open a new issue, wait for this to be reopened? (I don't know what the preferred way is)

Wolverine80, 2022-02-09:
I still get VOC after update, but the parameter name changed, so I had to edit some config elements, and also the value is now 1 million off. Data from the sensor reads 120, the quirk gives 120000000 µg/m³

snjallingur, 2022-02-27:
Hi,
I have the same problem (VOC sensor showing but does not report any values) with running on HA 2022.2.9 and ZHA integration on a ConbeeII. As far as I can see (and correct me if I am wrong) it seems to work with Z2Mqtt (according to @Wolverine80) but not ZHA. 
Going forward I don´t know if it makes further sense to comment on a closed issue or if we should open a new one?
BR

snjallingur, 2022-02-27:
So I did some further testing by spinning up the Z2MQTT Docker image and I can confirm the VOC sensor is reporting in Zigbee2MQTT but not ZHA. As @Wolverine80 mentioned I needed to do some calibration on the VOC sensor (3 digits) and after that the air quality is reporting.
Coming to the conclusion that this issue should for sure be reopened. But it might be the case that the problem is a more generic as mentioned above from @Hedda.

X52p, 2022-02-27:
It seems there is already a new open issue here:
#1273

rozsap, 2022-04-19:
Since I bought a new gateway (ZigStar LilyZig), I reinstalled the Home Assistant (2022.4.0). For a while, this frient sensor worked fine for me too. Then suddenly the VOC sensor became inaccessible and a new one came. I do not understand...

[image]

Hedda, 2022-04-19:
@rozsap suggest downgrade to 2022.3.x since issues with quirks in 2022.4.x that are still being worked on. Or alternativle update or very latest and help developers troubleshoot.

dmulcahey, 2022-04-19:
What problem is being worked on in quirks for this device? The only issues I am aware of were related to events for remotes

rozsap, 2022-07-10:
> What problem is being worked on in quirks for this device? The only issues I am aware of were related to events for remotes

Every second update I get the same error.

rozsap, 2022-09-15:
The air quality sensor has not been working for months. More recently, it also shows the battery as 0% with new batteries.

[image]

mex1254, 2022-10-03:
Has anyone managed to bring the voclevel to life?

rozsap, 2022-10-03:
I have reconfigured the device. Voclevel was good for a while, but then the Temperature didn't work. Now the other Voclevel has reappeared out of nowhere and only it works. The battery indicator still doesn't work.

rozsap, 2022-10-03:
With MQTT it shows:
[image]

cvicari, 2022-11-24:
Similar problem for me, device AQSZB-110 used to work with Z2Mqtt, but in ZHA reports temperature and humidity, no VOC and no battery.
It really seems there is only a little thing missing to have it fully supported

filipkotian, 2024-12-14:
@cvicari or @rozsap did you find a solution for this? I only see temperature and humidity.

cvicari, 2024-12-14:
@filipkotian it never worked for me and I did not try again. I am currently happy with zigbee2mqtt

rozsap, 2024-12-14:
> @cvicari or @rozsap did you find a solution for this? I only see temperature and humidity.

It still often happens that the Air quality or VOC level value is 'unavailable'.

filipkotian, 2024-12-14:
@rozsap do you use custom quirk for zha? Could you please share the code? I dont even see VOC. Thanks

rozsap, 2024-12-14:
> @rozsap do you use custom quirk for zha? Could you please share the code? I dont even see VOC. Thanks

I don't use custom quirk for ZHA.