zigpy/zha-device-handlers #755: v0.0.53 breaks Hue remote

leopold005, 2021-01-27:
I have been using Quirks v0.0.49 for a couple of months and all has been well.  However, on updating to v0.0.53, I noticed that none of my Hue remotes would trigger any automations, either with HA automations or using ControllerX under AppDaemon.  None of my other zigbee switches (Ikea, Aqara) were affected.

I tested the Hue remotes under ZHA_Event and the triggers are being picked up just fine, as far as I can tell.  I was contemplating if there was a subtle change which broke the automation within ControllerX, but the remotes would also not trigger new automations using them either.

On reverting back to v0.0.49 and restarting HA, the problem was resolved.

I haven't tried any other versions between these two, so I can't say for sure whether it's this version or a previous one where the break happens, but I'm suspecting there has been a change made to the quirk for the Hue remote which breaks mine.

My Hue remotes are model RWL021 with firmware version 0x42006f9d

My HA installation is as follows:
Type:  Supervised on Debian Buster
Core version: 2021.1.5
Supervisor version: 2021.01.7

Adminiuga, 2021-01-28:
you need to run it with the beta HA release.

leopold005, 2021-01-30:
Ah, fair enough.

closed 2021-01-30 by leopold005

bobbynobble, 2021-03-07:
> Ah, fair enough.

Have you updated HA beyond 2021.1.5 and if so are the dimmers working OK still?

leopold005, 2021-03-07:
I tried the update and the Hue remotes had an issue again. They would be visible in ZHA and checking the ZHA_event would see responses, but none of the triggers worked. So I’m stuck at 2021.1.5 until I can work out how to fix it.

bobbynobble, 2021-03-07:
Thanks for replying, that's literally the problem I was having, zha events not matching those used by ControllerX after updating past 2021.1.5. Have now fixed it this evening thanks to the devs help, check the end of the issue below but essentially just deleted the /config/deps/ folder and am now in full working order and bang up to date across the board...

https://github.com/zigpy/zha-device-handlers/issues/782

leopold005, 2021-03-07:
Thanks for the info.  I've given that a whirl, updated HA and the Hue remotes are still working properly!  Thank you for getting to the bottom of this and letting me know.

bobbynobble, 2021-03-08:
No probs, credit to @MattWestb for tracking it down.

MattWestb, 2021-03-08:
Thanks for the credit !!

Im glad i can helping :-))

I was having similar problems for one year ago then testing bellows (V8 support) and installing pre commits with pip that was not in the dev branch and they was later blocking then upgrading the the "normal" dev branch.