zigpy/zha-device-handlers #757: [BUG] Unavailable State

rul3zz, 2021-01-30:
I have switched all Zigbee devices from Deconz to ZHA.
I have several bulbs, philips hue and osram smart + controlled by wifi relays. When the relay is off, the bulb keeps its last state (On or OFF) rather than becoming unavailable.
If the bulb is ON and the relay is turned off, the status of the bulb remains ON.
If the lamp is OFF and the relay is turned off, the status of the lamp remains OFF.
Too much time passes before the state of the bulb becomes unavailable, which creates various types of problems on automations etc. But even worse, if a light bulb goes off the grid for any reason it is difficult to notice immediately.
This did not happen with Deconz, usually it took a few seconds or minutes to detect the unavailable state.

It would be nice if it were possible to detect the unavailable status faster

This is the History of the Relay, i just turn on the relay for a couple of minutes and the turn off:
[image]

This is the History of the light bulb, after one hour without power it's still ON
[image]

MattWestb, 2021-01-30:
Most Philips HUE and OSRAM lights is not sending attribute reports to the coordinator so there status is pulled from the coordinator. For not storming the network its doing in lazy pace then no command is sensed to the devices. Then one command is being sent to the device ZHA is pulling the new status in some seconds and its updated. An with zigbee standard lights is reporting the status changes self to the ZHA after some seconds.

If you is using zigbee group commands for turning your light on and off the status reporting is working normal.
If putting the power off your lights you with one WiFI relay you is getting problems in the network and no status updating from the devices (then its dont have any batteries for sending the status change).

Can you explaining way you is power the Zigbee light of with one WiFi relay instead sending one off command to cluster 0x0006 ??

rul3zz, 2021-01-30:
I use a wifi relay to control the light also from a physical switch. Without a wifi relay I would have to keep the switch on and control the bulb only via internet or voice and if someone turns off the switch the light becomes uncontrollable.
In my house everything is controllable both from internet and physically.
For special needs I have set only some wifi switches, disabling the relay, to control via mqtt the on/off of the zigbee bulb through tasmota rules, so the bulbs are always powered and i don't have this problem.
I hope I was clear

The problem I have exposed is not present on Deconz, the bulb becomes unavailable after a while.
On ZHA it takes too much time, about an hour or more, if possible it would be nice to reduce this time

MattWestb, 2021-01-30:
If you is using real Zigbee Light Link switches bonded to light groups you can controlling the light without WiFi / internet and then HA is offline. And your zigbee network is working god 24/7 without devices going on and off all the time and they can reporting there status OK to HA. 

Something like this is recommended if you like have the wall switch still working but not normally being used [Light Switch White Cover (SM200)](https://www.amazon.co.uk/Samotech%C2%AE-Light-Switch-Philips-Dimmer/dp/B07D5KJ4B6).

By switching the power of for the zigbee light you is killing the zigbee network and getting much strange problems (also in deCONZ in the long run). 
I think its better going 100% WiFi like SonOff or tuya WiFi devices and not messing all things up. But you can getting one more robust function with one stable zigbee network (then not switching the router devices on and off) then WiFi one.

rul3zz, 2021-01-30:
I'm using wifi relay switch like sonoff touch with tasmota, not zigbee switch. 
I don't have network problems with Deconz and also with ZHA, I'm just pointing out this difference which can have several implications. First of all, if a zigbee device goes offline, for any reason, it cannot be identified from the dashboard or through an automation notification that detects the unavailable status until after an hour

closed 2021-02-07 by rul3zz