zigpy/zha-device-handlers #758: [Device Support Request] Lonsonho X711A _TZ3000_9hpxg80k

PerJarlemark, 2021-01-31:
**Is your feature request related to a problem? Please describe.**
I'm having the same problem that I have found that many other seems to have with this device, most of the times when I add the device to the network it keeps turning itself of after 2 minutes
(if you disconnect it and re-add it "lots" of times, you can get lucky and it works as expected but usually it stops working again after updates/reconfiguration)

**Product name:**
Lonsonho Tuya Zigbee Smart Switch With / No Neutral (Model :X711A)

**Brand:** 
Lonsonho

**URL on a website:**
https://www.aliexpress.com/item/4000298926256.html?spm=a2g0s.9042311.0.0.74224c4dbyaaml

**Device signature**
```
{
  "node_descriptor": "NodeDescriptor(byte1=2, byte2=64, mac_capability_flags=128, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=0)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0100",
      "in_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0x0006"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZ3000_9hpxg80k",
  "model": "TS0011",
  "class": "zigpy.device.Device"
}
```

**Related issues**
https://github.com/zigpy/zha-device-handlers/issues/535

mrzak1, 2021-02-12:
I have same issue like you. Device is automatically turn off after 2 minutes. I am not sure what cause this.

Palthe, 2021-02-18:
Same issue, tried lots of settings in Home Assistant but I cannot get it to work properly

MattWestb, 2021-07-18:
Shall being fixed in #845.
Can some one confirming if its working or not with switches with manufacturer `"_TZ3000_9hpxg80k"`

PerJarlemark, 2021-07-23:
I may be able to test, is this implemented in latest release of Home Assistant now or do I need to Install "Custom deps deployment" addon?

MattWestb, 2021-07-23:
Its added in the released HA so it shall working "out of the box" = is shall loading if the device signature is OK and you is getting the quirk name on the device card that the device (if) is using. 

If you have installed local quirks you shall deleting them or you can  getting problem if the code base is changed.

PerJarlemark, 2021-07-23:
I can confirm it is detected with **Quirk "zhaquirks.tuya.ts001x.TuyaSingleNoNeutralSwitch"** in **Home Assistant 2021.7.4**

I have tested to remove and re-add the device several times and it seems this update has fixed the problem, it is **no longer turning off automatically after 2 minutes.** 
I guess I can now start unboxing my other not used 7pcs of this switch model.

Thanks for all your help in sorting this out

regards
Per

MattWestb, 2021-07-24:
Great tat is working !!

Some new versions have one "decoupling" function that is not implanted but i can coming later if its needed, but i think its better baying the remote version in the same design and without the relay for doing it more "slimmed".

Is it fitting inside the EU standard round wall box without problem or do you need using some "force" for getting it fitting inside ?

For user that have the switch installed, i think its not needed reparing the switch but perhaps only repowering it after updating HA (I dont  have the device but only looking how the quirk is working).

And you have getting little to do in your "semester" :-)))

Mvh  MW

PerJarlemark, 2021-07-24:
Physically it fits perfect in the EU round wall box,. Not sure what “decoupling” function does, but the only downside of the switch I have found is that it connect as and “End Device” so it is not routing in the mesh, I´m not sure this is a hardware limitation or somthing can be fixed in the Quirk?

Brg
Per

MattWestb, 2021-07-24:
The "decoupling" you is putting 2 switches and one (the master) is using the local relay and the other is not using its local relay instead its sending the commands to the "master" switch like one remote (one switch without relay) is doing.

If one "normal" is one end device sleeping or not it cant being changed from the host side then is normally being fix programmed in the firmware.

Normally its good having many routers in the mesh but if having one large and dens network its better having more end devices that have the radio on then idle but i think its not the thinking in this case the devs have making it like this more reused the code from the remote switch (without relay) and putting one relay or 2 on the back on it.

MW

closed 2022-01-27 by github-actions[bot]

aquemy, 2023-02-24:
> The "decoupling" you is putting 2 switches and one (the master) is using the local relay and the other is not using its local relay instead its sending the commands to the "master" switch like one [...]

Hey! I am having a similar device (1, 2 and 3 gangs) and there is no way to wire them in two-ways or three-ways switches.
If I understand correctly, this "decoupling" mode would allow me to have a pseudo-multiway by telling one switch to not physically use the relay but only send the order to the master one?

If so, could you indicate me how to configure this feature or add this to a custom quirk?

kayti, 2023-11-05:
Problem with autoswitching off is occuring again. Home assistant 2023.11.1. After pairing with ZHA and turning it on (manually with physical button or via HA), light turn off itself after 2 minutes. Any idea what I could do to make it right?