zigpy/zha-device-handlers #759: [Device Support Request] SilverCrest/LIDL Doorbell (HG06668)

kkplein, 2021-01-31:
The device signature:
```
{
  "node_descriptor": "NodeDescriptor(byte1=2, byte2=64, mac_capability_flags=128, manufacturer_code=4619, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=0)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0402",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x0500",
        "0x0b05"
      ],
      "out_clusters": [
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZ1800_ladpngdx",
  "model": "TS0211",
  "class": "zigpy.device.Device"
}
```
This is the device: https://zigbee.blakadder.com/Lidl_HG06668.html

I can add it to HA/ZHA just fine, but the initialization never finishes, with:
```
Device 0x3d6f (84:2e:14:ff:fe:f8:20:af) joined the network
Device 84:2e:14:ff:fe:f8:20:af changed id (0x53b7 => 0x3d6f)
[0x3d6f:zdo] ZDO request ZDOCmd.Device_annce: [0x3D6F, 84:2e:14:ff:fe:f8:20:af, 128]
Checking quirks for _TZ1800_ladpngdx TS0211 (84:2e:14:ff:fe:f8:20:af)
Considering <class ‘zhaquirks.xbee.xbee_io.XBeeSensor’>
Fail because endpoint list mismatch: {232, 230} {1}
Considering <class ‘zhaquirks.xbee.xbee3_io.XBee3Sensor’>
Fail because endpoint list mismatch: {232, 230} {1}
Considering <class ‘zhaquirks.smartthings.tag_v4.SmartThingsTagV4’>
Fail because device_type mismatch on at least one endpoint
```
Let me know if there is any way I can help to make this device work in hassio / zha. (or if this is the wrong place for this request)

kkplein, 2021-02-01:
Ok, I just did a trial install of homeassistant with the zigbee2mqtt-edge integration instead of the ZHA , and both the Lidl doorbell and the motion detector are working there. Guess I will simply move my configs over to zigbee2mqtt. (I find also that the zigbee2mqtt is a bit more verbose in logging, and exposes more interesting zigbee device / network configuration information)
If there is anything I can still do to help add the device to ZHA: simply reply here, and I'll try to help. Thank you.

Avdbergnmf, 2021-03-15:
Is there any Update on this?

I prefer ZHA over for example Deconz, but I'd really like this doorbell to work :P

It does show up as a binary sensor, but the problem is when I click it, it gets set to open and doesn't close untill I reset the device. It seems that it should be as simple as changing this device from a binary sensor to a button?

denperss, 2021-05-07:
any updates on this issue ? :(

Avdbergnmf, 2021-05-07:
In the end I fixed it by switching to Zigbee2MQTT :'), probably not the option you're looking for, but it is a option that could work for you.

kkplein, 2021-05-07:
Yep same here. Running Zigbee2MQTT now, and never looked back.

PEGARifrouf, 2021-08-31:
Hello,
Same issue here, with the heiman doorbell button. It is a clone of the Lidl one (or heiman made them for Lidl).
I can't use Zigbee2MQTT because I am using a pop zigbee stick (chipset Silicon Labs EFR32MG13).
I can help to debug this device providing tests and logs.
A thread is open on Home Assistant community.
Thanks.

denperss, 2021-09-01:
Its working just fine for me with xigbee2mqtt.. 

[image]

denperss, 2021-09-01:
https://www.zigbee2mqtt.io/devices/HG06668.html#lidl-hg06668

PEGARifrouf, 2021-09-01:
Thanks, unfortunately I have to change my zigbee controller to be able to use Zigbee2MQTT. EFR32MG13 chipset is under development in alpha releases.
Is there a way to improve the device integration in ZHA with Zigbee2MQTT definition of the device ?
Can the device definition be modified easily like on a configuration file or compilation is required ?

magicbenny, 2021-09-28:
I feel happy for all of you who could solve this by switching to mqtt! But for me it sounds a bit like "ok my Porsche is to slow so I will buy a Ferrari" :-) For me it's not an option to switch to mqtt. It would be an acceptable answer if somebody will say "no we will/cannot implement this" and we can think about dealing with the fact to ignore this Doorbell Button in the future. :-)  

Switching to mqtt does not make any sense for me because all of my devices (other silvercrest/lidl devices including) work fine with zha. The Doorbell Button is the only one which is not working right now. So it's not worth to move all other devices to a different integration.

Avdbergnmf, 2021-09-28:
@magicbenny Honestly, I was in the same boat, but I was on the fence about switching to MQTT anyway because people praise it so much. I asked around and I don't think there was any interest to integrate the device at the time. So I think you either buy a new doorbell, or get z2m running one way or another. (or integrate the device yourself if you have the knowhow)

PEGARifrouf, 2021-09-28:
@magicbenny I end up by buying a compatible to z2mqtt and more powerful zigbee controller. My doorbell (heiman) was not recognize either on z2mqtt but it was much easier to implement it myself (was not entire working), and to ask for it's integration into z2mqtt to the community to make it properly working (messaging characteristics was not exactly the same as the Lidl one). It took just a few days and now both Lidl and Heiman versions are officially integrated into z2mqtt. On zha I had no clue how to do this.
But I get your point, and it shall be integrated into zha, just don't know how to do/ask help for it

magicbenny, 2021-12-10:
any news on this? tried with z2m but hat some other issues so I went back to zha. everything works fine. the lidl motion sensors and all other stuff. but not the doorbell. I was wondering that there is no zha_event when I tell HA to listen. on the other hand button is working fine in Deconz and z2m. what is the difference to zha in that case? :(

PEGARifrouf, 2021-12-10:
Actually, zha does not support the liddl doorbell because it is not declared in his dependency "zipgy". I think that ZHA can customize a device but not add it directly, it needs to be on zigpy component. Other zigbee plugins or integtations use it, like the official zigbee Jeedom plugin.
See this page : `https://zigbee.blakadder.com/Lidl_HG06668.html`

Maybe you need to ask for integration on the zigpy component `https://github.com/zigpy/zigpy` ?

In my case the Lidl one was supported by Z2M wich uses "herdsman-converters", and you can create "external-herdsman-converters" meaning creating your own from scratch. My Heiman doorbell is the one used by lidl but they have differents implementations of tuya to be seemless with their respective box, si It was not working for me out of the box in Z2M but was quick to add with help.

This is my comprehension, I perhaps made some mistake by explaining ZHA <-> zigpy interactions.

magicbenny, 2021-12-10:
> In my case the Lidl one was supported by Z2M wich uses "herdsman-converters", and you can create "external-herdsman-converters" meaning creating your own from scratch. My Heiman doorbell is the one [...]

Well yes. As I already mentioned I tried Z2M for 3 days, repaired all my 50+ Zigbee Devices 3 times (!!!), had a lot of issues. Slow reaction, connection Losts. I don't wanna blame Z2M for that. Maybe there is something wrong in my installation/configuration. I have read so many good things about Z2M and made the decision to buy a sonoff Zigbee hub, flash it with Tasmota and switch from ZHA to Z2M. What happened you can see above. I would go so far and say that my HA Installations is very good maintained but still Z2M and my HA don't like each other. How ever I switched back to zha yesterday and after read al 50+ Devices for the fourth time everything works fine. Specially my so much hated Eurotronics TVRs are now working. The only thing which is not working is the Button. Well my solution is: ordered a Aquara doorbell button. It's supported by ZHA and all other systems like blakadder said. Maybe I will give Z2M a second chance in the future on a separate Raspi with HA as Testsystem. We´ll see.

thx for your efforts and explanations in that case.

cheery,
ben

PEGARifrouf, 2021-12-10:
I almost did your solution, but I had other issues linked to the POPP adapter (that I dont have any more with my the same Aquara sensors).
The key is to have a fully compatible hardware with Z2M, i have done a lot of research to choose the right coordinator for Z2M. In my opinion, ZHA supports more controllers and has the big advantage to be very well integrated to HA and it is pushed by HA devs. It is just a matter of time to have new zigbee devices supported, if only there was a way to support new devices manually in ZHA ,it would be perfect.
Even with the right controller I remember having difficulties to make it work, mostly "things to know" but who are not straight forward to make Z2M work.

MattWestb, 2021-12-12:
We have getting  one nice device signature but no log then adding the device and also then using the button and the tamper function on the device.

Its one ITSA Zone device so need getting to know the zone and the commands is using for the devs can making on quirk for it then its not doing it ala Zigbee standard (then its still one tuya device).

Joerg76, 2021-12-17:
Hallo, here is the log file from my lidl doorbell nwk 0x43b4. I hope you can fix it.

2021-12-17 17:38:17 DEBUG (MainThread) [zigpy_deconz.api] Device state changed response: [<DeviceState.128|APSDE_DATA_REQUEST_SLOTS_AVAILABLE|APSDE_DATA_INDICATION|2: 170>, 0]
2021-12-17 17:38:17 DEBUG (MainThread) [zigpy_deconz.api] Command Command.aps_data_indication (1, 1)
2021-12-17 17:38:17 DEBUG (MainThread) [zigpy_deconz.api] APS data indication response: [32, <DeviceState.APSDE_DATA_REQUEST_SLOTS_AVAILABLE|2: 34>, <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0x0000>, 1, <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0x43b4>, 1, 260, 1280, b'\t!\x00\x01\x00\x00\x00\x00\x00', 0, 175, 255, 19, 112, 82, 0, -52]
2021-12-17 17:38:17 DEBUG (MainThread) [zigpy.zcl] [0x43b4:1:0x0500] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=33 command_id=0>
2021-12-17 17:38:17 DEBUG (MainThread) [zigpy.zcl] [0x43b4:1:0x0500] ZCL request 0x0000: [<ZoneStatus.Alarm_1: 1>, <bitmap8.0: 0>, 0, 0]
2021-12-17 17:38:17 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x43B4:1:0x0500]: Updated alarm state: ZoneStatus.Alarm_1
2021-12-17 17:38:17 DEBUG (MainThread) [zigpy_deconz.zigbee.application] Sending Zigbee request with tsn 33 under 45 request id, data: b'18210b0000'
2021-12-17 17:38:17 DEBUG (MainThread) [zigpy_deconz.api] 'aps_data_indication' response from <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0x43b4>, ep: 1, profile: 0x0104, cluster_id: 0x0500, data: b'092100010000000000'
2021-12-17 17:38:17 DEBUG (MainThread) [zigpy_deconz.api] Command Command.aps_data_request (20, 45, 0, <DeconzAddressEndpoint address_mode=2 address=0x43B4 endpoint=1>, 260, 1280, 1, b'\x18!\x0b\x00\x00', 2, 0)
2021-12-17 17:38:17 DEBUG (MainThread) [zigpy_deconz.api] APS data request response: [2, <DeviceState.APSDE_DATA_REQUEST_SLOTS_AVAILABLE|2: 34>, 45]
2021-12-17 17:38:17 DEBUG (MainThread) [zigpy_deconz.api] Device state changed response: [<DeviceState.128|APSDE_DATA_REQUEST_SLOTS_AVAILABLE|APSDE_DATA_CONFIRM|2: 166>, 0]
2021-12-17 17:38:17 DEBUG (MainThread) [zigpy_deconz.api] Command Command.aps_data_confirm (0,)
2021-12-17 17:38:17 DEBUG (MainThread) [zigpy_deconz.api] APS data confirm response for request with id 45: 00
2021-12-17 17:38:17 DEBUG (MainThread) [zigpy_deconz.api] Request id: 0x2d 'aps_data_confirm' for <DeconzAddressEndpoint address_mode=ADDRESS_MODE.NWK address=0x43b4 endpoint=1>, status: 0x00
2021-12-17 17:38:27 DEBUG (MainThread) [zigpy_deconz.api] Device state changed response: [<DeviceState.128|APSDE_DATA_REQUEST_SLOTS_AVAILABLE|APSDE_DATA_INDICATION|2: 170>, 0]
2021-12-17 17:38:27 DEBUG (MainThread) [zigpy_deconz.api] Command Command.aps_data_indication (1, 1)
2021-12-17 17:38:27 DEBUG (MainThread) [zigpy_deconz.api] APS data indication response: [32, <DeviceState.APSDE_DATA_REQUEST_SLOTS_AVAILABLE|2: 34>, <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0x0000>, 1, <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0x43b4>, 1, 260, 1280, b'\t"\x00\x04\x00\x00\x00\x00\x00', 0, 175, 255, 23, 112, 82, 0, -52]
2021-12-17 17:38:27 DEBUG (MainThread) [zigpy.zcl] [0x43b4:1:0x0500] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=34 command_id=0>
2021-12-17 17:38:27 DEBUG (MainThread) [zigpy.zcl] [0x43b4:1:0x0500] ZCL request 0x0000: [<ZoneStatus.Tamper: 4>, <bitmap8.0: 0>, 0, 0]
2021-12-17 17:38:27 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x43B4:1:0x0500]: Updated alarm state: ZoneStatus.0
2021-12-17 17:38:27 DEBUG (MainThread) [zigpy_deconz.zigbee.application] Sending Zigbee request with tsn 34 under 46 request id, data: b'18220b0000'
2021-12-17 17:38:27 DEBUG (MainThread) [zigpy_deconz.api] 'aps_data_indication' response from <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0x43b4>, ep: 1, profile: 0x0104, cluster_id: 0x0500, data: b'092200040000000000'
2021-12-17 17:38:27 DEBUG (MainThread) [zigpy_deconz.api] Command Command.aps_data_request (20, 46, 0, <DeconzAddressEndpoint address_mode=2 address=0x43B4 endpoint=1>, 260, 1280, 1, b'\x18"\x0b\x00\x00', 2, 0)
2021-12-17 17:38:27 DEBUG (MainThread) [zigpy_deconz.api] APS data request response: [2, <DeviceState.APSDE_DATA_REQUEST_SLOTS_AVAILABLE|2: 34>, 46]
2021-12-17 17:38:27 DEBUG (MainThread) [zigpy_deconz.api] Device state changed response: [<DeviceState.128|APSDE_DATA_REQUEST_SLOTS_AVAILABLE|APSDE_DATA_CONFIRM|2: 166>, 0]
2021-12-17 17:38:27 DEBUG (MainThread) [zigpy_deconz.api] Command Command.aps_data_confirm (0,)
2021-12-17 17:38:27 DEBUG (MainThread) [zigpy_deconz.api] APS data confirm response for request with id 46: 00
2021-12-17 17:38:27 DEBUG (MainThread) [zigpy_deconz.api] Request id: 0x2e 'aps_data_confirm' for <DeconzAddressEndpoint address_mode=ADDRESS_MODE.NWK address=0x43b4 endpoint=1>, status: 0x00
2021-12-17 17:38:36 DEBUG (MainThread) [zigpy_deconz.api] Device state changed response: [<DeviceState.128|APSDE_DATA_REQUEST_SLOTS_AVAILABLE|APSDE_DATA_INDICATION|2: 170>, 0]
2021-12-17 17:38:36 DEBUG (MainThread) [zigpy_deconz.api] Command Command.aps_data_indication (1, 1)
[... cut ...] ttrib-5': 24, 'X-attrib-6': 1, 'path': 0, '0xff01-100': <Bool.false: 0>}]
2021-12-17 17:38:59 DEBUG (MainThread) [zigpy.zcl] [0x6088:1:0x0001] Voltage mV: [Min]:2820 < [RAW]:3005 < [Max]:3100, Battery Percent: 66.0
2021-12-17 17:38:59 DEBUG (MainThread) [zigpy_deconz.api] 'aps_data_indication' response from <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0x6088>, ep: 1, profile: 0x0104, cluster_id: 0x0000, data: b'1c5f11cd0a01ff421d0121bd0b0328140421a84305211800062401000000000a210000641000'

closed 2022-10-14 by github-actions[bot]