zigpy/zha-device-handlers #762: [BUG] ZHA aquara pressure sensor is not updating

devin122, 2021-02-01:
**Describe the bug**
I'm seeing update messages in the debug logs, but in the pressure is not updating in the ui

**Expected behavior**
measurements in the UI should match values in the logs.

**Screenshots**

[image]
.

**Additional context**
```
{
  "node_descriptor": "NodeDescriptor(byte1=2, byte2=64, mac_capability_flags=128, manufacturer_code=4151, maximum_buffer_size=127, maximum_incoming_transfer_size=100, server_mask=0, maximum_outgoing_transfer_size=100, descriptor_capability_field=0)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0302",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x0402",
        "0x0403",
        "0x0405",
        "0xffff"
      ],
      "out_clusters": [
        "0x0000",
        "0x0004",
        "0xffff"
      ]
    }
  },
  "manufacturer": "LUMI",
  "model": "lumi.weather",
  "class": "zhaquirks.xiaomi.aqara.weather.Weather2"
}
```

```
2021-01-31 21:27:36 DEBUG (MainThread) [zigpy.zcl] [0x0006:1:0x0403] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=False disable_default_response=True> manufacturer=None tsn=143 command_id=Command.Report_Attributes>
2021-01-31 21:27:36 DEBUG (MainThread) [zigpy.zcl] [0x0006:1:0x0403] ZCL request 0x000a: [[Attribute(attrid=0, value=<TypeValue type=int16s, value=136>), Attribute(attrid=20, value=<TypeValue type=int8s, value=-1>), Attribute(attrid=16, value=<TypeValue type=int16s, value=1360>)]]
2021-01-31 21:27:36 DEBUG (MainThread) [zigpy.zcl] [0x0006:1:0x0403] Attribute report received: measured_value=136, 20=-1, 16=1360
```

devin122, 2021-02-01:
ah, after a bit of digging it appears values outside of the documented range are intentionally ignored  see #296. Given that the sensor appears to be generating useful data < 300hPa, and that the original issue the invalid data was -1, I think expanding the lower bound of acceptable values to 0 makes sense

Adminiuga, 2021-02-01:
What are you doing with that sensor?

devin122, 2021-02-01:
I got some vacuum storage bags to keep my 3d printing filament dry, and I was curious just how strong a vacuum you could actually pull with the plastic hand pump.

Adminiuga, 2021-02-02:
@dmulcahey I guess in https://github.com/zigpy/zha-device-handlers/pull/296 the concern was the reporting of `-1` values. I guess accepting the pressure reports bellow 30hPa should be fine, even though it is outside of the manufacturer specs?

PR #763 (2021-02-15): Expand usable pressure range of Aquara weather sensor
While the device specs list the pressure sensor going down to 30kPa,
the device still generates usable data below that.

fixes #762
commit 4022b04a (2021-02-15): Expand usable pressure range of Aquara weather sensor (#763)

While the device specs list the pressure sensor going down to 30kPa,
the device still generates usable data below that.

fixes #762
closed 2021-02-15 by dmulcahey