zigpy/zha-device-handlers #764: [BUG] Quirk for Sengled E1E-G7F not firing HA events and throws errors in HA core

NobleWolf, 2021-02-03:
**Describe the bug**
Home Assistant Version: 2021.1.5
zha-quirks version: 0.0.53 (pr#707 included this remote for version 0.0.52)
Zigbee Radio: Phoscon Conbee ii

When I press any button on my Sengled E1E-G7F there are no zha_event events fired. I also listened to "*" _all_ events but no events fired for the remote there either.
And the following error and string of information appears in the HA Core log:

_Single short press "ON" button_
```
2021-02-03 12:29:54 ERROR (MainThread) [homeassistant] Error doing job: Exception in callback SerialTransport._read_ready()
Traceback (most recent call last):
  File "/usr/local/lib/python3.8/asyncio/events.py", line 81, in _run
    self._context.run(self._callback, *self._args)
  File "/usr/local/lib/python3.8/site-packages/serial_asyncio/__init__.py", line 119, in _read_ready
    self._protocol.data_received(data)
  File "/usr/local/lib/python3.8/site-packages/zigpy_deconz/uart.py", line 91, in data_received
    self._api.data_received(frame)
  File "/usr/local/lib/python3.8/site-packages/zigpy_deconz/api.py", line 360, in data_received
    getattr(self, "_handle_%s" % (command.name,))(data)
  File "/usr/local/lib/python3.8/site-packages/zigpy_deconz/api.py", line 485, in _handle_aps_data_indication
    self._app.handle_rx(
  File "/usr/local/lib/python3.8/site-packages/zigpy_deconz/zigbee/application.py", line 324, in handle_rx
    self.handle_message(device, profile_id, cluster_id, src_ep, dst_ep, data)
  File "/usr/local/lib/python3.8/site-packages/zigpy/application.py", line 223, in handle_message
    return sender.handle_message(
  File "/usr/local/lib/python3.8/site-packages/zigpy/device.py", line 279, in handle_message
    return endpoint.handle_message(
  File "/usr/local/lib/python3.8/site-packages/zigpy/endpoint.py", line 217, in handle_message
    handler(hdr, args)
  File "/usr/local/lib/python3.8/site-packages/zigpy/zcl/__init__.py", line 193, in handle_message
    self.handle_cluster_request(hdr.tsn, hdr.command_id, args)
TypeError: handle_cluster_request() takes 3 positional arguments but 4 were given
```

_Single short press "OFF" button_
```
2021-02-03 12:30:49 ERROR (MainThread) [homeassistant] Error doing job: Exception in callback SerialTransport._read_ready()
Traceback (most recent call last):
  File "/usr/local/lib/python3.8/asyncio/events.py", line 81, in _run
    self._context.run(self._callback, *self._args)
  File "/usr/local/lib/python3.8/site-packages/serial_asyncio/__init__.py", line 119, in _read_ready
    self._protocol.data_received(data)
  File "/usr/local/lib/python3.8/site-packages/zigpy_deconz/uart.py", line 91, in data_received
    self._api.data_received(frame)
  File "/usr/local/lib/python3.8/site-packages/zigpy_deconz/api.py", line 360, in data_received
    getattr(self, "_handle_%s" % (command.name,))(data)
  File "/usr/local/lib/python3.8/site-packages/zigpy_deconz/api.py", line 485, in _handle_aps_data_indication
    self._app.handle_rx(
  File "/usr/local/lib/python3.8/site-packages/zigpy_deconz/zigbee/application.py", line 324, in handle_rx
    self.handle_message(device, profile_id, cluster_id, src_ep, dst_ep, data)
  File "/usr/local/lib/python3.8/site-packages/zigpy/application.py", line 223, in handle_message
    return sender.handle_message(
  File "/usr/local/lib/python3.8/site-packages/zigpy/device.py", line 279, in handle_message
    return endpoint.handle_message(
  File "/usr/local/lib/python3.8/site-packages/zigpy/endpoint.py", line 217, in handle_message
    handler(hdr, args)
  File "/usr/local/lib/python3.8/site-packages/zigpy/zcl/__init__.py", line 193, in handle_message
    self.handle_cluster_request(hdr.tsn, hdr.command_id, args)
TypeError: handle_cluster_request() takes 3 positional arguments but 4 were given
```

_Single short press "increase brightness" button_
```
2021-02-03 12:31:28 ERROR (MainThread) [homeassistant] Error doing job: Exception in callback SerialTransport._read_ready()
Traceback (most recent call last):
  File "/usr/local/lib/python3.8/asyncio/events.py", line 81, in _run
    self._context.run(self._callback, *self._args)
  File "/usr/local/lib/python3.8/site-packages/serial_asyncio/__init__.py", line 119, in _read_ready
    self._protocol.data_received(data)
  File "/usr/local/lib/python3.8/site-packages/zigpy_deconz/uart.py", line 91, in data_received
    self._api.data_received(frame)
  File "/usr/local/lib/python3.8/site-packages/zigpy_deconz/api.py", line 360, in data_received
    getattr(self, "_handle_%s" % (command.name,))(data)
  File "/usr/local/lib/python3.8/site-packages/zigpy_deconz/api.py", line 485, in _handle_aps_data_indication
    self._app.handle_rx(
  File "/usr/local/lib/python3.8/site-packages/zigpy_deconz/zigbee/application.py", line 324, in handle_rx
    self.handle_message(device, profile_id, cluster_id, src_ep, dst_ep, data)
  File "/usr/local/lib/python3.8/site-packages/zigpy/application.py", line 223, in handle_message
    return sender.handle_message(
  File "/usr/local/lib/python3.8/site-packages/zigpy/device.py", line 279, in handle_message
    return endpoint.handle_message(
  File "/usr/local/lib/python3.8/site-packages/zigpy/endpoint.py", line 217, in handle_message
    handler(hdr, args)
  File "/usr/local/lib/python3.8/site-packages/zigpy/zcl/__init__.py", line 193, in handle_message
    self.handle_cluster_request(hdr.tsn, hdr.command_id, args)
TypeError: handle_cluster_request() takes 3 positional arguments but 4 were given
```

_Single short press "decrease brightness" button_
```
2021-02-03 12:31:55 ERROR (MainThread) [homeassistant] Error doing job: Exception in callback SerialTransport._read_ready()
Traceback (most recent call last):
  File "/usr/local/lib/python3.8/asyncio/events.py", line 81, in _run
    self._context.run(self._callback, *self._args)
  File "/usr/local/lib/python3.8/site-packages/serial_asyncio/__init__.py", line 119, in _read_ready
    self._protocol.data_received(data)
  File "/usr/local/lib/python3.8/site-packages/zigpy_deconz/uart.py", line 91, in data_received
    self._api.data_received(frame)
  File "/usr/local/lib/python3.8/site-packages/zigpy_deconz/api.py", line 360, in data_received
    getattr(self, "_handle_%s" % (command.name,))(data)
  File "/usr/local/lib/python3.8/site-packages/zigpy_deconz/api.py", line 485, in _handle_aps_data_indication
    self._app.handle_rx(
  File "/usr/local/lib/python3.8/site-packages/zigpy_deconz/zigbee/application.py", line 324, in handle_rx
    self.handle_message(device, profile_id, cluster_id, src_ep, dst_ep, data)
  File "/usr/local/lib/python3.8/site-packages/zigpy/application.py", line 223, in handle_message
    return sender.handle_message(
  File "/usr/local/lib/python3.8/site-packages/zigpy/device.py", line 279, in handle_message
    return endpoint.handle_message(
  File "/usr/local/lib/python3.8/site-packages/zigpy/endpoint.py", line 217, in handle_message
    handler(hdr, args)
  File "/usr/local/lib/python3.8/site-packages/zigpy/zcl/__init__.py", line 193, in handle_message
    self.handle_cluster_request(hdr.tsn, hdr.command_id, args)
TypeError: handle_cluster_request() takes 3 positional arguments but 4 were given
```
Let me know if you want more from my logs.

**To Reproduce**
Steps to reproduce the behavior:
1. Pair Sengled E1E-G7F remote
2. In Home Assistant (HA) go to "Developer Tools"
3. Start listening to "zha_event" events
4. Press any button on remote
5. See no events
6. See above error in Supervisor Core Log

**Expected behavior**
"zha_event" events fired for every button press combination for the remote.

**Screenshots**
Not applicable.

**Additional context**

Device Signature
```
{
  "node_descriptor": "NodeDescriptor(byte1=2, byte2=64, mac_capability_flags=128, manufacturer_code=4448, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=0)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0104",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x0020",
        "0xfc11"
      ],
      "out_clusters": [
        "0x0003",
        "0x0004",
        "0x0006",
        "0x0008",
        "0xfc10"
      ]
    }
  },
  "manufacturer": "sengled",
  "model": "E1E-G7F",
  "class": "zhaquirks.sengled.e1e_g7f.SengledE1EG7F"
}
```

Adminiuga, 2021-02-03:
Upgrade to 2021.2.0 release -- recommended
or install latest zigpy lib in you instance

compeek, 2021-02-04:
I have been checking every Home Assistant update to see when the new device handler would show up. I upgraded to 2021.2.0 today, and my Sengled E1E-G7F is showing the custom device handler, and it's working as expected for me.

NobleWolf, 2021-02-04:
@compeek and @Adminiuga it is now working for me. I did not see the zigpy update in HA v2021.2.1 so I tried to install zigpy 0.0.53 manually, but I guess it didn't work. Now that I've upgraded to HA v2021.2.1 it is working.
I'm also polishing an HA Blueprint for this device and will post later today.

closed 2021-02-04 by NobleWolf