zigpy/zha-device-handlers #765: [BUG] Problems with quirks on raspberry pi

CharlieBailly, 2021-02-03:
**Describe the bug**
Hi all, I already opened an issue #667 because I was not able to test custom quirks on a raspberry pi install, but I solved it by using a virtual machine install. However, I was hoping that my working quirk (working on a virtual machine install) would work on my main raspberry install once my pull request would be merged. But it is not the case : I am running HassOS 2021.2.0 and my custom quirk #701 is available in the current version, but it is not working. So I am suspecting that zhaquirks are not working properly on a raspberry pi (4 4GB in my case) because :

1. I was unable to test custom quirks on a raspberry pi install and I needed to do my test on a virtual machine
2. Even though my code is available in the current zhaquirk version, my raspberry pi won't apply the quirk to my tuya siren device.

The code of my quirk is [here](https://github.com/zigpy/zha-device-handlers/blob/6c480990876cbbe1590efe2cfe5e49fd7cd07a02/zhaquirks/tuya/siren.py#L180)

To be clear, I am running HassOS 2021.2.0 on a raspberry pi 4 4GB, and here are the logs that I get during the pairing :
```txt
[0x0000:zdo] ZDO request ZDOCmd.Mgmt_Permit_Joining_req: [60, <Bool.false: 0>]
[0x7114:1:0x0702] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=125 command_id=Command.Report_Attributes>
[0x7114:1:0x0702] ZCL request 0x000a: [[Attribute(attrid=0, value=<TypeValue type=uint48_t, value=908>)]]
[0x7114:1:0x0702] Attribute report received: current_summ_delivered=908
Device 0xedd1 (84:2e:14:ff:fe:db:84:32) joined the network
[0xedd1] Requesting 'Node Descriptor'
Tries remaining: 2
[0xedd1] Extending timeout for 0x91 request
Device 0xedd1 (84:2e:14:ff:fe:db:84:32) joined the network
Skip initialization for existing device 84:2e:14:ff:fe:db:84:32
Device 0xedd1 (84:2e:14:ff:fe:db:84:32) joined the network
Skip initialization for existing device 84:2e:14:ff:fe:db:84:32
[0xedd1:zdo] ZDO request ZDOCmd.Device_annce: [0xEDD1, 84:2e:14:ff:fe:db:84:32, 128]
[0xedd1:zdo] ZDO request ZDOCmd.Node_Desc_req: [0x0000]
[0xedd1:zdo] Unsupported ZDO request:ZDOCmd.Node_Desc_req
Received frame on uninitialized device 84:2e:14:ff:fe:db:84:32 (Status.NEW) for endpoint: 1
Received frame on uninitialized device 84:2e:14:ff:fe:db:84:32 (Status.NEW) for endpoint: 1
Received frame on uninitialized device 84:2e:14:ff:fe:db:84:32 (Status.NEW) for endpoint: 1
Received frame on uninitialized device 84:2e:14:ff:fe:db:84:32 (Status.NEW) for endpoint: 1
Received frame on uninitialized device 84:2e:14:ff:fe:db:84:32 (Status.NEW) for endpoint: 1
Received frame on uninitialized device 84:2e:14:ff:fe:db:84:32 (Status.NEW) for endpoint: 1
Received frame on uninitialized device 84:2e:14:ff:fe:db:84:32 (Status.NEW) for endpoint: 1
Received frame on uninitialized device 84:2e:14:ff:fe:db:84:32 (Status.NEW) for endpoint: 1
[0xedd1] Node Descriptor: NodeDescriptor(byte1=2, byte2=64, mac_capability_flags=128, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=0)
[0xedd1] Discovering endpoints
Tries remaining: 3
[0xedd1] Extending timeout for 0x93 request
[0x8C07:1:0x0b04]: async_update
[0x7114:1:0x0b04]: async_update
[0xB9E3:1:0x0b04]: async_update
[0xDA83:1:0x0b04]: async_update
[0x5E27:1:0x0b04]: async_update
[0xedd1] Discovered endpoints: [1]
[0xedd1:1] Discovering endpoint information
Tries remaining: 3
[0xedd1] Extending timeout for 0x9f request
Received frame on uninitialized device 84:2e:14:ff:fe:db:84:32 endpoint 1 for cluster: 61184
[0xedd1:zdo] ZDO request ZDOCmd.Active_EP_rsp: [<Status.SUCCESS: 0>, 0xEDD1, [1]]
[0xedd1:zdo] Unsupported ZDO request:ZDOCmd.Active_EP_rsp
Received frame on uninitialized device 84:2e:14:ff:fe:db:84:32 endpoint 1 for cluster: 61184
[0x8c07:1:0x0b04] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=149 command_id=Command.Read_Attributes_rsp>
[0x6540:1:0x0b04]: async_update
[0x7114:1:0x0b04] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=151 command_id=Command.Read_Attributes_rsp>
[0xB787:1:0x0b04]: async_update
[0x7114:1:0x0b04] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=151 command_id=Command.Read_Attributes_rsp>
[0x7114:1:0x0b04] ZCL request 0x0001: [[ReadAttributeRecord(attrid=1291, status=<Status.SUCCESS: 0>, value=<TypeValue type=int16s, value=0>)]]
[0xb9e3:1:0x0b04] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=153 command_id=Command.Read_Attributes_rsp>
[0xda83:1:0x0b04] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=155 command_id=Command.Read_Attributes_rsp>
[0x5e27:1:0x0b04] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=157 command_id=Command.Read_Attributes_rsp>
[0x6540:1:0x0b04] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=161 command_id=Command.Read_Attributes_rsp>
[0xedd1:1] Discovered endpoint information: SizePrefixedSimpleDescriptor(endpoint=1, profile=260, device_type=81, device_version=1, input_clusters=[0, 4, 5, 61184], output_clusters=[25, 10])
Unknown cluster 61184
[0xedd1] Extending timeout for 0xa5 request
[0xb787:1:0x0b04] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=163 command_id=Command.Read_Attributes_rsp>
Received frame on uninitialized device 84:2e:14:ff:fe:db:84:32 endpoint 1 for cluster: 61184
Received frame on uninitialized device 84:2e:14:ff:fe:db:84:32 endpoint 1 for cluster: 61184
[0xedd1:1:0x0000] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=165 command_id=Command.Read_Attributes_rsp>
[0xedd1:1] Manufacturer: _TZE200_d0yu2xgi
[0xedd1:1] Model: TS0601
Checking quirks for _TZE200_d0yu2xgi TS0601 (84:2e:14:ff:fe:db:84:32)
[... 144 lines cut ...]
[0x6540:1:0x0702] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=45 command_id=Command.Report_Attributes>
[0x6540:1:0x0702] ZCL request 0x000a: [[Attribute(attrid=0, value=<TypeValue type=uint48_t, value=32>)]]
[0x6540:1:0x0702] Attribute report received: current_summ_delivered=32
Duplicate 45 TSN
[0xedd1:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=5 command_id=2>
[0xedd1:1:0xef00] Unknown cluster-specific command 2
[0xedd1:1:0xef00] ZCL request 0x0002: b'\x00\x02j\x02\x00\x04\x00\x00\x00)'
[0xedd1:1:0xef00] No handler for cluster command 2
[0xedd1:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=6 command_id=2>
[0xedd1:1:0xef00] Unknown cluster-specific command 2
[0xedd1:1:0xef00] ZCL request 0x0002: b'\x00\x02e\x04\x00\x01\x04'
[0xedd1:1:0xef00] No handler for cluster command 2
[0xedd1:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=6 command_id=2>
[0xedd1:1:0xef00] Unknown cluster-specific command 2
[0xedd1:1:0xef00] ZCL request 0x0002: b'\x00\x02e\x04\x00\x01\x04'
[0xedd1:1:0xef00] No handler for cluster command 2
[0xedd1:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=7 command_id=17>
[0xedd1:1:0xef00] Unknown cluster-specific command 17
[0xedd1:1:0xef00] ZCL request 0x0011: b'\x00\x06@'
[0xedd1:1:0xef00] No handler for cluster command 17

```

I really don't understand why quirks are so weird on a raspberry pi, maybe someone can help me !
Thanks in advance !

MattWestb, 2021-02-03:
I is no quirk maker but understand little of how it working.
Your log is saying something is not matching:
```
Checking quirks for _TZE200_d0yu2xgi TS0601 (84:2e:14:ff:fe:db:84:32)
Considering <class 'zhaquirks.xiaomi.mija.smoke.MijiaHoneywellSmokeDetectorSensor'>
Fail because device_type mismatch on at least one endpoint
```
First look i cant see any that is wrong but was comparing with my tuya RTV that have the same zigbee module = the same endpoints and cluster but different MODL_INFO.
I can see 2 lines that is different, out clusters have switched positions:
```
#  output_clusters=[25, 10]>

        OUTPUT_CLUSTERS: [Ota.cluster_id, Time.cluster_id],
```
You can copy the same part  from the `zhaquirks/tuya/valve.py` and only changing  the MODEL_INFO and it shall being rigged.

I have testing some quirks on PI 3b+  in docker container its working but is not easy to getting it working.

Keep trying it must being possible getting it rigged !!!

CharlieBailly, 2021-02-04:
Thanks for your answer, but my quirk is tested and is working, and the pull request has been merged, the only problem is that ZHA is not trying the good quirks on my pi and I can't figure out why.
I will try again on a virtual machine install when I'll have the time.

MattWestb, 2021-02-04:
Im very sorry but your ZHA is trying to using your quirk if you is reading the log you have provided:
```
Checking quirks for _TZE200_d0yu2xgi TS0601 (84:2e:14:ff:fe:db:84:32)
Considering <class 'zhaquirks.xiaomi.mija.smoke.MijiaHoneywellSmokeDetectorSensor'>
Fail because device_type mismatch on at least one endpoint
```
I have copying the TRVs end point config from the valve with the same zigbee module and putting your  MODELS_INFO in ther:
```

    signature = {
        #  endpoint=1 profile=260 device_type=81 device_version=0 input_clusters=[0, 4, 5, 61184]
        #  output_clusters=[10, 25]>
        MODELS_INFO: [("_TZE200_d0yu2xgi", "TS0601")],
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.SMART_PLUG,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaManufClusterAttributes.cluster_id,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            }
        },
    }
```
As you can see its the both lines with out cluster changed and its very likely doing the ZHA not trigger the quirk.

If you like doing the test on your PI you must following the instruction or ZHA is writing over all changes if not doing right on every reboot. 
More safer is using one vm but its more work to doo for getting it working OK.

Adminiuga, 2021-02-04:
How did you install the quirks?
- Remove that configuration
- Remove HA container and tell supervisor to rebuild it

Adminiuga, 2021-02-04:
the fact that it checks for this quirk:
```
Considering <class 'zhaquirks.xiaomi.mija.smoke.MijiaHoneywellSmokeDetectorSensor'>
Fail because device_type mismatch on at least one endpoint
```
first, is a taletell sign that your setup is broken, as it shouldn't check for this quirk. Something is screwed up with your zha-quirks installation.

CharlieBailly, 2021-02-05:
> How did you install the quirks?
> 
>     * Remove that configuration
> 
>     * Remove HA container and tell supervisor to rebuild it

I didn't install the quirk since it is now part of zhaquirks.

CharlieBailly, 2021-02-05:
> the fact that it checks for this quirk:
>
> ```
> Considering <class 'zhaquirks.xiaomi.mija.smoke.MijiaHoneywellSmokeDetectorSensor'>
> Fail because device_type mismatch on at least one endpoint
> ``` [...]

Ok, it is strange since I only configured ZHA using the GUI or by changing my sonoff zigbee mini devices type to switch instead of light in my `configuration.yaml` file

Adminiuga, 2021-02-05:
> I didn't install the quirk since it is now part of zhaquirks.

Part of `zhaquirks` is not the same as "part of HA Core 2021.1.4". keyword -- versioning. check zha's manifest.json

All I can say: as long as the quirk signature is correct, it would match your device. If it doesn't then it either your installation is broken or you are using the wrong version.

CharlieBailly, 2021-02-05:
> Part of `zhaquirks` is not the same as "part of HA Core 2021.1.4". keyword -- versioning. check zha's manifest.json
>
> All I can say: as long as the quirk signature is correct, it would match your [...]

Ok then my quirk is part of HA Core 2021.2 (the version I have). Moreover I can see the changes of my quirk if I open `tuya/siren.py`, so it must be "installed".
The quirk signature is correct since it was working on another install.
So according to you, my installation is broken, but that is really weird, I'll check that when I have more time.
Thanks for your answer.

MattWestb, 2021-03-09:
If you have installing packages with pip you can deleting them.
Look on this issue https://github.com/zigpy/zha-device-handlers/issues/782#issuecomment-792307991 and going thru all steps for finding if you have some miss match.

MattWestb, 2021-03-09:
@Adminiuga Is it possible putting one log warning / info if using some "pip installed" libs so you can see that in the log with one easy cat / grep  command ??

It can helping easier finding the source for strange problems like this.

CharlieBailly, 2021-03-09:
These are good ideas, good work @MattWestb !