zigpy/zha-device-handlers #769: [BUG] Phillips Hue Light Model LCA001 does not use quirk

SHxKM, 2021-02-06:
**Describe the bug**
This Hue smart light bulb (model: 'LCA001') used to be paired and was identified correctly as "Phillips Hue" bulb, I could also reach out and modify the desired clusters through ZHA. At some point in time during tests I unpaired it, but when I re-paired it was identified as "signify netherlands b.v" as the manufacturer. The bigger problem is that the clusters I was modifying are no longer exposed through ZHA.

**To Reproduce**
Steps to reproduce the behavior:
1. Pair a Hue smart light bulb with model LCA001
2. Try to find Cluster and cluster attribute that is exposed by the quirk
3. They are nowhere to be found

**Expected behavior**
To find certain clusters and cluster attributes, espeically the "power_on_XXX" attributes in certain clusters.

**Additional context**
Device signature with the current pairing:

```
{
  "node_descriptor": "NodeDescriptor(byte1=1, byte2=64, mac_capability_flags=142, manufacturer_code=4107, maximum_buffer_size=82, maximum_incoming_transfer_size=128, server_mask=11264, maximum_outgoing_transfer_size=128, descriptor_capability_field=0)",
  "endpoints": {
    "11": {
      "profile_id": 260,
      "device_type": "0x010d",
      "in_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0008",
        "0x0300",
        "0x1000",
        "0xfc01",
        "0xfc02"
      ],
      "out_clusters": [
        "0x0019"
      ]
    },
    "242": {
      "profile_id": 41440,
      "device_type": "0x0061",
      "in_clusters": [],
      "out_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "Signify Netherlands B.V.",
  "model": "LCA001",
  "class": "zigpy.device.Device"
}
```

Previously working Python script used to modify the power on behavior of this same bulb successfully - no longer works as some attributes are no longer exposed:

```
ieees = data.get("ieees")
logger.info("reconfiguring lights on/off behaviour")
for ieee in ieees:
    logger.info(f"reconfiguring light {ieee} on/off behaviour")
    # set PhillipsOnOffCluster -> power_on_state
    service_data1 = {
        "ieee": ieee,
        "endpoint_id": 11,
        "cluster_id": 6,
        "cluster_type": "in",
        "attribute": 16387,
        "value": 255,
    }
    hass.services.call("zha", "set_zigbee_cluster_attribute", service_data1, False)
    time.sleep(5)
    # set PhillipsLevelControlCluster -> power_on_level
    service_data2 = {
        "ieee": ieee,
        "endpoint_id": 11,
        "cluster_id": 8,
        "cluster_type": "in",
        "attribute": 16384,
        "value": 255,
    }
    hass.services.call("zha", "set_zigbee_cluster_attribute", service_data2, False)
    time.sleep(5)
    # control PhillipsColorCluster -> power_on_color_temprature
    service_data3 = {
        "ieee": ieee,
        "endpoint_id": 11,
        "cluster_id": 768,
        "cluster_type": "in",
        "attribute": 16400,
        "value": 65535,
    }
    hass.services.call("zha", "set_zigbee_cluster_attribute", service_data3, False)
    time.sleep(5)
    # service_data4 = {
    #     "ieee": ieee,
    #     "endpoint_id": 11,
    #     "cluster_id": 768,
    #     "cluster_type": "in",
    #     "attribute": 3,
    #     "value": 65536,
    # }
    # hass.services.call("zha", "set_zigbee_cluster_attribute", service_data4, False)
    # time.sleep(5)
    logger.info(f"Done reconfiguring light {ieee}")
logger.info(f"Done reconfiguring all lights on/off behaviour")
```

SHxKM, 2021-02-06:
I checked the Python virtual environment that and I had HA installed in, unfortunately even after upgrading `zha-quirks` to the latest version these clusters aren’t there.

SHxKM, 2021-02-06:
Upgrading to 2021.2.1 worked beautifully. Device is now properly recognized.

I'm wondering though why didn't individually upgrading zha-quirks and zigpy work, hmmm..

MattWestb, 2021-02-06:
Changes in ZHA-Quirks and Zigpy that is braking if you is not having the right versions.
Look in the latest release notes ZHA-Quirks.

SHxKM, 2021-02-06:
> Changes in ZHA-Quirks and Zigpy that is braking if you is not having the right versions.
> Look in the latest release notes ZHA-Quirks.

Makes sense, but I had already updated zha-quirks and zigpy, so I'm wondering why didn't HA use that until I upgraded it.

closed 2021-02-06 by SHxKM