zigpy/zha-device-handlers #780: [BUG] CINLINELE TS0601_TZE200 classified as router instead of switch

strannick-ru, 2021-02-14:
**Describe the bug**
After adding CINLINELE TS0601_TZE200 to ZHA, the entity is classified as a router.
This device is not only a router, but a switch.
It does not seem possible to change the domain of an entity from router to switch.

I tried changing device_config in the configuration.yaml as described in the [ZHA documentation](https://www.home-assistant.io/integrations/zha/#modifying-the-device-type), but it didn't help. 

**To Reproduce**
Steps to reproduce the behavior:
1. Add CINLINELE TS0601_TZE200 to Home Assistant via ZHA integration
2. Look at Device Type: shows Router.

**Expected behavior**
Entity should be have the domain switch, ID should show switch.

**Screenshots**
[image]

**Additional context**
Running Home Assistant 2021.2.3 + ZHA on Docker with Sonoff ZigBee Bridge (Tasmota flashed).

CINLINELE TS0601_TZE200 device signature:
```{
  "node_descriptor": "NodeDescriptor(byte1=1, byte2=64, mac_capability_flags=142, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=0)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0051",
      "in_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0xef00"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZE200_oisqyl4o",
  "model": "TS0601",
  "class": "zigpy.device.Device"
}```

TheJulianJES, 2021-02-21:
Isn't it expected to show either ``Device Type: Router`` or ``Device Type: EndDevice``?
Or is the problem that there are simply no entities? You're also talking about changing the domain from router to switch but there isn't a "router" domain for entities.

Could you post a screenshot of the whole page (so we can also see if there are any entities).
Please also explain what's not working correctly. Afaik, saying ``Device Type: Router`` is fully expected and correct.

MattWestb, 2021-02-21:
"class": "zigpy.device.Device" = no quirk and tuya = not working.

TheJulianJES, 2021-02-21:
> "class": "zigpy.device.Device" = no quirk and tuya = not working.

So it doesn't show any entities then? (I do have some Tuya devices that *just work* but that seems to be rare)
Still, the issue then is about the device not properly presenting the three relays it has and not that it displays ``Device Type: Router``. That's why I was so confused about this.

Adminiuga, 2021-02-23:
Yeah, original request "this is a router but also a switch" does not have any bearing with ZHA or zigpy. Perhaps need a quirk to be added as a switch.

strannick-ru, 2021-02-24:
> Or is the problem that there are simply no entities?
Yes. Sorry, I described the problem incorrectly.

I tried writing quirk (it is in [linked PR](https://github.com/zigpy/zha-device-handlers/pull/784) here), it works, but with a small problem, I do not know how to fix it.
When I turn on the light with this switch, everything works, but with a delay in the interface and in the logs, I see this:
```
File "/usr/src/homeassistant/homeassistant/components/zha/light.py", line 236, in async_turn_on
  if not isinstance(result, list) or result[1] is not Status.SUCCESS:
```
and this:
```
WARNING (MainThread) [zigpy.zcl] [0x64e4:1:0xef00] Unknown cluster-specific command 36
```

May you tell me, please, what I did wrong and how I can improve this quirk?

TheJulianJES, 2021-02-27:
Perhaps this should be re-opened because of the (now merged) quirk also throwing an exception.

dmulcahey, 2021-02-27:
Sure, it was closed by the merge automatically. I reopened it.

dmulcahey, 2021-02-27:
Should I back that PR out? I missed this as I was merging from my phone...

TheJulianJES, 2021-02-27:
I think it's okay to leave it in. I guess an exception is better than the device not working at all(?)

closed 2021-09-02 by github-actions[bot]