zigpy/zha-device-handlers #782: [BUG] Quirk for Hue Dimmer has been changed to reflect old firmware that was replaced in October 2019!

bobbynobble, 2021-02-15:
**Describe the bug**
The quirk for the Hue Dimmer, zhaquirks.philips.rwl021.PhilipsRWL021, which worked perfectly with the current firmware released in October 2019 has recently been replaced by one that reflects a previous firmware. This has lost several functions and renamed others breaking current set up in Home Assistant.

**To Reproduce**
Events in ZHA are currently...
Click 'I' - returns only 'on'
Click 'up' - returns 'step'
Hold 'up' - returns 'step' as above, followed by 'step' again but with different args
Release 'up' - returns 'stop'
Click 'down' - returns 'step'
Hold 'down' - returns 'step' as above, followed by 'step' again but with different args
Release 'down' - returns 'stop'
Press ''O' - returns 'off_with_effect'

**Expected behavior**
Events in ZHA used to, and should, be...
Click 'I' - on_short_release
Hold 'I' - on_hold
Release 'I' - on_long_release
Click 'up' - up_short_release
Hold 'up' - up_hold
Release 'up' - up_long_release
Click 'down' - down_short_release
Hold 'down' - down_hold
Release 'down' - down_long_release
Click 'O' - off_short_release
Hold 'O' - off_hold
Release 'O' - off_long_release

Fix literally only requires replacing with the pre-existing version.

**Additional context**
```
{
  "node_descriptor": "NodeDescriptor(byte1=2, byte2=64, mac_capability_flags=128, manufacturer_code=4107, maximum_buffer_size=71, maximum_incoming_transfer_size=45, server_mask=0, maximum_outgoing_transfer_size=45, descriptor_capability_field=0)",
  "endpoints": {
    "1": {
      "profile_id": 49246,
      "device_type": "0x0830",
      "in_clusters": [
        "0x0000"
      ],
      "out_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0008"
      ]
    },
    "2": {
      "profile_id": 260,
      "device_type": "0x000c",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x000f",
        "0xfc00"
      ],
      "out_clusters": [
        "0x0019"
      ]
    }
  },
  "manufacturer": "Philips",
  "model": "RWL021",
  "class": "zhaquirks.philips.rwl021.PhilipsRWL021"
}

```

Adminiuga, 2021-02-15:
Can you be a more specific about which version of HA or `zha-device-handlers` broke it? Last change to RWL021 quirk was 8 month ago, that's easily 8 to 10 versions ago.

bobbynobble, 2021-02-15:
Oh, don't know what's happened then. Not sure exactly which version changed things, It was fine with everything up to date, from memory, at the end of the year and it's just after getting everything back up to date recently that the Hue dimmers events changed.

Is there some sort of forced update or some such I can do to try to get things back on track?

bobbynobble, 2021-02-15:
Just seen this... https://github.com/zigpy/zha-device-handlers/issues/755

Sounds familiar but I'm not aware I've picked up a beta version of anything?

TheJulianJES, 2021-02-15:
I'm running firmware ``6.1.1.28573`` / ``0x42006f9d`` on my dimmer switch and it works just fine.
Which version are you running?

bobbynobble, 2021-02-15:
> I'm running firmware `6.1.1.28573` / `0x42006f9d` on my dimmer switch and it works just fine.
> Which version are you running?

Same :(

bobbynobble, 2021-02-21:
After a break due to the 5.11 OS update breaking HA completely, I am back in business using 5.9 combined with a 2020.12 snapshot, like this all my Hue dimmers now return the correct events again, confirming there is nothing wrong with the dimmers themselves or the firmware they are running.

Next I updated all the addons and HACS integrations and all was still good, this only leaves HA itself so next I updated to the current version, 2021.2 and instantly the dimmers reverted to producing the old style zha events. Restoring the 2020.12 snapshot again made them work correctly once more. This would seem to confirm there is something is not right somewhere in the 2021 updates.

Will now try them incrementally to work out exactly where it all goes wrong and report back.

bobbynobble, 2021-02-21:
@TheJulianJES  whatever was introduced with HA 2021.2.0 is what has messed up the Hue Dimmers. 

2021.1.5 works as it should, updating to 2021.2.0 changes them to the old events, roll back to 2021.1.5 and all is good again.

TheJulianJES, 2021-02-21:
The latest version of "ZHA quirks" (zha-device-handlers) is only compatible with the latest zigpy (0.32.0).
Could it be that some custom component is pulling an older version of zigpy?

MattWestb, 2021-02-21:
@Adminiuga  was having one user that was having the same problem (HA updated but ZHA-quirk was false version and braking Philips HUE quirk) but i cant finding the posting.
The user was deleting HA rebooting and installing HA and ZHA and it was working.

TheJulianJES, 2021-02-21:
Here was one user who tried to upgrade both libraries manually: https://github.com/zigpy/zha-device-handlers/issues/769
But I guess you mean something different.

MattWestb, 2021-02-21:
Yes i was looking for the post in HA and Zigpy  git but i was not finding it but it was then the upgrade of the quirk "API" in Zigpy so only some mounts ago.

bobbynobble, 2021-02-21:
> The latest version of "ZHA quirks" (zha-device-handlers) is only compatible with the latest zigpy (0.32.0).
> Could it be that some custom component is pulling an older version of zigpy?

Not that I'm aware of, what should I be looking for and how can I tell which versions it's using?

TheJulianJES, 2021-02-21:
I guess you could use Porinter to get into the homeassistant docker and run ``cat /usr/local/lib/python3.8/site-packages/zigpy/__init__.py``.
How to use Portainer: https://github.com/zigpy/zha-device-handlers#testing-quirks-in-development-in-docker-based-install

MattWestb, 2021-02-21:
If you is  on docker open one command in the HA container and` ls /usr/local/lib/python3.8/site-packages/` and you is getting somthing like this:
```
zeroconf
zeroconf-0.28.8.dist-info
zha_quirks-0.0.53.dist-info
zhaquirks
zhong_hong_hvac
zhong_hong_hvac-1.0.9.dist-info
ziggo_mediabox_xl-1.1.0.dist-info
ziggo_mediabox_xl.py
zigpy
zigpy-0.32.0.dist-info
zigpy_cc
zigpy_cc-0.5.2.dist-info
zigpy_deconz
zigpy_deconz-0.11.1.dist-info
zigpy_xbee
zigpy_xbee-0.13.0.dist-info
zigpy_zigate
zigpy_zigate-0.7.3.dist-info
zigpy_znp
zigpy_znp-0.3.0.dist-info
zipp-3.4.0.dist-info
zipp.py
zm_py-0.5.2.dist-info
zoneminder
zope
zope.interface-5.2.0-py3.8-nspkg.pth
zope.interface-5.2.0.dist-info
zwave_js_server
zwave_js_server_python-0.17.2.dist-info
bash-5.0#
```
And the `zha_quirks-0.0.53.dist-info` shall being the installed version is you not have  installing on in the config folder.

bobbynobble, 2021-02-22:
This is what I got...

HA 2021.1.5
zigpy - 0.30.0
zha-quirks - 0.0.51

HA 2021.2.0
zigpy - 0.32.0
zha-quirks - 0.0.53

Adminiuga, 2021-02-22:
@TheJulianJES does it work for you on 2021.2.x?

TheJulianJES, 2021-02-22:
Yeah, I was also running the HA 2021.2.x beta. However, I manually updated zha-quirks a couple of times on 2021.2.3 now, but there weren't like any changes that could do this I guess. I'll see if I can revert later and if the issue then occurs.

bobbynobble, 2021-02-27:
> Yeah, I was also running the HA 2021.2.x beta. However, I manually updated zha-quirks a couple of times on 2021.2.3 now, but there weren't like any changes that could do this I guess. I'll see if I can revert later and if the issue then occurs.

Any luck on finding the problem?

TheJulianJES, 2021-02-28:
I wasn't able to find the problem. Can you perhaps upgrade to the latest Home Assistant beta and check there? (Maybe also re-pair the dimmer)
2021.3.0b3 is currently the latest version.

bobbynobble, 2021-02-28:
> I can't reproduce the issue. Can you perhaps upgrade to the latest Home Assistant beta and check there? (Maybe also re-pair the dimmer)
> 2021.3.0b3 is currently the latest version.

Will do

bobbynobble, 2021-02-28:
> I can't reproduce the issue. Can you perhaps upgrade to the latest Home Assistant beta and check there? (Maybe also re-pair the dimmer)
> 2021.3.0b3 is currently the latest version.

Exactly the same behaviour with 2021.3.0b3 even after repairing a remote.

Get this error in the logs on start up now...

`Traceback (most recent call last):
  File "/usr/local/lib/python3.8/site-packages/bellows/ezsp/__init__.py", line 297, in handle_callback
    handler(*args)
  File "/usr/local/lib/python3.8/site-packages/bellows/zigbee/application.py", line 219, in ezsp_callback_handler
    self._handle_frame(*args)
  File "/usr/local/lib/python3.8/site-packages/bellows/zigbee/application.py", line 264, in _handle_frame
    self.handle_message(
  File "/usr/local/lib/python3.8/site-packages/zigpy/application.py", line 223, in handle_message
    return sender.handle_message(
  File "/usr/local/lib/python3.8/site-packages/zigpy/device.py", line 279, in handle_message
    return endpoint.handle_message(
  File "/usr/local/lib/python3.8/site-packages/zigpy/endpoint.py", line 217, in handle_message
    handler(hdr, args, dst_addressing=dst_addressing)
  File "/usr/local/lib/python3.8/site-packages/zigpy/zcl/__init__.py", line 201, in handle_message
    self.handle_cluster_request(hdr, args, dst_addressing=dst_addressing)
TypeError: handle_cluster_request() got an unexpected keyword argument 'dst_addressing'`

bobbynobble, 2021-03-07:
Nothing? Don't want to go back to zigbee2mqtt :)

TheJulianJES, 2021-03-07:
If you have another Zigbee stick, are you willing to setup a clean HassOS installation on some test machine and only install ZHA and join the dimmer?
Since I can't reproduce your issue, I'm not sure how else this can be debugged. A clean HassOS installation should basically give us both the same environment.

bobbynobble, 2021-03-07:
Thanks. Yes, I still have the old CC2531 I used before swapping to the Sonoff ZBBridge, will give it a go and let you know.

Adminiuga, 2021-03-07:
> line 201, in handle_message self.handle_cluster_request(hdr, args, dst_addressing=dst_addressing) TypeError: handle_cluster_request()

You have incorrect version of zha-device-handlers and/or zigpy installed. Update both to the latest one define in manifest.json

bobbynobble, 2021-03-07:
> If you have another Zigbee stick, are you willing to setup a clean HassOS installation on some test machine and only install ZHA and join the dimmer?
> Since I can't reproduce your issue, I'm not sure how else this can be debugged. A clean HassOS installation should basically give us both the same environment.
> 
> I'll also try to find out what the error in your logs is about.

Progress! Works perfectly on new install, breathing easier now thanks. Just need to fix my existing insistence without starting from scratch?

@Adminiuga  when I checked versions before they were confirmed as correct, see post above, so what do I need to do to update?

Adminiuga, 2021-03-07:
That error you would only get if zigpy and zha-device-handlers version combination is incorrect. How it got to that stage? - impossible to tell from here. Perhaps you are still overriding it somewhere.
Nobody else is reporting same issue and there are quite a few users of hue remotes.

bobbynobble, 2021-03-07:
How would I check, not done anything consciously? Every other device in ZHA works perfectly so why specifically the Hue dimmer picking up the wrong info?

Adminiuga, 2021-03-07:
Because hue dimmer quirk has the handle_cluster_command() method. How were you installing unreleased version of zha-device-handlers? Start from there

bobbynobble, 2021-03-07:
I haven't installed anything that's been unreleased, all changes came with full HA updates and it was the change to 2012.2 that caused this issue as I said previously.

Adminiuga, 2021-03-07:
> Works perfectly on new install,

But not on your old install? It does not add up. Again, if there were issues with the versions pinned by zha, we would hear by now from far more people.

bobbynobble, 2021-03-07:
I know it doesn't, hence why I'm here. Can the ZBBridge be affecting things in any way?

TheJulianJES, 2021-03-07:
> I know it doesn't, hence why I'm here. Can the ZBBridge be affecting things in any way?

No. Can you list all folders you have in your custom_components?

MattWestb, 2021-03-07:
And `/usr/local/lib/python3.8/site-packages/` pleas.

bobbynobble, 2021-03-07:
Custom components...
hacs
alexa_media
spotcast
upnp_availablity
panasonic_viera
fcm_android

Addons...
AppDaemon 4
Check Home Assistant configuration
Duck DNS
File editor
Home Assistant Google Drive Backup
Mosquitto broker
Portainer
SSH & Web Terminal
Samba share
TasmoAdmin
Terminal & SSH

bobbynobble, 2021-03-07:
> And `/usr/local/lib/python3.8/site-packages/` pleas.

When I run 'ls /usr/local/lib/python3.8/site-packages/' in Portainer the list whizzes up the screen and vanishes, how do I stop it so I can copy it?

MattWestb, 2021-03-07:
The interesting lines is:
```
zha_quirks-0.0.54.dist-info
zhaquirks
zigpy
zigpy-0.32.0.dist-info
zigpy_cc
zigpy_cc-0.5.2.dist-info
zigpy_deconz
zigpy_deconz-0.11.1.dist-info
zigpy_xbee
zigpy_xbee-0.13.0.dist-info
zigpy_zigate
zigpy_zigate-0.7.3.dist-info
zigpy_znp
zigpy_znp-0.4.0.dist-info
```
And its one the last page in portainer so you dont need scrolling.

bobbynobble, 2021-03-07:
Ah...was sending the command before opening the console window so the console just vanished after running, got it now...

zha_quirks-0.0.54.dist-info
zhaquirks
zhong_hong_hvac
zhong_hong_hvac-1.0.9.dist-info
ziggo_mediabox_xl-1.1.0.dist-info
ziggo_mediabox_xl.py
zigpy
zigpy-0.32.0.dist-info
zigpy_cc
zigpy_cc-0.5.2.dist-info
zigpy_deconz
zigpy_deconz-0.11.1.dist-info
zigpy_xbee
zigpy_xbee-0.13.0.dist-info
zigpy_zigate
zigpy_zigate-0.7.3.dist-info
zigpy_znp
zigpy_znp-0.4.0.dist-info
zipp-3.4.1.dist-info
zipp.py
zm_py-0.5.2.dist-info
zoneminder
zope
zope.interface-5.2.0-py3.8-nspkg.pth
zope.interface-5.2.0.dist-info
zwave_js_server
zwave_js_server_python-0.21.1.dist-info

MattWestb, 2021-03-07:
I was forgetting EZSP =  bellows is also needed but you cant scrolling to it must do it with ls like:
`ls /usr/local/lib/python3.8/site-packages/bello`

And the interesting line is the:
`usr/local/lib/python3.8/site-packages/bellows-0.22.0.dist-info:`

bobbynobble, 2021-03-07:
```
bash-5.0# ls /usr/local/lib/python3.8/site-packages/bello
ls: /usr/local/lib/python3.8/site-packages/bello: No such file or directory
```

MattWestb, 2021-03-07:
Sorry i was missing one star.
`ls /usr/local/lib/python3.8/site-packages/bello*`

bobbynobble, 2021-03-07:
No probs...

/usr/local/lib/python3.8/site-packages/bellows-0.22.0.dist-info:
COPYING               METADATA              WHEEL                 top_level.txt
INSTALLER             RECORD                dependency_links.txt
LICENSE               REQUESTED             entry_points.txt

MattWestb, 2021-03-07:
Then all Zigpy libs in the HA container is current version.
If you dont have changing some of the files after the last HA update in the container all shall being OK,

Do you have somthing in the config deps ?

`ls /config/deps/`

Its being temp version if installed  libs.

If yes looks for the names of the libs in the  tree.

bobbynobble, 2021-03-07:
Just get this...

`bash-5.0# ls /config/deps/
lib
bash-5.0# ls /config/deps/lib
python3.8`

MattWestb, 2021-03-07:
Then you is having somthing installed.
Lock thru the tree structure wot lib is installed.

MattWestb, 2021-03-07:
Or you is deleting all in the deps folder and restarting HA and its away.

bobbynobble, 2021-03-07:
Fantastic, all up to date and working as it should now.

Thank you for your patience and apologies for assuming it was something in your update.

closed 2021-03-08 by bobbynobble